arkypita/LaserGRBL #351: Laser GRBL has stopped working - Windows is checking for solution to the problem

ihab2010, 2018-05-06:
Hello,
It is my first time to Laser GRBL.
I like your software too much, I have seen it last week with one of my students.
Since then, it did not work.
I change the Arduino Uno card, I changed USB cable, I have changed USB port.
I have uninstalled Laser Grbl 2.8.45 and installed 2.8.30 instead.
It did not solve the problem.
What else could it be?
Best regards.
Ihab  

[image]

arkypita, 2018-05-07:
do you have .net framework 3.5 installed?
https://www.microsoft.com/it-it/download/details.aspx?id=21

ihab2010, 2018-05-07:
Thank you arkypita for your prompt reply.
Yes, you right.
Once I installed .net framework 3.5 
 Laser Grbl open.
But unfortunately when I connect status became connected for few seconds and disconnected.
What could be the problem now?
Waiting for your prompt reply.
I am very excited to start using it on my project.
Thanks again.

arkypita, 2018-05-08:
too generic

is grbl installed on your arduino? what version?
is correct USB driver installed to see arduino com port?

ihab2010, 2018-05-09:
Thank you arkypita for your prompt reply.
Yes,  you right.
My Arduino was having an old version of Grbl.
I thought it will work.
now I uploaded a new verion

[image]

now it works.
Many thanks for your understanding an cooperation.
Ihab

closed 2018-05-18 by arkypita