arkypita/LaserGRBL #352: Homing fails

Delaser, 2018-05-07:
Hopefully user error, but I'm stumped on the issue.

Just updated my laser to GRBL 1.1f
Using an uno/gshield V3
LaserGRBL v2.8.45

Everything was working fine before the update.
Post update, when I hit the find Home button, from both bottom tray, custom button and Grbl menu, I get:
Homing Fail. Could not find limit switch within search distances.
Try increasing max travel or decreasing pull-off distance or check wiring.

Now, that would make sense, however the homing button doesn't even attempt to move towards my limit switches, it just hangs for a moment then throws the error.

My hardware switches otherwise work fine, laser triggers ok, jogging works no issues.

My config export is here:
 
> $0=10 (Step pulse time)
$1=25 (Step idle delay)
$2=0 (Step pulse invert)
$3=1 (Step direction invert)
$4=0 (Invert step enable pin)
$5=0 (Invert limit pins)
$6=0 (Invert probe pin)
$10=1 (Status report options)
$11=0.010 (Junction deviation)
$12=0.002 (Arc tolerance)
$13=0 (Report in inches)
$20=0 (Soft limits enable)
$21=1 (Hard limits enable)
$22=1 (Homing cycle enable)
$23=3 (Homing direction invert)
$24=2500.000 (Homing locate feed rate)
$25=5000.000 (Homing search seek rate)
$26=0 (Homing switch debounce delay)
$27=10.000 (Homing switch pull-off distance)
$30=1000 (Maximum spindle speed)
$31=0 (Minimum spindle speed)
$32=1 (Laser-mode enable)
$100=160.000 (X-axis travel resolution)
$101=160.000 (Y-axis travel resolution)
$102=1.000 (Z-axis travel resolution)
$110=2000.000 (X-axis maximum rate)
$111=2000.000 (Y-axis maximum rate)
$112=2000.000 (Z-axis maximum rate)
$120=500.000 (X-axis acceleration)
$121=500.000 (Y-axis acceleration)
$122=100.000 (Z-axis acceleration)
$130=160.000 (X-axis maximum travel)
$131=190.000 (Y-axis maximum travel)
$132=200.000 (Z-axis maximum travel)

Thanks for your time.

arkypita, 2018-05-07:
1. how many axes you use? two or three?
2. read this: https://github.com/grbl/grbl/wiki/Connecting-Grbl#grbls-pins
3. if you have update GRBL and you have problem with GRBL please DO NOT ASK HELP in LaserGRBL project board but in grbl one!!!!

closed 2018-05-07 by arkypita

Delaser, 2018-05-07:
1 - 2 Axis

2 - My wiring is fine, it worked 100% before upgrade

3 - I can't tell if its a problem with the GRBL or with your LaserGRBL not talking to the GRBL, thought I'd arbitrarily pick LaserGRBL to ask about first.

arkypita, 2018-05-07:
Ok. If you have updated and you have 2 axis limit switches... did you have changed config.h of grbl library to manage XY homing instead of XYZ homing?

arkypita, 2018-05-07:
https://github.com/gnea/grbl/blob/5c8dcefcbdc72e953fb07c5f8bb8ef4925ac6b5b/grbl/config.h#L109

Delaser, 2018-05-07:
Ahhha. No. I didn't know that was a thing. I don't recall having to edit anything in the config when I originally setup.

Ill take a poke at that in the morning and report back.

Thanks much!

arya72, 2019-03-20:
I have build my own LaserGRBL machine and its working, but the 'Homing' isn't work. I have edit 'config.h', i comment the red one and uncomment the blue one but its still not working. How to setting the 'Homing' in LaserGRBL? I use NO limit switch, two for each axis. Thx @arkypita 
[image]

arkypita, 2019-03-20:
I usually use the version that home X and Y the same time:
` #define HOMING_CYCLE_0 ((1<<X_AXIS)|(1<<Y_AXIS))  // NOT COMPATIBLE WITH COREXY: Homes both X-Y in one cycle. `
But should be ok also your config.

Did you have enabled homing in $$ configuration?

[image]

What append when you push Homing button?

[image]

arya72, 2019-03-20:
Yes I did it.
This is what happened when I pushed the Homing Button
[image]

arya72, 2019-03-20:
This is my GRBL configuration:

$0=10 (Step pulse time)
$1=25 (Step idle delay)
$2=0 (Step pulse invert)
$3=0 (Step direction invert)
$4=0 (Invert step enable pin)
$5=0 (Invert limit pins)
$6=0 (Invert probe pin)
$10=1 (Status report options)
$11=0.010 (Junction deviation)
$12=0.002 (Arc tolerance)
$13=0 (Report in inches)
$20=1 (Soft limits enable)
$21=1 (Hard limits enable)
$22=1 (Homing cycle enable)
$23=0 (Homing direction invert)
$24=1000.000 (Homing locate feed rate)
$25=5000.000 (Homing search seek rate)
$26=250 (Homing switch debounce delay)
$27=1.000 (Homing switch pull-off distance)
$30=255 (Maximum spindle speed)
$31=0 (Minimum spindle speed)
$32=1 (Laser-mode enable)
$100=80.000 (X-axis travel resolution)
$101=80.000 (Y-axis travel resolution)
$102=80.000 (Z-axis travel resolution)
$110=10000.000 (X-axis maximum rate)
$111=10000.000 (Y-axis maximum rate)
$112=10000.000 (Z-axis maximum rate)
$120=1000.000 (X-axis acceleration)
$121=1000.000 (Y-axis acceleration)
$122=1000.000 (Z-axis acceleration)
$130=400.000 (X-axis maximum travel)
$131=420.000 (Y-axis maximum travel)
$132=200.000 (Z-axis maximum travel)

arkypita, 2019-03-20:
If "homing fail" appare exactly when you press homing button, and axis does not move at all, it means that grbl see the switch as "already pressed". With $5=0 this means that you have logic low level on Limit-X or Limit-Y pin.

If axis moves one or both direction and "crash" to the fisical limit without stops, and "homing fail" appare some seconds later from when you press homing button, it means that your switch is not well connected.

However this problem is not related to LaserGRBL itself, but to your hardware setup or grbl configuration.

arya72, 2019-03-20:
So, if I invert the limit pins with $ 5 = 1 can it solve my problem? or do you have other solutions?

arkypita, 2019-03-20:
With NO switch $5=0 should be ok because of internal pull-up keeps pin value high unless you close the circuit to the ground with switch.

Did you put any pull up resistor? Filtering capacitor?
Maybe there is some noise from stepper then trigger your switch pin.

There is a lot of literature about homing and limit switches. Just search on web starting with gnea/grbl site and google.

https://github.com/gnea/grbl/wiki/Wiring-Limit-Switches
https://github.com/gnea/grbl/wiki/Frequently-Asked-Questions#homing-cycle-isnt-working-right-the-movements-are-all-going-in-wrong-directions-or-its-not-stopping-when-hitting-the-limit-switch
https://www.google.it/search?client=opera&q=grbl+homing+fail&sourceid=opera&ie=UTF-8&oe=UTF-8

arya72, 2019-03-20:
I didn't put any pull up resistor or filtering capacitor, I'll try it first, thanks for helping me

arkypita, 2019-03-20:
For NO I suggest the schematic NO with noise filtering

[image]

arya72, 2019-03-26:
May I look your "config.h"? @arkypita