arkypita/LaserGRBL #353: Как изменить отправную точку лазера?

Jackyboy1988, 2018-05-10:
Доброго веремени суток всем! Подскажите, пожалуйста, возможно ли изменить точку отправки лазера с нижней левой на верхнюю левую в LaserGRBL? Пользовался EleksCam, так там  можно было позиционировать картинку вокруг отправной точки, а в LaserGRBL не могу понять как. Помогите, пожалуйста! :)

Arhimed-Ru, 2018-05-10:
Открой тему 
translation of the program into another language (Russian localization)
там ссылка на полностью русифицированную версию - может станет понятней.
А вообще - почитай тут странички. Где то видел подобный вопрос и ответ на него (на английском), но уже не помню где.

Revenger72, 2018-05-22:
Тоже хотел это узнать (потому, что пытаюсь использовать с лазером сo2), но автор толком ничего не ответил (только типа читать мануал), но реально изменения точки Home в программе нет. Можно захомиться (если концевики в нижнем левом то проще) и затем обнулить рабочие координаты на ноль в любом месте стола. Хотя это не удобно.

Arhimed-Ru, 2018-05-22:
Вот тут Xillius178  показывал:
https://github.com/arkypita/LaserGRBL/issues/345

Revenger72, 2018-05-23:
Не совсем понял тот пост... Просто поменять направление моторов и задав в Арткаме нулевую точку "верх/лево" лазергрбл сама определит ноль слева вверху? Просто я задавал точку ноль в левом верхнем, и при движении стрелками в программе вниз, координаты шли в минус. Или стрелками слева в программе лазергрбл тогда пользоваться наоборот? Или все изображения нужно будет подразумевать как перевёрнутые?

Arhimed-Ru, 2018-07-05:
Давеча меня просили сделать полный русский перевод версии 2.8.51
Т.к. там тему закрыли - даю ссылку на скачивание архива тут:
https://drive.google.com/open?id=1IvdAXS6rn9QBXKK8iVx1-kfm_DzMC766
[image]

Revenger72, 2018-07-05:
Прелестно! Попробую с переводом позже (сгорел нах лазер), спасибо за труды!
Кстати, с какой версии кнопки двух функций пошли? Чот я качал 2.8.5 кажется, там не увидел возможности такой (или не внимательно смотрел).

Mikalay31, 2018-07-22:
День добрый. Смотрю у Вас много новых кнопок. Не поделитесь ?

StuartB4, 2018-07-22:
2.8.51

Mikalay31, 2018-07-23:
English:  I understood you ! I'm already downloading. Thank you !     ( RUS:  Я понял тебя ! Уже скачиваю. Спасибо ! )

Mikalay31, 2018-07-26:
I see Arhimed-Ru has a lot of buttons. Discard the text of the commands that were used ... Or a link where I can read about them. Thank you !

Nicholas.