arkypita/LaserGRBL #354: Second instance lock up

BallscrewBob, 2018-05-12:
Have not been able to nail this one down in any form but since the latest release a second instance sometime just locks up solid. 
It is a random event and usually I only have these running with no other major tasks as I can often do multiple jobs on either copy.

The first instance is still fully functional in every respect.

The second copy will not respond to any commands / mouse presses at all.
I wait until the other instance has completed then shut down the running version normally and then use task manager to close out the errant copy.

At that point I run CCLEANER and WISE REGISTRY cleaner and do a computer restart.

This is the log section from the second instance.
From that I would guess it locks up just after completion.
I had told it to do another pass as the first was a little light.

`
11/05/2018 2:39:45 PM	OpenFile    	Open D:\ARD ESP PI stuff\CNC\NC Files\pics\Triumph-Motorcycles.jpg
11/05/2018 2:41:09 PM	SetStatus   	Machine status [Jog]
11/05/2018 2:41:12 PM	SetStatus   	Machine status [Idle]
11/05/2018 2:42:14 PM	EnqueueProgram	Running program, 262944 lines
11/05/2018 2:42:14 PM	SetStatus   	Machine status [Run]
11/05/2018 3:46:39 PM	ProgramEnd  	Job Executed: 27929 lines, 0 errors, 1 hour,9 min
11/05/2018 3:47:13 PM	SetStatus   	Machine status [Idle]
11/05/2018 3:56:12 PM	SendImmediate	Send Immediate Command [0x9C]
11/05/2018 3:56:12 PM	SendImmediate	Send Immediate Command [0x9C]
11/05/2018 3:56:13 PM	SendImmediate	Send Immediate Command [0x9C]
11/05/2018 3:56:13 PM	SendImmediate	Send Immediate Command [0x9C]
11/05/2018 3:56:13 PM	SendImmediate	Send Immediate Command [0x9C]
11/05/2018 3:56:13 PM	SendImmediate	Send Immediate Command [0x9C]
11/05/2018 3:56:16 PM	SendImmediate	Send Immediate Command [0x9C]
11/05/2018 3:56:16 PM	SendImmediate	Send Immediate Command [0x9C]
11/05/2018 3:56:16 PM	SendImmediate	Send Immediate Command [0x9C]
11/05/2018 3:56:16 PM	SendImmediate	Send Immediate Command [0x9C]
11/05/2018 3:56:16 PM	SendImmediate	Send Immediate Command [0x9C]
11/05/2018 3:56:22 PM	SendImmediate	Send Immediate Command [0x92]
11/05/2018 3:56:23 PM	SendImmediate	Send Immediate Command [0x94]
11/05/2018 3:56:23 PM	SendImmediate	Send Immediate Command [0x92]
11/05/2018 3:56:23 PM	SendImmediate	Send Immediate Command [0x94]
11/05/2018 3:56:23 PM	SendImmediate	Send Immediate Command [0x94]
11/05/2018 3:56:25 PM	SendImmediate	Send Immediate Command [0x94]
11/05/2018 3:56:25 PM	SendImmediate	Send Immediate Command [0x94]
11/05/2018 3:56:25 PM	SendImmediate	Send Immediate Command [0x94]
11/05/2018 3:56:25 PM	SendImmediate	Send Immediate Command [0x94]
11/05/2018 3:56:26 PM	SendImmediate	Send Immediate Command [0x94]
11/05/2018 3:56:26 PM	SendImmediate	Send Immediate Command [0x94]
11/05/2018 3:56:29 PM	EnqueueProgram	Running program, 27929 lines
11/05/2018 3:56:29 PM	SetStatus   	Machine status [Run]
11/05/2018 5:11:20 PM	ProgramEnd  	Job Executed: 27929 lines, 0 errors, 1 hour,14 min
11/05/2018 5:11:54 PM	SetStatus   	Machine status [Idle]
11/05/2018 5:43:08 PM	SendImmediate	Send Immediate Command [0x9C]
11/05/2018 5:43:08 PM	SendImmediate	Send Immediate Command [0x9C]
11/05/2018 5:43:08 PM	SendImmediate	Send Immediate Command [0x9C]
11/05/2018 5:43:08 PM	SendImmediate	Send Immediate Command [0x9C]
11/05/2018 5:43:08 PM	SendImmediate	Send Immediate Command [0x9C]
11/05/2018 5:43:09 PM	SendImmediate	Send Immediate Command [0x9C]
11/05/2018 5:43:09 PM	SendImmediate	Send Immediate Command [0x9C]
11/05/2018 5:43:09 PM	SendImmediate	Send Immediate Command [0x9C]
11/05/2018 5:43:09 PM	SendImmediate	Send Immediate Command [0x9C]
11/05/2018 5:43:10 PM	SendImmediate	Send Immediate Command [0x9C]
11/05/2018 5:43:10 PM	SendImmediate	Send Immediate Command [0x9C]
11/05/2018 5:43:10 PM	SendImmediate	Send Immediate Command [0x9C]
11/05/2018 5:43:12 PM	SendImmediate	Send Immediate Command [0x94]
11/05/2018 5:43:12 PM	SendImmediate	Send Immediate Command [0x94]
11/05/2018 5:43:13 PM	SendImmediate	Send Immediate Command [0x94]
11/05/2018 5:43:13 PM	SendImmediate	Send Immediate Command [0x94]
11/05/2018 5:43:13 PM	SendImmediate	Send Immediate Command [0x94]
11/05/2018 5:43:13 PM	SendImmediate	Send Immediate Command [0x94]
11/05/2018 5:43:14 PM	SendImmediate	Send Immediate Command [0x94]
11/05/2018 5:43:14 PM	SendImmediate	Send Immediate Command [0x94]
11/05/2018 5:43:14 PM	SendImmediate	Send Immediate Command [0x94]
11/05/2018 5:43:14 PM	SendImmediate	Send Immediate Command [0x94]
11/05/2018 5:43:14 PM	SendImmediate	Send Immediate Command [0x92]
11/05/2018 5:43:15 PM	SendImmediate	Send Immediate Command [0x94]
11/05/2018 5:43:15 PM	SendImmediate	Send Immediate Command [0x94]
11/05/2018 5:43:15 PM	SendImmediate	Send Immediate Command [0x94]
11/05/2018 5:43:15 PM	SendImmediate	Send Immediate Command [0x94]
11/05/2018 5:43:21 PM	SetStatus   	Machine status [Jog]
11/05/2018 5:43:25 PM	SetStatus   	Machine status [Idle]
11/05/2018 5:43:28 PM	SetStatus   	Machine status [Jog]
11/05/2018 5:43:32 PM	SetStatus   	Machine status [Idle]
11/05/2018 5:43:37 PM	SetStatus   	Machine status [Jog]
11/05/2018 5:43:40 PM	SetStatus   	Machine status [Idle]
11/05/2018 5:43:58 PM	SetStatus   	Machine status [Jog]
11/05/2018 5:44:05 PM	SetStatus   	Machine status [Idle]
11/05/2018 5:44:11 PM	EnqueueProgram	Running program, 27929 lines
11/05/2018 5:44:11 PM	SetStatus   	Machine status [Run]
11/05/2018 7:05:57 PM	ProgramEnd  	Job Executed: 27929 lines, 0 errors, 1 hour,21 min
11/05/2018 7:06:31 PM	SetStatus   	Machine status [Idle]
`

BallscrewBob, 2018-05-17:
Additional information

`
See the end of this message for details on invoking 
just-in-time (JIT) debugging instead of this dialog box.

************** Exception Text **************
System.NullReferenceException: Object reference not set to an instance of an object.
   at LaserGRBL.UserControls.CommandLog.OnPaint(PaintEventArgs e)
   at System.Windows.Forms.Control.PaintWithErrorHandling(PaintEventArgs e, Int16 layer, Boolean disposeEventArgs)
   at System.Windows.Forms.Control.WmPaint(Message& m)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

************** Loaded Assemblies **************
mscorlib
    Assembly Version: 2.0.0.0
    Win32 Version: 2.0.50727.8784 (QFE.050727-8700)
    CodeBase: file:///C:/Windows/Microsoft.NET/Framework64/v2.0.50727/mscorlib.dll
----------------------------------------
LaserGRBL
    Assembly Version: 2.8.45.0
    Win32 Version: 2.8.45.0
    CodeBase: file:///C:/Program%20Files%20(x86)/LaserGRBL/LaserGRBL.exe
----------------------------------------
System.Drawing
    Assembly Version: 2.0.0.0
    Win32 Version: 2.0.50727.5495 (Win7SP1GDR.050727-5400)
    CodeBase: file:///C:/Windows/assembly/GAC_MSIL/System.Drawing/2.0.0.0__b03f5f7f11d50a3a/System.Drawing.dll
----------------------------------------
System
    Assembly Version: 2.0.0.0
    Win32 Version: 2.0.50727.8770 (QFE.050727-8700)
    CodeBase: file:///C:/Windows/assembly/GAC_MSIL/System/2.0.0.0__b77a5c561934e089/System.dll
----------------------------------------
System.Windows.Forms
    Assembly Version: 2.0.0.0
    Win32 Version: 2.0.50727.5491 (Win7SP1GDR.050727-5400)
    CodeBase: file:///C:/Windows/assembly/GAC_MSIL/System.Windows.Forms/2.0.0.0__b77a5c561934e089/System.Windows.Forms.dll
----------------------------------------
System.Core
    Assembly Version: 3.5.0.0
    Win32 Version: 3.5.30729.5420 built by: Win7SP1
    CodeBase: file:///C:/Windows/assembly/GAC_MSIL/System.Core/3.5.0.0__b77a5c561934e089/System.Core.dll
----------------------------------------
System.Xml
    Assembly Version: 2.0.0.0
    Win32 Version: 2.0.50727.8773 (QFE.050727-8700)
    CodeBase: file:///C:/Windows/assembly/GAC_MSIL/System.Xml/2.0.0.0__b77a5c561934e089/System.Xml.dll
----------------------------------------
System.Configuration
    Assembly Version: 2.0.0.0
    Win32 Version: 2.0.50727.5483 (Win7SP1GDR.050727-5400)
    CodeBase: file:///C:/Windows/assembly/GAC_MSIL/System.Configuration/2.0.0.0__b03f5f7f11d50a3a/System.Configuration.dll
----------------------------------------
Accessibility
    Assembly Version: 2.0.0.0
    Win32 Version: 2.0.50727.5483 (Win7SP1GDR.050727-5400)
    CodeBase: file:///C:/Windows/assembly/GAC_MSIL/Accessibility/2.0.0.0__b03f5f7f11d50a3a/Accessibility.dll
----------------------------------------

************** JIT Debugging **************
To enable just-in-time (JIT) debugging, the .config file for this
application or computer (machine.config) must have the
jitDebugging value set in the system.windows.forms section.
The application must also be compiled with debugging
enabled.

For example:

<configuration>
    <system.windows.forms jitDebugging="true" />
</configuration>

When JIT debugging is enabled, any unhandled exception
will be sent to the JIT debugger registered on the computer
rather than be handled by this dialog box.

`

BallscrewBob, 2018-05-17:
Additional Information for this issue.

> See the end of this message for details on invoking 
just-in-time (JIT) debugging instead of this dialog box.

************** Exception Text **************
System.NullReferenceException: Object reference not set to an instance of an object.
   at LaserGRBL.UserControls.CommandLog.OnPaint(PaintEventArgs e)
   at System.Windows.Forms.Control.PaintWithErrorHandling(PaintEventArgs e, Int16 layer, Boolean disposeEventArgs)
   at System.Windows.Forms.Control.WmPaint(Message& m)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

************** Loaded Assemblies **************
mscorlib
    Assembly Version: 2.0.0.0
    Win32 Version: 2.0.50727.8784 (QFE.050727-8700)
    CodeBase: file:///C:/Windows/Microsoft.NET/Framework64/v2.0.50727/mscorlib.dll
----------------------------------------
LaserGRBL
    Assembly Version: 2.8.45.0
    Win32 Version: 2.8.45.0
    CodeBase: file:///C:/Program%20Files%20(x86)/LaserGRBL/LaserGRBL.exe
----------------------------------------
System.Drawing
    Assembly Version: 2.0.0.0
    Win32 Version: 2.0.50727.5495 (Win7SP1GDR.050727-5400)
    CodeBase: file:///C:/Windows/assembly/GAC_MSIL/System.Drawing/2.0.0.0__b03f5f7f11d50a3a/System.Drawing.dll
----------------------------------------
System
    Assembly Version: 2.0.0.0
    Win32 Version: 2.0.50727.8770 (QFE.050727-8700)
    CodeBase: file:///C:/Windows/assembly/GAC_MSIL/System/2.0.0.0__b77a5c561934e089/System.dll
----------------------------------------
System.Windows.Forms
    Assembly Version: 2.0.0.0
    Win32 Version: 2.0.50727.5491 (Win7SP1GDR.050727-5400)
    CodeBase: file:///C:/Windows/assembly/GAC_MSIL/System.Windows.Forms/2.0.0.0__b77a5c561934e089/System.Windows.Forms.dll
----------------------------------------
System.Core
    Assembly Version: 3.5.0.0
    Win32 Version: 3.5.30729.5420 built by: Win7SP1
    CodeBase: file:///C:/Windows/assembly/GAC_MSIL/System.Core/3.5.0.0__b77a5c561934e089/System.Core.dll
----------------------------------------
System.Xml
    Assembly Version: 2.0.0.0
    Win32 Version: 2.0.50727.8773 (QFE.050727-8700)
    CodeBase: file:///C:/Windows/assembly/GAC_MSIL/System.Xml/2.0.0.0__b77a5c561934e089/System.Xml.dll
----------------------------------------
System.Configuration
    Assembly Version: 2.0.0.0
    Win32 Version: 2.0.50727.5483 (Win7SP1GDR.050727-5400)
    CodeBase: file:///C:/Windows/assembly/GAC_MSIL/System.Configuration/2.0.0.0__b03f5f7f11d50a3a/System.Configuration.dll
----------------------------------------
Accessibility
    Assembly Version: 2.0.0.0
    Win32 Version: 2.0.50727.5483 (Win7SP1GDR.050727-5400)
    CodeBase: file:///C:/Windows/assembly/GAC_MSIL/Accessibility/2.0.0.0__b03f5f7f11d50a3a/Accessibility.dll
----------------------------------------

************** JIT Debugging **************
To enable just-in-time (JIT) debugging, the .config file for this
application or computer (machine.config) must have the
jitDebugging value set in the system.windows.forms section.
The application must also be compiled with debugging
enabled.

For example:

<configuration>
    <system.windows.forms jitDebugging="true" />
</configuration>

When JIT debugging is enabled, any unhandled exception
will be sent to the JIT debugger registered on the computer
rather than be handled by this dialog box.

BallscrewBob, 2018-05-17:
************** Loaded Assemblies **************
mscorlib
    Assembly Version: 2.0.0.0
    Win32 Version: 2.0.50727.8784 (QFE.050727-8700)
    CodeBase: file:///C:/Windows/Microsoft.NET/Framework64/v2.0.50727/mscorlib.dll
----------------------------------------
LaserGRBL
    Assembly Version: 2.8.45.0
    Win32 Version: 2.8.45.0
    CodeBase: file:///C:/Program%20Files%20(x86)/LaserGRBL/LaserGRBL.exe
----------------------------------------
System.Drawing
    Assembly Version: 2.0.0.0
    Win32 Version: 2.0.50727.5495 (Win7SP1GDR.050727-5400)
    CodeBase: file:///C:/Windows/assembly/GAC_MSIL/System.Drawing/2.0.0.0__b03f5f7f11d50a3a/System.Drawing.dll
----------------------------------------
System
    Assembly Version: 2.0.0.0
    Win32 Version: 2.0.50727.8770 (QFE.050727-8700)
    CodeBase: file:///C:/Windows/assembly/GAC_MSIL/System/2.0.0.0__b77a5c561934e089/System.dll
----------------------------------------
System.Windows.Forms
    Assembly Version: 2.0.0.0
    Win32 Version: 2.0.50727.5491 (Win7SP1GDR.050727-5400)
    CodeBase: file:///C:/Windows/assembly/GAC_MSIL/System.Windows.Forms/2.0.0.0__b77a5c561934e089/System.Windows.Forms.dll
----------------------------------------
System.Core
    Assembly Version: 3.5.0.0
    Win32 Version: 3.5.30729.5420 built by: Win7SP1
    CodeBase: file:///C:/Windows/assembly/GAC_MSIL/System.Core/3.5.0.0__b77a5c561934e089/System.Core.dll
----------------------------------------
System.Xml
    Assembly Version: 2.0.0.0
    Win32 Version: 2.0.50727.8773 (QFE.050727-8700)
    CodeBase: file:///C:/Windows/assembly/GAC_MSIL/System.Xml/2.0.0.0__b77a5c561934e089/System.Xml.dll
----------------------------------------
System.Configuration
    Assembly Version: 2.0.0.0
    Win32 Version: 2.0.50727.5483 (Win7SP1GDR.050727-5400)
    CodeBase: file:///C:/Windows/assembly/GAC_MSIL/System.Configuration/2.0.0.0__b03f5f7f11d50a3a/System.Configuration.dll
----------------------------------------
Accessibility
    Assembly Version: 2.0.0.0
    Win32 Version: 2.0.50727.5483 (Win7SP1GDR.050727-5400)
    CodeBase: file:///C:/Windows/assembly/GAC_MSIL/Accessibility/2.0.0.0__b03f5f7f11d50a3a/Accessibility.dll
----------------------------------------

BallscrewBob, 2018-05-17:
Sorry about the formatting GIT doesn't seem to like a regular copy paste.

BallscrewBob, 2018-05-18:
Also seeing this.  

[image]

arkypita, 2018-05-31:
Last screenshot was reported by some other user #303 but it is difficult to diagnose because i cannot reproduce in my environment...

arkypita, 2018-05-31:
Is this issue sporadic and random or it can be reproduced with some sequence/action?

BallscrewBob, 2018-05-31:
It is sporadic but does seem to happen on larger files more often so I think there is a link there somewhere.

StuartB4, 2018-05-31:
A bit off topic but I noticed BallscrewBob mentioned it happening with larger files.
What is the maximum image pixel size LaserGRBL can accept.

I just wondered because out of curiosity I tried to load an image with a size of 10,000 x 7500 pixels.
LaserGRBL starts the conversion process but then this pops up, but I did'nt understand what it meant.
Clicking Ok just stops the process.

[image]

arkypita, 2018-07-05:
> What is the maximum image pixel size LaserGRBL can accept.
@StuartB4 It is not fixed, it depends of architecture (x86 or x64) and available RAM.

> LaserGRBL can handle up to 22000*22000 (480Mpx) that mean an image of 2.2mt x 2.2mt (7x7feet) @ 10lines/mm of resolution.
> 
> I have tested it with 1mt x 1mt @ 10lines/mm (100Mpx) and it take about 10 minutes to generate the GCode on a 2.8GHz quad-core - using 5GB of RAM. It may be a little slow (but it is nothing compared to the execution time), however i have margin for improvement working on further optimizations.

https://github.com/arkypita/LaserGRBL/issues/139

arkypita, 2018-07-05:
@BallscrewBob can you tell me something about your PC?
Processor, OS, RAM?

BallscrewBob, 2018-07-05:
Quad core AMD bulldozer, 8 gig ram, all settings trimmed for max power usage but not OC'd, 4gb Nvidia card (not top of the line but respectable) It is also my main test box for the Arduino CREATE team too so it is cleaned up very regular, Security is AVAST (free) backed by MBAM (free). ...Nothing too fancy but a solid reliable workhorse.

BallscrewBob, 2018-07-05:
This was not a large job by any means.
I did get an option box pop up with the details below and there was an option to continue.
It seems to have worked Attaching g-code also.

[dog 01 cnc.zip](https://github.com/arkypita/LaserGRBL/files/2167119/dog.01.cnc.zip)

See the end of this message for details on invoking 
just-in-time (JIT) debugging instead of this dialog box.

************** Exception Text **************
System.NullReferenceException: Object reference not set to an instance of an object.
   at LaserGRBL.UserControls.CommandLog.OnPaint(PaintEventArgs e)
   at System.Windows.Forms.Control.PaintWithErrorHandling(PaintEventArgs e, Int16 layer, Boolean disposeEventArgs)
   at System.Windows.Forms.Control.WmPaint(Message& m)
   at System.Windows.Forms.Control.WndProc(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

************** Loaded Assemblies **************
mscorlib
    Assembly Version: 2.0.0.0
    Win32 Version: 2.0.50727.8784 (QFE.050727-8700)
    CodeBase: file:///C:/Windows/Microsoft.NET/Framework64/v2.0.50727/mscorlib.dll
----------------------------------------
LaserGRBL
    Assembly Version: 2.8.50.0
    Win32 Version: 2.8.50.0
    CodeBase: file:///C:/Program%20Files%20(x86)/LaserGRBL/LaserGRBL.exe
----------------------------------------
System.Drawing
    Assembly Version: 2.0.0.0
    Win32 Version: 2.0.50727.5495 (Win7SP1GDR.050727-5400)
    CodeBase: file:///C:/Windows/assembly/GAC_MSIL/System.Drawing/2.0.0.0__b03f5f7f11d50a3a/System.Drawing.dll
----------------------------------------
System
    Assembly Version: 2.0.0.0
    Win32 Version: 2.0.50727.8770 (QFE.050727-8700)
    CodeBase: file:///C:/Windows/assembly/GAC_MSIL/System/2.0.0.0__b77a5c561934e089/System.dll
----------------------------------------
System.Windows.Forms
    Assembly Version: 2.0.0.0
    Win32 Version: 2.0.50727.5491 (Win7SP1GDR.050727-5400)
    CodeBase: file:///C:/Windows/assembly/GAC_MSIL/System.Windows.Forms/2.0.0.0__b77a5c561934e089/System.Windows.Forms.dll
----------------------------------------
System.Core
    Assembly Version: 3.5.0.0
    Win32 Version: 3.5.30729.5420 built by: Win7SP1
    CodeBase: file:///C:/Windows/assembly/GAC_MSIL/System.Core/3.5.0.0__b77a5c561934e089/System.Core.dll
----------------------------------------
System.Xml
    Assembly Version: 2.0.0.0
    Win32 Version: 2.0.50727.8773 (QFE.050727-8700)
    CodeBase: file:///C:/Windows/assembly/GAC_MSIL/System.Xml/2.0.0.0__b77a5c561934e089/System.Xml.dll
----------------------------------------
System.Configuration
    Assembly Version: 2.0.0.0
    Win32 Version: 2.0.50727.5483 (Win7SP1GDR.050727-5400)
    CodeBase: file:///C:/Windows/assembly/GAC_MSIL/System.Configuration/2.0.0.0__b03f5f7f11d50a3a/System.Configuration.dll
----------------------------------------

************** JIT Debugging **************
To enable just-in-time (JIT) debugging, the .config file for this
application or computer (machine.config) must have the
jitDebugging value set in the system.windows.forms section.
The application must also be compiled with debugging
enabled.

For example:

<configuration>
    <system.windows.forms jitDebugging="true" />
</configuration>

When JIT debugging is enabled, any unhandled exception
will be sent to the JIT debugger registered on the computer
rather than be handled by this dialog box.

arkypita, 2018-07-05:
I could run your gcode and I did not get the error. This means that the problem is sporadic and can happen due to a combination of PC-dependent factors.

At least I have time I whould like to write a debugging version to send to you

BallscrewBob, 2018-07-05:
Not  problem...plenty of time here too for a while ;)

closed 2018-10-13 by arkypita