arkypita/LaserGRBL #356: Adding Slovenian language

Focus3, 2018-05-13:
Good day,

wanted to ask what's the procedure for translation to another language and adding as an option (u der the Language tab). 

I would love to translate from English to Slivenian, just not sure how. Also not sure if it's okay with the creator.

Focus3.

Arhimed-Ru, 2018-05-13:
https://github.com/arkypita/LaserGRBL/issues/28

Focus3, 2018-05-14:
Arhimed-Ru: Thank you for replying. Will try to translate as soon.

Focus3.

Focus3, 2018-05-15:
[Added Slovenian files.zip](https://github.com/arkypita/LaserGRBL/files/2002990/Added.Slovenian.files.zip)

Focus3, 2018-05-15:
I have not yet had the change to work in Radialix. Seems like a powerfull programe.
I got alot of help from the post that you refered - #28  Did everything the way it was described. Now will see what ArkyPita thinks about it.

Arhimed-Ru, 2018-05-16:
You can wait
:)

arkypita, 2018-05-18:
when I have time I will add your submitted translation

Focus3, 2018-06-22:
Good day,

just wanted to ask how things are with the Language support? Need any additional help/info?

Focus3.

JurajJurajFN, 2018-11-18:
I want to translate to my language, but I see that it is going nowhere, because from Julay, there is still not Slovenian translation. So if I translate it to my languege, you will not put it in software. Or am I wrong? Are you (arkypita) interest to translate LaserGRBL to other languages? I think if you want to support a lot of languages, you can make it easier to translate. Make some txt translations, that anybody can change after instalation. You will get support in many languages, and we will get our languages instantly. (Focus3 spent a lot of time with translation, and he get nothing until now)

arkypita, 2018-11-19:
>Make some txt translations, that anybody can change after instalation.

Unfortunately the .NET framework (with which it is written LaserGRBL) does not provide the possibility to translate with an external file. The only way to translate with .net is by compiling resource files that are embedded within the executable.

This implies that it is necessary to work on my side to include the translations made by the users, and their updates, especially if they send them as files instead of using pull request that is the standard method provided by github for code contribution.

Furthermore, as I wrote previously:

> I have plans to do a big code and UI refactoring, so i am waiting to add new translation because they will be obsolete early.

Unfortunately I have not been able to do this process yet, and I do not think I will do it anymore, because as you may have noticed I preferred to focus on bugfixes and adding new features. I will focus on the inclusion of new translations when the development will be at a sufficiently stable stage, so that it does not result in a partial job.