arkypita/LaserGRBL #357: How to increase laser power in lasergrbl

brucect, 2018-05-14:
The machine is running and the laser is on but it is not burning, it runs the program fine but no engraving
please help.
Laser is a 500mW blue\purple laser

StuartB4, 2018-05-15:
Is the laser focussed correctly, to the smallest dot size.
Are you using PWM to control the laser power. if so, is the PWM value set High enough.
0=off/lowest power, 255= fully on or highest power.

brucect, 2018-05-15:
To check these where do I go to find this

brucect, 2018-05-15:
focused
yes
paper
How can I check the power of the laser to make sure it is running at the proper setting??
when I have it only connected to 12v it burns perfectly.

brucect, 2018-05-15:
Laser is focused
Yes using PWM
Power? I assume it is very low because it doesn't burn. Don't know how to
change it.

StuartB4, 2018-05-16:
If you are using LaserGRBL to generate the GCode and are using PWM the laser power is generated with it using the letter S at the end of the line of code, like this for burning a greyscale image.
The numbers after the S are the laser power.
X10.5 S177
X10.571 S181
X10.643 S176
X10.714 S190
X10.786 S185
X10.857 S177

But you may need to change the settings when you prepare your image just before you press the Create button to generate GCode.

[image]

brucect, 2018-05-16:
So if I put in 170 in the SMax it will be about half of full power?

brucect, 2018-05-16:
Ok, I'm confused. I have been putting 1000, 3000, 5000  in the S-Max and my
laser seems to run hotter at 5000 than at 1000.
The other thing is how do I get the machine to run faster, it seems that
when I load a picture and I crop it to just the face it says at the bottom
6 hours, it seems my X axis is dithering as it moves but the line is not
broken. I am using off white card stock, the beam is a much of a pin point
as it can be adjusted. My face of the laser is 3 inches away from my table.
My laser is a Blue/Purple laser 500mW with TTL control Board. with the
laser not connected to anything just the 12 v power supply thru the board
it will burn thru standard card board box in a few seconds.
Thank You for helping me understand this program. I have inkscape but I
really don't understand how to use it.

brucect, 2018-05-16:
when I put in 1500 in the S-Max I'm getting readings s1223

        s1276

        s1294

        s1323
as the laser burns.

StuartB4, 2018-05-16:
S128 is half power. 

In the image I posted with the red circle it says "Speed" at the top and Engraving speed.
If you increase the number it should go faster but it depends on the settings you have in GRBL, assuming you are using GRBL. 
If so goto Grbl on the top menu in LaserGRBL.
[image]

Look at the Grbl Configuration.
You should get this image:
[image]

The first 2 highlighted items refer to Spindle Speed, but when using a laser it means laser power and should be set to Max=255 Min=0. To change the values just click in the Value column and change the value and move to the next one.

The next one down is Laser Mode.
If you are using GRBL 1.1 then you should make sure Laser Mode Enable is set to 1

Then you may need to change the X Axis Maximum rate and the Y Axis Maximum rate.
This is in millimeters per minute.
Move to the last 2 highlighted items and change the values to what you want up to the limit of your machine. If you use too high a setting you will have problems with the machine or with the image quality as it could miss steps travelling too fast. 
If you enter 1000 in the grbl settings and then try to set a higher value in Lasergrbl's Speed setting and press enter it will show a red box around the number you enter. Showing you that the number you entered is higher than the grbl setting.

When you have changed the settings to how you want them press the Write button at the bottom and that will write the values in to grbl and store them until you change them again.

Focus3, 2018-05-16:
Hi,

This might sound odd. But, you wrote that the distance between the Laser and work surface is 3 inches?

I had issues with some materials, where the laser power just wasn't strong enough (Set to power S=255. Focused to a tiny dot-perfectly).

And after some testing i found that my distance (Laser to workboard) was too long. It was 4 inches. By reducing the distance i was able to Engrave perfectly. Again, that was by some materials.

The thing is that Laser power is reduced with distance and it doesn't hurt to keep that in mind.

Focus3.

StuartB4, 2018-05-16:
There are a lot of changes to 1.1. 
It might be a good idea to update but make a note of your settings as they are now so you don't forget anything. You can alwasy tweak them later.

Here are the changes/additions to 1.1.
> 
1: Real-time Overrides : Alters the machine running state immediately with feed, rapid, spindle speed, spindle stop, and coolant toggle controls. This awesome new feature is common only on industrial machines, often used to optimize speeds and feeds while a job is running. Most hobby CNCs try to mimic this behavior, but usually have large amounts of lag. Grbl executes overrides in realtime and within tens of milliseconds.

2: Jogging Mode : The new jogging commands are independent of the G-code parser, so that the parser state doesn't get altered and cause a potential crash if not restored properly. Documentation is included on how this works and how it can be used to control your machine via a joystick or rotary dial with a low-latency, satisfying response.

3: Laser Mode : The new "laser" mode will cause Grbl to move continuously through consecutive G1, G2, and G3 commands with spindle speed changes. When "laser" mode is disabled, Grbl will instead come to a stop to ensure a spindle comes up to speed properly. Spindle speed overrides also work with laser mode so you can tweak the laser power, if you need to during the job. Switch between "laser" mode and "normal" mode via a $ setting.

4: Dynamic Laser Power Scaling with Speed : If your machine has low accelerations, this option will automagically scale the laser power based on how fast Grbl is traveling, so you won't have burnt corners when your CNC has to make a turn! Enabled by the M4 spindle CCW command when laser mode is enabled.

5: Sleep Mode : Grbl may now be put to "sleep" via a $SLP command. This will disable everything, including the stepper drivers. Nice to have when you are leaving your machine unattended and want to power down everything automatically. Only a reset exits the sleep state.

6: Significant Interface Improvements: Tweaked to increase overall performance, include lots more real-time data, and to simplify maintaining and writing GUIs. Based on direct feedback from multiple GUI developers and bench performance testing. NOTE: GUIs need to specifically update their code to be compatible with v1.1 and later.

7: New Status Reports: To account for the additional override data, status reports have been tweaked to cram more data into it, while still being smaller than before. Documentation is included, outlining how it has been changed. Improved Error/Alarm Feedback : All Grbl error and alarm messages have been changed to providing a code. Each code is associated with a specific problem, so users will know exactly what is wrong without having to guess. Documentation and an easy to parse CSV is included in the repo.

8: Extended-ASCII realtime commands : All overrides and future real-time commands are defined in the extended-ASCII character space. Unfortunately not easily type-able on a keyboard, but helps prevent accidental commands from a G-code file having these characters and gives lots of space for future expansion.

9: Message Prefixes : Every message type from Grbl has a unique prefix to help GUIs immediately determine what the message is and parse it accordingly without having to know context. The prior interface had several instances of GUIs having to figure out the meaning of a message, which made everything more complicated than it needed to be. New OEM specific features, such as safety door parking, single configuration file build option, EEPROM restrictions and restoring controls, and storing product data information.

10: New safety door parking motion as a compile-option: Grbl will retract, disable the spindle/coolant, and park near Z max. When resumed, it will perform these tasks in reverse order and continue the program. Highly configurable, even to add more than one parking motion. See config.h for details.

11: New '$' Grbl settings for max and min spindle rpm. Allows for tweaking the PWM output to more closely match true spindle rpm. When max rpm is set to zero or less than min rpm, the PWM pin D11 will act like a simple enable on/off output.

12: Updated G28 and G30 behavior from NIST to LinuxCNC G-code description. In short, if an intermediate motion is specified, only the axes specified will move to the stored coordinates, not all axes as before.

Lots of minor bug fixes and refactoring to make the code more efficient and flexible.
NOTE: Arduino Mega2560 support has been moved to an active, official Grbl-Mega project. All new developments here and there will be synced when it makes sense to.

brucect, 2018-05-16:
I haven't changed anything so I can just load this in my regular grbl file
and re-compile it??

brucect, 2018-05-16:
StuartB4, This is what is in my Grbl file. I have no $30 $31 etc.

$0=10 (step pulse, usec)
$1=25 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=0 (dir port invert mask:00000000)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=3 (status report mask:00000011)
$11=0.010 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)
$13=0 (report inches, bool)
$20=0 (soft limits, bool)
$21=0 (hard limits, bool)
$22=1 (homing cycle, bool)
$23=0 (homing dir invert mask:00000000)
$24=25.000 (homing feed, mm/min)
$25=250.000 (homing seek, mm/min)
$26=250 (homing debounce, msec)
$27=1.000 (homing pull-off, mm)
$100=250.000 (x, step/mm)
$101=250.000 (y, step/mm)
$102=250.000 (z, step/mm)
$110=1000.000 (x max rate, mm/min)
$111=1000.000 (y max rate, mm/min)
$112=0.000 (z max rate, mm/min)
$120=10.000 (x accel, mm/sec^2)
$121=10.000 (y accel, mm/sec^2)
$122=10.000 (z accel, mm/sec^2)
$130=200.000 (x max travel, mm)
$131=200.000 (y max travel, mm)
$132=200.000 (z max travel, mm)
ok

brucect, 2018-05-17:
> Tried to load and compile Grbl v1.1 and I get this:+
> C:\Users\bruce_000\Documents\ARDUINO 1.8.5\arduino-1.8.5\libraries\
> grbl\examples\grblUpload\grblUpload.ino:27:18: fatal error: grbl.h: No
> such file or directory
>
>  #include <grbl.h>
>
>                   ^
>
> compilation terminated.
>
> exit status 1
> Error compiling for board Arduino/Genuino Uno.
>
> On Wed, May 16, 2018 at 5:55 PM, Bruce Kramer <brucekramer07@gmail.com>

StuartB4, Well, I uploaded Grbl 1.1 and it is giving me issues as I sent
the error to you, now my machine will not home out properly.

here is the config file for Grbl on homing

// NOTE: Defaults are set for a traditional 3-axis CNC machine. //Z-axis
first to clear, followed by X & Y.
//#define HOMING_CYCLE_0 (1<<Z_AXIS) // REQUIRED: First move Z to clear
workspace.
#define HOMING_CYCLE_1 ((1<<X_AXIS)|(1<<Y_AXIS))  // OPTIONAL: Then move
X,Y at the same time.
// #define HOMING_CYCLE_2                         // OPTIONAL: Uncomment
and add axes mask to enable

// NOTE: The following are two examples to setup homing for 2-axis machines.
 #define HOMING_CYCLE_0 ((1<<X_AXIS)|(1<<Y_AXIS))  // NOT COMPATIBLE WITH
COREXY: Homes both X-Y in one cycle.

// #define HOMING_CYCLE_0 (1<<X_AXIS)  // COREXY COMPATIBLE: First home X
// #define HOMING_CYCLE_1 (1<<Y_AXIS)  // COREXY COMPATIBLE: Then home Y

// Number of homing cycles performed after when the machine initially jogs
to limit switches.
// This help in preventing overshoot and should improve repeatability. This
value should be one or
// greater.

StuartB4, 2018-05-17:
I've never used homing on my machine so I dont really know much about it.
This may help.
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration

arkypita, 2018-05-18:
@brucect if you need two axis homing you should comment out [line 105 and 106](https://github.com/gnea/grbl/blob/5c8dcefcbdc72e953fb07c5f8bb8ef4925ac6b5b/grbl/config.h#L105) and decomment [line 110](https://github.com/gnea/grbl/blob/5c8dcefcbdc72e953fb07c5f8bb8ef4925ac6b5b/grbl/config.h#L110)

Your code will be:

```
// NOTE: Defaults are set for a traditional 3-axis CNC machine. Z-axis first to clear, followed by X & Y.
//#define HOMING_CYCLE_0 (1<<Z_AXIS)                // REQUIRED: First move Z to clear workspace.
//#define HOMING_CYCLE_1 ((1<<X_AXIS)|(1<<Y_AXIS))  // OPTIONAL: Then move X,Y at the same time.
// #define HOMING_CYCLE_2                         // OPTIONAL: Uncomment and add axes mask to enable

// NOTE: The following are two examples to setup homing for 2-axis machines.
#define HOMING_CYCLE_0 ((1<<X_AXIS)|(1<<Y_AXIS))  // NOT COMPATIBLE WITH COREXY: Homes both X-Y in one cycle. 

// #define HOMING_CYCLE_0 (1<<X_AXIS)  // COREXY COMPATIBLE: First home X
// #define HOMING_CYCLE_1 (1<<Y_AXIS)  // COREXY COMPATIBLE: Then home Y
```

brucect, 2018-05-18:
Is there any way to check the max allowable number setting for the laser
and also the max for the allowable speed of X and Y am axis??
Thank you. Again

arkypita, 2018-05-19:
$30 and $31 are absolutly arbitrary. Arduino use 8bit pwm modulation to set laser power or spindle speed so the value of $31/$31 will be scaled to 0-255. I suggest to use 0-255 as values.

Max speed of X and Y depend on specific hardware, stepper, driver and microstepping configuration, the can be measured experimentally setting an high limit speed (like 10'000) and use jogging to move with different speeds. When the motor hang or vibrate you have reached the max speed of your hardware

brucect, 2018-05-19:
Thank you thank you, you have been a BIG help, I am learning much.

brucect, 2018-05-20:
I have a question on the Arduino download.
I have uploaded Arduino-1.8.5-windows to One Drive.
Documents/arduino-1.8.5-windows/arduino-1.8.5/libraries (paste
grbl)/grbl/examples/grblupload/grblupload
Now this opens the sketch box, click on the right arrow to upload. I get
the following error:
C:\users\bruce_000\documents\arduion-1.8.5-windows\arduino
1.8.5\libraries\grbl\examples\grblupload\grblupload.ino:27:18: fatal error:
grbl.h: no such file or directory
#include <grbl.h>
compilation terminated.
exit status 1
Error compiling for board Arduino/Genuino Uno

What am I doing wrong here?
Thanks
Bruce

closed 2018-07-05 by arkypita

david19830108, 2018-09-30:
Hi!

I live in Hungary! I would like to ask you some help!

I bought a second machine:

https://www.ebay.com/itm/112880197588?ul_noapp=true
7w laser, 30 * 40cm working area
I got a MNlaser program

Can I change my machine to be compatible with grbl controller? What should I do with it? the base frame, stepping motors and cables are grbl compatible on my machine?

I'm sorry, but I'm really new, I just want to use grbl software and it does not work with my grbl program now!

I found a grbl control panel:
https://www.ebay.com/itm/372436007572?ul_noapp=true

Would this be right for my machine?
Help me please what to do! :)

Thanks for the answer!

Psychokiller1888, 2019-01-18:
I came on here after fighting the day over this one... Kept putting 255 in my laser max setting... But the $30 setting says: **RPM**. Who has a spindle that spins only at 255 rpm? In my case, setting the laser max to 10'000 finally did it! It should be better explained and understood of course, that the settnig reflects $30 and 31. Cheers!

LordPueblo, 2019-08-28:
Hi People!
I'm from Russia)
I'd adding grbl library in Arduino and getting error when uploading to machine (compiling is ok)

The text of error is:

Arduino: 1.8.7 (Windows 7), Плата:"Arduino/Genuino Uno"

In file included from C:\Users\Pueblo\Documents\Arduino\libraries\grbl/grbl.h:43:0,

                 from C:\Users\Pueblo\Documents\Arduino\libraries\grbl\examples\grblUpload\grblUpload.ino:27:

C:\Users\Pueblo\Documents\Arduino\libraries\grbl/nuts_bolts.h:51:0: warning: "max" redefined

 #define max(a,b) (((a) > (b)) ? (a) : (b))

 ^

In file included from sketch\grblUpload.ino.cpp:1:0:

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:93:0: note: this is the location of the previous definition

 #define max(a,b) ((a)>(b)?(a):(b))

 ^

In file included from C:\Users\Pueblo\Documents\Arduino\libraries\grbl/grbl.h:43:0,

                 from C:\Users\Pueblo\Documents\Arduino\libraries\grbl\examples\grblUpload\grblUpload.ino:27:

C:\Users\Pueblo\Documents\Arduino\libraries\grbl/nuts_bolts.h:52:0: warning: "min" redefined

 #define min(a,b) (((a) < (b)) ? (a) : (b))

 ^

In file included from sketch\grblUpload.ino.cpp:1:0:

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:92:0: note: this is the location of the previous definition

 #define min(a,b) ((a)<(b)?(a):(b))

 ^

In file included from C:\Users\Pueblo\Documents\Arduino\libraries\grbl/grbl.h:43:0,

                 from C:\Users\Pueblo\Documents\Arduino\libraries\grbl\examples\grblUpload\grblUpload.ino:27:

C:\Users\Pueblo\Documents\Arduino\libraries\grbl/nuts_bolts.h:55:0: warning: "bit" redefined

 #define bit(n) (1 << n) 

 ^

In file included from sketch\grblUpload.ino.cpp:1:0:

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:123:0: note: this is the location of the previous definition

 #define bit(b) (1UL << (b))

 ^

Скетч использует 29868 байт (92%) памяти устройства. Всего доступно 32256 байт.
Глобальные переменные используют 1484 байт (72%) динамической памяти, оставляя 564 байт для локальных переменных. Максимум: 2048 байт.
avrdude: ser_open(): can't open device "\\.\COM4": Не удается найти указанный файл.

Проблема загрузки в плату. Помощь по загрузке: http://www.arduino.cc/en/Guide/Troubleshooting#upload .

Этот отчёт будет иметь больше информации с
включенной опцией Файл -> Настройки ->
"Показать подробный вывод во время компиляции"