arkypita/LaserGRBL #360: TTL LASER with Arduino UNO and how to add $30=1; $31; $32 to Grbl Laser

ihab2010, 2018-05-21:
Hello ,
I have my X-Axis motor Working
I have my Y-Axis motor working
I have a laser
https://www.ebay.com/itm/EleksMaker-445nm-7000mW-Blue-Laser-Module-TTL-PWM-For-Laser-Engraver-Machine-/112710815996

 which has power adapter and driver board  like photo
[image]
It also has TTL + and TTL - and PWM
It does not have any pdf document.
I tried to read about connecting it to Arduino UNO Pin 11 or Pin 12 and
Now need to add lines about spindle  in the  G code lines 
When I try to add it gives me Red alarm (like No entry sign)
how to add $30=1; $31; $32 to Grbl Laser 
How to make my laser up and running?
Can you help

StuartB4, 2018-05-21:
What version of GRBL do you have.
Version 1.1 includes $30,$31,$32.
Version 0.9 does'nt.

TTL+ goes to Pin 11 and TTL - goes to GND on the Arduino.

If you start LaserGRBL and click Connect and then click the Grbl menu (top left), then select Grbl Configuration you should see a list of GRBL settings like this if you have version 1.1:

[image]
Obviously your settings wont be the same as the ones in the picture, but you can then edit the settings to suit your machine. 
$30=255 (Maximum Spindle Speed / Laser Power)
$31=0     (Mimimum Spindle Speed / Laser Power
$32=1     (Laser Mode Enabled)

To edit the list just move the mouse to the Value column, select the one you want and left click it twice but not quick, a normal double click seems to be too fast to register and enable editing.
Enter the values you want in the first one and press enter and it will jump down to the next one.
When you have finished the settings click the Write button at the bottom. The settings will then be stored in GRBL on the Arduino until you change them again.

ihab2010, 2018-05-25:
Hello,
Thank you for your prompt reply.
I changed to Grbl 1.1
The commands $30 to $32 are in the table now.
But, still, laser, is running (I mean fan and weak light laser)
I can not burn anything. even when sending my gcode fro any photo.
Would you please let me know how can connect my power to laser while getting my pwm or TTL connecting , there are 2 sockets one with 2 pins TTL + and TTL -  and the other has three pins PWM and + amd - , I put my power in this  + and - , what  can I do with the rest (PWM , + , -) ?
Is there any testing way to make sure that the laser cn burn and not giving weak light only?
Thanks

StuartB4, 2018-05-25:
Hi. 
You should just be able to use the TTL + from the laser driver board connected to pin 11 of the Arduino and the TTL - from the laser driver board connected to the GND of the Arduino. 

I had a look at the Ebay link for the laser you have and it says the laser does not have adjustable focus.
"Light form: Unadjustable, Point-shape and focus"

That means you will have to set the height of the laser from the workpiece exactly the right distance so it focuses to the smallest point.

Do you have PWM button enabled in LaserGRBL settings?
Like this.
[image]
[image]

ihab2010, 2018-05-25:
Thank You StuartB4 for your prompt reply.I need to test the high power laser to make sure it is burning without connection to Grbl.I tried to contact the supplier , they said it will not work with Arduion, I believe they want to push me to buy their Mana SE card.Do you think if I connect +5 Volt from battery to +TTL and - TTL would give me one burning point , without damaging the laser?Or
Do I need to connect PWM on the laser driver board to TTL + as well?Please advise.Thanks again.
Best regards,Dr . Eng. Ihab El-Sayed

StuartB4, 2018-05-25:
Hi.

I can't be 100% sure, but if It was mine and I was doing it I would connect it like this.
**I dont want you to damage the laser** but it seems the most likely way.

The driver board needs 12 Volts maximum @ 2 Amps or more, more is fine up to as many as you like, the board will only use what current it needs to run. 

[image]

ihab2010, 2018-05-27:
Hello StuartB4,Thank you for your email. I applied the above circuit. But I did not any Led lighting-up on the laser driver board.What could be the reason?The weak blue laser is on, the fan of the laser is on, but the led on the laser driver board does not light up.There is one PWM pin on the driver board, it is not connected to any pin.

Best regards,Dr . Eng. Ihab El-Sayed 
Associate Professor College of EngineeringIndustrial Engineering DepartmentTel. +96648460008  Ext. 8208Fax. +96648475837 www.drihab.com Email: isayed@taibahu.edu.sa Email : dr_ihab2010@yahoo.com SKYPE: Dr. Ihab AachenMobile: +966555781096  Saudi Arabia 
Mobile : +201001082038   Egypt

StuartB4, 2018-05-27:
Hi.
You could also try connecting the  unused PWM pin to pin 11 on the Arduino instead of the TTL+.

[image]

If they are connected then you can still use the TTL- for the GND on the Arduino.

It may be that they were telling the truth about that laser not working on an Arduino. Though I 
find that hard to believe as they use an Arduino Nano on the MANA SE board anyway.

ihab2010, 2018-06-01:
> Hi ihab2010.
>
> Just wondered if you actually got the laser to work properly.

Hi Stuart,Thank you for your email and concern.No not yet.It is working only (Fan +weak mode).up till now I did not see the driver board with the two small led lighting up , that is why , I believe is something wrong in connection.
Best regards,Dr . Eng. Ihab El-Sayed

StuartB4, 2018-06-01:
Sorry to hear that.
Seems a bit strange. May be it's something to do with the chip on the board that makes it incompatible with the Arduino.

I was just looking at other driver boards with similar connections.
It would be possible to swap the driver board for a different type, but that means spending more money, and it looks like you've spent quite a bit already.

I was looking at this one.
https://www.ebay.co.uk/itm/5A-analog-linear-PWM-laser-diode-driver-with-thermal-protection/152549325331

But you would have to desolder the laser LD+ and LD- from yours and connect it to this, and find a way to fit the fan because this one does not have a connection for it. You would also have to add a heatsink to the Mosfet Transistor on the board to dissipate the heat.
But it looks like this one does work with the Arduino. 

[image]

closed 2018-07-05 by arkypita

OldShark, 2018-11-27:
Hi all. Have the same problem here. Have a LD 2.5w with controller board 3-pin only. I trying use with Arduino UNO and GRBL, Benbox. Nothing.

OldShark, 2018-11-28:
Not connecting the pin 11 to the laser signal

[image]

Connected the pin 11 to the laser signal

[image]

Repair the image of the ocisloscope, which when connecting the arduino pin 11 to the laser signal, voltage drops about half. Because????? I duno.
I tested with arduino uno CNC Shield V3 and with arduino nano cnc shield V4

argee2022, 2022-03-02:
I'm not sure if the job done. 

I see this after years. And here is my understanding

Two connectors are both Key controller:
1), 3 pin connector is power control you must give 12V/GND/PWM( about Frq10K~30KHz & duty cycle about 10~40% ). PWM use a 5V clock should be OK. On Uno we can use D11 and with the command "M3S500", we should get a big power configured. 

2), 2 pin should also be gate control as a switch or relay to turn laser on. if 3Pin connect right connected, this one should connect to GND & D13 of GRBL-1.1f. This is the spindle DIR and can also be the enable to the laser. M3.M4 change the D13 level from 0V to 5V.

And that means after Uno bootup, write

S500 //setup PWM power
M3 //disable or enable laser, depend on the hardware of laser.
M4 //disable or enable laser 

FYI