arkypita/LaserGRBL #361: X Axis Jitter

Wetsparks, 2018-05-21:
Help, I'm going mad with this one.
It never used to happen, but just lately I keep getting jitter on the X axis. It's as though the belt is slipping on the pully, but it's not, it's comming from the motor. When it happens it can throw the axis out by as much as 20mm.
I have changed the POLO driver and checked the voltage setting. I have also tried many combinations in the settings to see if that made a differance but without any luck.
I'm using the latest update of GRBL v2.8.45.
Any help would be most welcome.

Philip.

StuartB4, 2018-05-22:
What type of board are your motor drivers plugged in to?
It might be worth checking the driver socket solder joints and the motor connector joints.

StuartB4, 2018-05-22:
If it's a protoneer Arduino shield then you should also check the 10 uF capacitor that sits laying down in between the motor driver connectors.

Wetsparks, 2018-05-23:
Hi Stuart,
Yes it's a Protoneer board.
Did as you suggested and changed the 100uF on the X driver section. Swapped it out with the unused A driver section. Works a treat now with no jitters.
Many thanks,

Philip.

closed 2018-05-23 by Wetsparks

StuartB4, 2018-05-23:
Hi Philip.
Thats good to know, glad I could help.
Good luck with it.