arkypita/LaserGRBL #364: Problem when aborting, machine goes to oposite corner

sinfocomp, 2018-05-25:
when i hit the stop button the machine goes all the way to the oposite home corner, and tottaly mess the position, it would be better to stop at the last commanded position so we can continue or move from there.

StuartB4, 2018-05-25:
You can use the hand button in the bottom right corner of the screen to do that, and the button with the man on to continue from where it stopped.

arkypita, 2018-05-28:
The squared red button is not a "pause job" button but it is a "stop and abort current job".
If you want to pause job, simply use hand button as suggested by @StuartB4

closed 2018-05-28 by arkypita

sinfocomp, 2018-05-28:
i know i can pause but the problem is that after abort i need to homme and set offsets again, that is the problem of aborting with the red button, i might as well just hit reset and loose position anyway.
the thing is i need to abort without loosing position so i can move the machine fix whatever problem i got then continue on the line i need to continue, so i need to either fix the abort option or add a start from line xx comand, please help