arkypita/LaserGRBL #366: lasergrbl goes on then off

brucect, 2018-05-27:
when I connect lasergrbl and load a picture, set up to engrave, the program stops and disconnects. Why does it do that??

arkypita, 2018-05-28:
not normal and not reported by any other user
post your log here (menu ? - open session log - copy and paste)

Tha14, 2018-05-28:
I've also had some issues with it in the past, it crashes without any logs. Any debugging mode we can use to find the issue? Mine seems to be caused from loose connections(which I am currently fixing).

brucect, 2018-05-28:
For some reason unknown to me, when I restarted LaserGrbl it cleared any
issues. I did have to change a couple items, line $10=3
$23=1
$110=635.000
#111=635.000
#112=635.000
Not sure if that fixed it or what. I went in to check if anything changed,
and found since the last time I ran it. And the 5 items had changed so I
went back to my latest print out and sure enough, they were different.
I noticed my laser isn't burning again. I have Grbl Configuration spindle
speed ($30)at 255 and ($31) at 0 $32 is at 1. When I pause the run, s shows
in the 20's nothing higher than 28.
Thanks

brakthehun, 2018-06-01:
The Arduino has a maximum pulse rate of 30khz.
What is your step resolution?? ($100 / $101)
Your laser might not trigger until it gets enough PWM (S).

closed 2018-07-05 by arkypita