arkypita/LaserGRBL #368: How to connect laser module to cnc sheild v3.0

sankajalakshan, 2018-05-29:
I want to add 2.5w ttl/pwm laser module to my corexy plotter.

https://goo.gl/images/2awCuV

1. What is the best method TTL or PWM?
2. What is the Wiring digram off TTL and PWM? 
3. How to supply power?

Please help...

arkypita, 2018-05-29:
1. TTL and PWM means the same (different way to call the "on-off" tecnology
2. See grbl website, not LaserGRBL one
3. Same as 2

StuartB4, 2018-05-29:
If it a Protoneer Shield look here:
https://forum.protoneer.co.nz/index.php

sankajalakshan, 2018-05-29:
Im using v3.0 cnc sheild and grbl 1.1f (latest version) . Currently i made a 4xidraw corexy pen plotter and it's working well. So i need to improve it adding a laser module.

sankajalakshan, 2018-05-29:
Thanks StuartB4.

When we have the 3 pin module (VCC, GND, PWM), I have to connect VCC into external 12v 2+A power source (+) and GND into external 12v 2+A power source (-) and PWM into cnc sheild Z+.

Is that correct?

Do i need to connect the GND into (-) of the external power source and the GND of the arduino borad/cnc sheild?

sankajalakshan, 2018-05-30:
Thanks a lot

sankajalakshan, 2018-05-30:
I have some issues to clarify.
Please help.
If i want to engrave a wood can i use 500mw laser? If it is, when i using a 2500mw laser, that wood will engrave deeper than 500mw laser.

Is there any code or method to control the power of 2500mw laser. Like when i need to engrave something lightly and when i need to engrave somthing hardly can i adjust the power of laser?

StuartB4, 2018-05-30:
Yes you can burn wood with a 500mw, and yes the more powerful the laser the more it will burn.
If you are using LaserGRBL to burn then the settings for the laser power are in the GCode when it is generated.
So you can use a 2.5 watt laser but only burn lightly.
It's to do with the S setting and the speed your machine moves.
It can have a high speed (mm per minute) setting but if the S for the laser power value is high it will still burn more than if you have a high speed and low power S setting.

The settings can be changed when you import your image in to LaserGRBL.

[image]

sankajalakshan, 2018-05-30:
Thanks a lot.....

StuartB4, 2018-05-30:
These were all burned with LaserGRBL.
The ones with the overlaying squares and the chess board one were all done with the same laser power using a 300mw red laser but at different speeds between 80 and 200 mm per minute. 

The greyscale images were done with a 2 watt blue laser. The pencil is just so you get an idea of how small they are.
My machine is made from old DVD drives so it only has a maximum work size of 40 x 40 mm's

[image]

sankajalakshan, 2018-05-30:
Thanks a lot. If anything to add to learn, please be kind enough to update me.  And you are explaining this well. Thanks again.

closed 2018-07-05 by arkypita

sankajalakshan, 2018-07-09:
i made this and laser not turn on by the lasergrbl. however after connecting the power to the laser module when pushing the switch laser on but not burning.
[image]
[image]
[image]

StuartB4, 2018-07-09:
Do you have PWM mode enabled in LaserGRBL settings?

[image]

[image]

sankajalakshan, 2018-07-09:
yes it also enabled.

StuartB4, 2018-07-09:
Can you check the config.h file in the Arduino Libraries folder.
If you look at this picture you can see where it should be on your computer.
You may have a different user name to mine but it should be in the same place.

[image]

I edit this file with Notepad++ which is a free text editor and better than notepad as it gives you line numbers at the begining of the lines.

When you find config.h scroll down to line number 339 and make sure there are no //'s  in front of the # symbol.  If there are then you have to remove them, save the file and recompile GRBL and re-upload it to the Arduino.
[image]

sankajalakshan, 2018-07-09:
i checked, it is also not commented. i measure the voltage by multi meter on D11(Z+0) pin and GND, it shows the maximum voltage is 4.35v.

ingrc91, 2018-11-01:
good day to all.
I have a problem with the laser configuration, I will send you an image of your circuit
[image]
it stays on

StuartB4, 2018-11-01:
Which version of GRBL are you using?

ingrc91, 2018-11-01:
[image]

ingrc91, 2018-11-01:
the laser stays on does not turn off

ingrc91, 2018-11-01:
Can you help me please
I'm just starting in the world of laser cnc

StuartB4, 2018-11-01:
Check your config.h file in the Arduino Libraries.

[image]

I edit this file with Notepad++ which is a free text editor and better than notepad as it gives you line numbers at the begining of the lines.

When you find config.h scroll down to where it says #define VARIABLE SPINDLE and make sure there are no //'s  in front of the # symbol.  If there are then you have to remove them, save the file and recompile GRBL and re-upload it to the Arduino.
[image]

Also check in GRBL Configuration in LaserGRBL to make sure you have $31 and $32 set correctly.
$31=0
$32=1

arkypita, 2018-11-01:
First, check proper working of your driver.
Detach pwm cable from shield, probe it to 5v and gnd. Laser should turn on and off respectly.

ingrc91, 2018-11-01:
> Primero, verifique el correcto funcionamiento de su conductor. 
> Desconecte el cable pwm del blindaje, pruebe a 5v y gnd. El láser debe encenderse y apagarse respetuosamente.
I did the same thing but only turn on the fan and when it has just carried out the trajectory it keeps on the laser fan but not the laser.

ingrc91, 2018-11-01:
[image]

arkypita, 2018-11-01:
> did the same thing but only turn on the fan and when it has just carried out the trajectory it keeps on the laser fan but not the laser.

So: you confirm that pwm pin is not controlling laser on/off?

If yes, your is an hardware problem. Could you post a picture of your boars and wiring?

ingrc91, 2018-11-01:
[image]
![image](https://user-images.githubusercontent.com/44658063/47865127-6657df80-ddc9-11e8-92c6-5c8ef21b9459.pn
[image]

arkypita, 2018-11-01:
@fantilator true is not needed

arkypita, 2018-11-01:
@ingrc91 I see a white cable connected to Z (ttl out, shield side) but a red cable connected to ttl, driver side.

Maybe you have inverted ttl and vcc?

ingrc91, 2018-11-01:
el cable rojo le hice una extensión por eso en la conexión con el cnc shield sale en blanco

ingrc91, 2018-11-01:
Pueden ayudarme por favor mi láser hace trayectoria completa quema todo

arkypita, 2018-11-02:
I cannot exactly read what is written on PCB, but take a look at the 2pin connector on the left

[image]

Is this marked GND - TTL?

StuartB4, 2018-11-02:
Yes it is Arkypita. You can see it a bit better on the other picture with the shield.
May be it's a TTL out for something else. Seems strange having 2 TTL ports.

ingrc91, 2018-11-02:
I did the test and apparently made short because it does not want to recognize the computer my arduino, you can appreciate the 4 led that comes in the plate are on
[image]

StuartB4, 2018-11-02:
That does'nt sound good. Is your Arduino now broken?

Just a thought.
Have you got PWM mode enabled in LaserGRBL?

[image]

ingrc91, 2018-11-02:
apparently it burned the arduino because it no longer wants to know the com port, with respect to the image that you showed me if the PWM support was selected

gisep, 2018-11-02:
Hi.
I've got a similar laser driver.
I think the mistake is in the Power.
You have to put the power supply in the 3-pins socket: "VCC" and "ground" to the power supply and "PWM" in the Z pin.

gisep, 2018-11-02:
This is my scheme:
from power supply i've got + and - that go in series in the VCCs and GROUNDs of the shield and of the driver. In my confusing picture i have the power supply on the left: VCC is orange and GROUND is brown. The yellow cable (PWM) from the 3-pins that becomes purple, goes in the Z+
[image]

I hope to help you!

OldShark, 2018-11-21:
Hi all. I have the same problem here.

bsmith07, 2019-06-14:
So, the way this works is different depending on your power supply situation.  If you are using the same power supply for your Arduino and your laser, you're going to have problems, however using the z pin with the spindle enable (a jumper on the board depending on your shield version I believe) using separate power supplies should set you up. Make sure your grbl setting for $30=255 and laser mode is enabled.  I'm currently trying to figure out how to use pwm when they're on the same psu, but I'm hitting a wall.  The ttl drivers seem not to like the signal when the board shares a ground.  Go figure.

thunderfrog007, 2019-06-20:
Hi all,
i am new with laser and would ask for some help please.
I bought this 15W laser module on ebay
https://www.ebay.de/itm/15W-Laser-Head-15000mW-Blue-Laser-Module-for-CNC-Laser-Engraving-Machine-w-PWM-E/153488235356?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649

thunderfrog007, 2019-06-20:
Hi all,
i am new with laser and would ask for some help / clarification.
I bought this 15W laser at ebay 
https://www.ebay.de/itm/15W-Laser-Head-15000mW-Blue-Laser-Module-for-CNC-Laser-Engraving-Machine-w-PWM-E/153488235356?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649

and i want to control with cnc shield V3. 
Main question is about driver: do i need additional ACC driver, or is this already on board ?
Can i connect my power supply 12V/20A directly to the laser module, this way ?

[image]

Thanks in advance

StuartB4, 2019-06-20:
Yes you can, the current the laser uses is regulated by the ACC board, you could use a 12v - 100 Amp supply, the laser will only use what it needs. The yellow lead should go to a pin on the CNC Shield that supplies the PWM signal. 
That comes from the Arduino Pin 11. Connect the shield to the Arduino.
If you have a multimeter you can check the continuity between the Arduino Pin 11 and either the Z+ or Z-  or SpnEn pins on the shield. If the Z+ or Z- and Arduino pin 11 have continuity then you connect the yellow wire to either of them, they are both connected together.
If Arduino pin 11 and SpnEn pin on the shield have continuity then you connect the yellow wire to SpnEn pin.
You may also need to add a GND wire from the Laser power supply to the shield to make sure the PWM is properly connected.
You might not need it if you are using the same power supply to power the shield/stepper motors and the laser.

thunderfrog007, 2019-06-20:
Thanks Stuart !
So i was right, the green marked module at my laser is the ACC driver... this was my confusion, too many modules compared, and many have separate ACC modules.
GND is clear, i use same power supply for shield and laser, GND is connected together.
PWM is also clear, i will check once laser module arrives, hopefully this saturday :)
Thanks again

StuartB4, 2019-06-21:
I assume you know the only way to focus that laser is by moving the whole thing up and down above the workpiece. It has no adjustable focus ring on the lens.

thunderfrog007, 2019-06-21:
Yeah, sure.
I already mounted additional fan to my MPCNC to avoid smoke at the lens.

lexelby, 2019-07-04:
Folks that have their laser always on are in the same boat as me: their laser expects a LOW to turn on and a HIGH to turn off.  This is a common enough variant that the CamTool CNC V3.3 board that comes with some CNC kits has two laser ports: one for "normal" PWM and one that's inverted.

I think all that's necessary is to invert the Z+ signal.  I've read that that can be done with an NPN transistor.

lexelby, 2019-07-05:
I've successfully connected my laser by inverting the PWM signal!  I followed this to hook up the NPN transistor: https://github.com/gnea/grbl/issues/22#issuecomment-341931756

I connected the SW pin of my fan (white wire) to V0 in the above schematic.  I connected "+V" to the 5V pin on the cnc shield.  I connected the 12V pin (red) to the 12V power supply by hooking it into the upper right pin in the red header, Vmot on the [cnc shield v3 schematic](https://i0.wp.com/blog.protoneer.co.nz/wp-content/uploads/2013/07/Arduino-CNC-Shield-Scematics-V3.XX_.jpg).

Everything works perfectly.  The laser stays off initially, though the fan comes on as expected. I can enable laser mode and dial up the power and the laser brightens up just as it should.

I hooked all this up on a breadboard, but I'm planning to connect it all in the red headers.  Since I'm not using the "A" axis, there are several empty pin slots that I can put the resistors and transistor in, and 5V and ground are right there too.  I'll probably only have to bridge a couple solder pads on the bottom to make everything connect up as it needs to.

shivmahesh, 2019-07-22:
StuartB4 sir,pls guide me on how to connect Laser module MXD 1230( which has only two wire +ve red and-ve black.) to arduino uno and cnc Shield v3.I will be using laser GRBL for my cnc machine .thanks.

StuartB4, 2019-07-22:
Hi shivmahesh.
I've just been looking at the MXD1230 laser module but all the ones I saw are only 5mW.
Is yours the same. If it is it won't burn anything. You need something more powerful, like 200mW plus.

OhmsLawBreaker, 2019-07-24:
Hey all,

First off im running and arduino UNO with a version 3.00 CNC shield on GRBL v1.1. Im not sure of the model number or make of the lazer im using, theres nothing on it but its a 2.5 w blue lazer thats 3 wires (See attached photo), it came with the chineese woodpecker CNC 3018 kit.

I'm having the same problem. My laser stays on when connected to Z+ pin. I do have PWN power enabled and $32=1 laser mode enabled. As I was making the attached drawing I noticed that my laser is by default 0=ON and 1=OFF.

Anyone have any clue as to what i'm doing wrong?

[image]

lexelby, 2019-07-24:
You're in exactly the same situation I was, @OhmsLawBreaker.  Did your control board catch fire?  Mine did because the purple DRV8825s weren't properly tuned by the kit maker.

You need to invert the signal coming from the z+ pin just like I did.  See the first link in my comment for a schematic showing how to make a simple logic inverter.

OhmsLawBreaker, 2019-07-24:
@lexelby I'm wondering if there is a way to do this in the code rather then using the transistor? I'm in a bit of a time crunch and it would take me some time to order some transistors in.

I defiantly agree that the transistor would work very well (as you have proved) but I'm almost sure the output of the PWN pin should be able to be changed in the program??

lexelby, 2019-08-04:
Oh sure, you could do it in software.  The tricky bit is figuring out where in the code you'll need to make the changes, and I have no idea.

Emoslapper, 2019-10-26:
WOW!!! IT WORKS! IT WORKS!!!
I want to say a BIG BIG Thanks you to everyone in this thread\form\tab.. Whatever I am over the moon and just wanted to thank this whole group on the help, time and effort that they have put in.
Big shout out to StuartB4, arkypita & lexelby for contributing so so much and helping everyone with their suggestions & have lead me down the path of successes!
So this is what I did and what setup I have that is working.
Checked all the setups for Grbl config & Lasergrbl setup as StuartB4 has shown a few times ;)
I set my LaserGrbl like this I am using Dyi DVD Laser….

[image]
 
So this is the wire diagram I came up with… I know this will make electrical engineers cringe……Sorry

[image]

And again thank you all :D To helping to getting this Arduino Uno + cnc shield v3 + laser working :D

thecuress, 2019-12-03:
hi i want to connect this laser on my cnc shield v3 (5500mw)https://www.amazon.it/Macchina-incisione-Incisore-Controllo-focalizzabile/dp/B07VV6VTJD/ref=sr_1_1_sspa?keywords=laser+5500mw&qid=1575405193&s=tools&sr=1-1-spons&psc=1&spLa=ZW5jcnlwdGVkUXVhbGlmaWVyPUEyM0JPUjJLRVZVS1JIJmVuY3J5cHRlZElkPUEwMjQ3OTY5M01VTDRVNk1DSlJHWCZlbmNyeXB0ZWRBZElkPUEwOTQ2NTY4MzdQR0FYMkVDMkxCVCZ3aWRnZXROYW1lPXNwX2F0ZiZhY3Rpb249Y2xpY2tSZWRpcmVjdCZkb05vdExvZ0NsaWNrPXRydWU=

i want to kmow the connection ..tks

hiddenvs, 2019-12-26:
What if I have such laser module ?
[image]

I tried to connect it to CNC shield V3 with Arduino Uno and GRBL v1.1f. LaserGRBL works with it, but I have (of course) allways 0 or 100% power out from that module. Has someone figured out how to use displayed PWM/TTL signal input ?

Actually is PWM/TTL connected to Spindle Dir on CNC shield and controlled via M3 (off) / M4 (on), which is working. But I can control intensity (power, whatever). Zero or full power of 5500mW is there. 
I am new in laser engraving, I have read about modules a lot on forums, but allways found 3-wire connection modules in use cases. Google was also not very helpful. Thanks in advance for any idea.

StuartB4, 2019-12-27:
PWM wire should be connected to Z- or Z+, does'nt matter which one, they are both connected together in the circuit on the bottom of the CNC Shield.  It goe's to Pin 11 (PWM out) on the Arduino if you have GRBL 1.1.

hiddenvs, 2019-12-27:
> 
> 
> PWM (red) wire should be connected to Z- or Z+, does'nt matter which one, they are both connected together in the circuit on the bottom of the CNC Shield. It goe's to Pin 11 (PWM out) on the Arduino if you have GRBL 1.1. The Black PWM - wire should go to any one of the ground pins in the black row on the right of the CNC shield.

Thanks for reply. One question in addition. What if I am using 4-axis system (Reprap 3D printer) and want to have limit switches connected (now I have them not connected due to tests with laser module). Is there any option to re-route the PWM signal for laser somewhere else, like f.e. to pins for SPINDLE DIR or SPINDLE EN ? This is maybe stupid question, but without any experience it's better to ask, before I "destroy" something... Thanks in advance for answer.

StuartB4, 2019-12-27:
It may be because the laser does not have a PWM driver board. If it is just the laser and the fan you will need to buy a seperate PWM controller board. Can you see inside underneath the fan if there is any other electronics like a small circuit board. If not then something like this will be needed.
But that's the only way to control the power with that CNC shield, from the Z+ or Z- and Ground.
My laser has the PWM board inside the heatsink for the laser and fan and is connected that way.

[image]

StuartB4, 2019-12-27:
The reason it is working at full power or nothing when you connect it to the Spn Dir pin is because it's a fixed voltage/current. PWM is a signal varying voltage between 0 volts and 5 volts. It comes from Pin 11 on the Arduino and goes straight to Z+ and Z- on the Version 3 shield. On the laster version it goes to SpnEN pin.

[93 more replies]