arkypita/LaserGRBL #369: Limited On/Off Commands

Catch229, 2018-05-29:
In the configuration for the machine you are given drop down menus for the on/off commands with the options of M3, M4 and M5 respectively. This is great for most normal situations but I recently added a laser to my 3D printer via a fan output that is PWM. The issue is that the fan output is controlled by M106 (on) and M107 (off). I have been using Notepad++ to just replace the commands in the GCode but I think adding the option to have custom on/off commands wouldn't be a bad idea for the future.

Thank you!

arkypita, 2018-05-30:
Already in development roadmap

closed 2018-05-30 by arkypita

contagon, 2018-06-08:
@Catch229 Could I ask how you've been using Notepad++ to switch the codes to M106/M107? I've tried a few different ways to do it, but the steppers always seem to work fine, but I can never get the laser to work correctly.

Thanks!