arkypita/LaserGRBL #370: set up two custom buttons

brucect, 2018-05-29:
arkypita, I want to place two custom buttons, one laser on and one laser off. for focusing the laser. This is what I input but I get an error with M3, M5 is ok.

M3
G01 S100     "undefined feed rate"

M5
G00 S0

did I forget something or what
Thanks

StuartB4, 2018-05-29:
I use this:

$32=0
M3 S10

Obviously the S value can be whatever value you choose upto 255.

brucect, 2018-05-30:
> This is what I have to laser.

And this is what's it looks like

brucect, 2018-05-30:
Sorry.  This what the laser is burning.
What do I need to reset??

arkypita, 2018-05-30:
Undefined feed rate means you should set F value before any G1

arkypita, 2018-05-30:
G1 means "move the laser" so speed is required. Why use move command for focusing?

brucect, 2018-05-30:
Ok, I used your post and the two buttons work fine now, thank you.
Now, the feed rate. Not sure how to do that, it's an override?
Thanks

brucect, 2018-06-01:
StuartB4, How do I set my work area so Lasergrbl doesn't run away every
time I go to burn.  I have and end stop at each axis location:   at X-, X+
and at Y-, and Y+. My work area is 10" wide and 11" long. This  is about
1/2" away from each end stop. I think this may be the last hurdle to
complete the laser engraver, I appreciate all the help you and arkpita have
given, I can't thank you both enough.
Bruce

StuartB4, 2018-06-01:
What do you mean by run away.
If you load an image and prepare it for burning you should have set any offset from the lasers home position to the start point on the work piece. 
So if your laser is @ X0 ,Y0 (usually bottom left) and you work piece is positioned at say X20 , Y20 away from X0 , Y0 then you would have set the offsets to X20 , Y20 and the laser should move to there as the start of the burning point.
[image]

May be something to do with $130 and $131 in GRBL Settings.
[image]

brucect, 2018-06-01:
StuartB4, got it, just didn't understand at first.
Thank You

StuartB4, 2018-06-01:
Ok no problem.
Just one other thing to mention.
I assume your steps per millimeter for your motors are set correctly.

You can check by starting LaserGRBL and clicking on Connect.

Make sure the laser is in the home position.
Then Left click the little up and down slider once (the one on the right) at the bottom left of LaserGRBL main window.
Then use the keyboard up or down arrow/cursor keys on your keyboard to move the slider up or down to set the slider to 30. 

Then Left click the Right Arrow on LaserGRBL next to the House icon once, the X Axis should move 30 mm.
Get a ruler and actually measure how far the Axis has moved from the Home position.
If it's 30 mm then you are ok.
Then do the same for the Y Axis, if that moves 30mm you are ok.
It either Axis moves more or less than 30mm then you need to edit the values in the Steps Per mm in GRBL settings.

If you do need to change the values just click on Grbl menu top left of LaserGrbl main window, click on Grbl Configuration and you should get a window like this.

[image]

Your settings will be different to mine though.
Move the mouse to the Value coloumn and click/double click to select either X ($100) or Y ($101) row.
Then you can change the values of the Steps/mm. When you have changed the first one press Enter to move to the next one if needed.

When you have edited the values click on the Write button at the bottom and then click Close.
Those settings will then be saved in to GRBL forever, until you change them again.
It may take a few attempts to get them just right, just repeat the process until it actually moves 30mm when you tell it to.

brucect, 2018-06-01:
my steppers are jumpered out Micro

$0=10
$1=25
$2=0
$3=6
$4=0
$5=0
$6=0
$10=3
$11=0.020
$12=0.002
$13=0
$20=0
$21=0
$22=0
$23=2
$24=50.000
$25=635.000
$26=250
$27=1.000
$30=1000
$31=0
$32=1
$100=314.961
$101=314.961
$102=314.961
$110=635.000
$111=635.000
$112=635.000
$120=50.000
$121=50.000
$122=50.000
$130=225.000
$131=125.000
$132=170.000

StuartB4, 2018-06-01:
Thats ok, they are probably fine.
But if you think they might be moving too far or not far enough at some time, you can check them using the method I posted earlier.

Tell an axis to move a certain amount, then measure the movement, and adjust setting if necesseary.

brucect, 2018-06-02:
Yes, after doing what you said about the 30mm it's ok.
Now, where ever I set Zero for X and Y on my work area where is where the
program starts, correct?
Should I set each end stop for each axis limit of travel, so if for some
reason the finished size is larger than my work area the axis won't just
run off the grid and keep running? Does that make sense?
Thanks for the help again.
Bruce

closed 2018-07-05 by arkypita