arkypita/LaserGRBL #373: How to stop over burning at the end of lines.

StuartB4, 2018-05-31:
Every time a burn an image with quite a dark background I get too much burning at the end of the lines.
The laser switches of at the end of a line and comes back on for the start of the next one, but for some reason I get overburning down both left and right edges of the image. 
If I'm using card the laser almost burns straight through it.
The rest of the image away from the edges is fine.
I've tried lowering the number of lines but then the picture is not good enough.
Tried turning the laser power down but then it does not burn the rest of the image dark enough.
Tried changing the burning speed but still get the same result.

These 2 images were set at 5 Lines.
The more lines I set it to for better quality the darker the overburning gets.

[image]

[image]

arkypita, 2018-05-31:
This is due to the deceleration of the laser. Using M4 (laser power corrected by speed) can help fixing this problem

arkypita, 2018-05-31:
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Laser-Mode

https://github.com/gnea/grbl/issues/34

arkypita, 2018-05-31:
Another solution is to add a very light margin to the image  (not white, as white will be optimized/skipped by lasergrbl). An "acceleration border" feature should be already in dev roadmap

StuartB4, 2018-05-31:
Ok, I will have a look at that.
Thankyou.

closed 2018-06-12 by arkypita