arkypita/LaserGRBL #377: reestablish cnc code

markk13, 2018-06-06:
help please, I need to re-establish the arduino code of my cnc2-1720 has grbl v2.5, delete the arduino code by mistake

StuartB4, 2018-06-06:
Do you mean LaserGRBL 2.5 ?
What make is your machine ?

StuartB4, 2018-06-07:
If you want to reinstall GRBL on to the Arduino Nano board that your machine has then you could go here:  https://github.com/gnea/grbl
Click on the green "Clone or download" link and click Download Zip.
Then read this about how to install (flash)  it to the Arduino.
https://github.com/gnea/grbl/wiki/Flashing-Grbl-to-an-Arduino

Or you can just download the hex file and download a little program called XLoader
Hex file is here: https://github.com/gnea/grbl/releases
XLoader is here: http://xloader.russemotto.com

Then read this about how to flash the hex file to the Arduino.

1: Download and extract XLoader.
2: Open XLoader and select your Arduino's COM port from the drop down menu on the lower left.
3: Select the appropriate device from the dropdown list titled "Device".
4: Check that Xloader set the correct baud rate for the device: 57600 for Duemilanove/Nano (ATmega 328) or 115200 for Uno (ATmega 328).

5: Now use the browse button on the top right of the form to browse to your grbl hex file.
Once your grbl hex file is selected, click "Upload"
After clicking upload, you'll see the RX/TX lights going wild on your Arduino. The upload process generally takes about 10 seconds to finish. Once completed, a message will appear in the bottom left corner of XLoader telling you how many bytes were uploaded. If there was an error, it would show instead of the total bytes uploaded.

markk13, 2018-06-10:
I already reinstalled everything, but I can not achieve this result

https://www.youtube.com/watch?time_continue=22&v=conZiopJF3k

StuartB4, 2018-06-10:
What is it doing or not doing exactly?

StuartB4, 2018-06-11:
It's hard to say why that happens, could be a hardware problem / fault or GRBL settings.
Could be interference in the cables if they are not properly shielded.
Can you upload a picture of your GRBL settings?
You can get them by clicking on Grbl at the top left of the main window, then clicking on Grbl Configuration. 

1. Does it stop everytime you use it?
2. Does it stop in the same place everytime or after the same amount of time, like 5 minutes or is it
different for every image you burn?
3. Are you using LaserGrbl to create your GCode, or are you importing GCode files made by another program?
4. Have you tried burning in Horizontal, Vertical and Diagonal? If so, does it do it with all 3 options.
5. Does it do it in Line To Line Tracing mode and Vectorize mode.
6. Do you have any Power Saving modes set in your computer. Some things can interrupt programs when they switch to save power. Like this in Control Panel, Power Options.

[image]

markk13, 2018-06-14:
[image]
[image]

arkypita, 2018-06-14:
http://lasergrbl.com/faq

arkypita, 2018-06-14:
Similar to:
https://github.com/arkypita/LaserGRBL/issues/378
https://github.com/arkypita/LaserGRBL/issues/321

StuartB4, 2018-06-14:
If you now have GRBL 1.1 you should make sure the PWM/TTL signal wire from the laser goes to pin 11 on the Arduino, and also the Ground or "-" from the PWM/TTL should go to an Arduino GND connection.

Also make sure you enable "Support PWM" in LaserGRBL settings, like this:

[image]

Also change these settings:
$30 only needs to be 255
$32 should be 1

[image]

Then change this setting to M4 if it is M3.

markk13, 2018-06-14:
I'm doing tests and the same thing happens, it stops almost in the same place, I charge images JPG
[image]

here stopped now (picture)

markk13, 2018-06-14:
ok, now I do the configurations, and test, thank you

markk13, 2018-06-14:
Friend,

[image]
 when change $30 = 255
$32 = 1

the laser stops recording, i dont know why

StuartB4, 2018-06-14:
I've just had a look at the machine from this link you supplied:
https://es.aliexpress.com/item/500-mW-2500-MW-5500-MW-escritorio-DIY-violeta-Laser-grabado-m-quina-CNC-impresora-rea/32850627495.html

I can't find any mention of the machine being able to do greyscale engraving.
It does not look like it has PWM/TTL feature.
It may just be a fixed power laser, which means you wont be able to do photographs like in the video on youtube that you posted.

https://www.youtube.com/watch?time_continue=22&v=conZiopJF3k

markk13, 2018-06-14:
I can grayscale like in the video, but it stops the machine and the statistical laser,
[image]

in 
#321 

say: "Grbl 1.1f takes up a significant part of the board's memory (~93%) on a 328p chip with 32kB and this leads to instability in operation.

I've since flashed grbl 0.9j which fills the memory to about 70% and, despite the downside of less features, this appears to have resolved the issue in my case."

hom flash Grlb to 0.9j? because i have arduino 328p chip

StuartB4, 2018-06-14:
I have an Genuine Arduino UNO, and even though GRBL 1.1 does take up most of the memory I don't have any issues with it, it just works how it should. 
Can you burn a picture and show us the results.

arkypita, 2018-06-14:
> Grbl 1.1f takes up a significant part of the board's memory (~93%) on a 328p chip with 32kB and this leads to instability in operation. I've since flashed grbl 0.9j which fills the memory to about 70% and, despite the downside of less features, this appears to have resolved the issue in my case."

The message "Low-memory available, stability problems may occur"  emitted by arduino IDE when flashing grbl is only a warning that warn that the program usa almost all the memory space of mcu, so the program should be well programmed to not allocate dynamic buffers that can fill up all the remaining space.

The grbl developers ensure that grbl is stable despite this warning ([see grbl faq](https://github.com/gnea/grbl/wiki/Frequently-Asked-Questions#i-get-this-low-memory-available-stability-problems-may-occur-warning-after-i-compile-grbl-is-there-a-problem))

All arduino board, original or compatible, use atmega328p chip and have exactly the same memory space so, definitly, this message is not the cause of your problems, and grbl v1.1 is ok to be used on atmega328p

arkypita, 2018-06-14:
The image you have uploaded clearly shows that you have communication problems between PC and arduino.

[image]

This lines show that LaserGRBL has received weird data from serial line that cannot be interpretated.
If you had read the faq i have linked here to you some post ago, instead of ignoring it, you can easly find causes and solutions.

arkypita, 2018-06-14:
again: http://lasergrbl.com/faq

arkypita, 2018-06-14:
> I've just had a look at the machine from this link you supplied: https://es.aliexpress.com/item/500-mW-2500-MW-5500-MW-escritorio-DIY-violeta-Laser-grabado-m-quina-CNC-impresora-rea/32850627495.html

> I can't find any mention of the machine being able to do greyscale engraving.
> It does not look like it has PWM/TTL feature.

I also think that this engraver is not able of power modulation, so should be configured with no "PWM Support" in LaserGRBL raster import settings and maybe also with $32=0

> It may just be a fixed power laser, which means you wont be able to do photographs like in the video on youtube that you posted.

You cannot do a "PWM grayscale" but dithered grayscale is available.

closed 2018-07-05 by arkypita

dungjk, 2018-07-06:
Hi @markk13 
There some thing you should consider to check
**Hardware**
- Check if your board is compatible with GRBL 0.9+ or not https://github.com/gnea/grbl/wiki/Connecting-Grbl. Some China's borad still not compatible with GRBL 0.9+ and some got hardware issue on it. If laser control pin is connected to D11 (PWM enabled), you're lucky, your board is compatible with GRBL 0.9+, that enable PWM for laser module. If not, you can modify the board a little bit to swap the pin. Check the link above.
- Make sure that you connect laser module correctly.

**Software**
- Check if GRBL confuration is correct, $32 should be set to 1
- Follow instruction to use LaserGRBL http://lasergrbl.com/en/configuration/. Or/And see @StuartB4  comment above.

kyroncfel, 2018-07-11:
Hi, Just a little note on what I have found in my journey with GRBL and the UNO.
I found that the firmware v1.1f has given my lots of issues with engraving and homing.
So i went a step back to grbl v1.1e and all have been good ever since. Just want to share as
it may help someone.

StuartB4, 2018-07-11:
Thats good, thanks for letting us know.

hk89, 2018-07-12:
Would be awesome if you tell us what issues you found with grbl 1.1f

markk13, 2018-07-22:
friend i found the problem, the problem was my PC Processor, i have an intel inside, no problem with core i3, i5 or higher

arkypita, 2018-07-23:
@markk13 "intel inside" is not a type of processor, it is only a sticker that say that the PC use a intel processor (like i3, i5 etc). Btw cpu does not matter with LaserGRBL-grbl working.

@kyroncfel thousand of user use LaserGRBL with grbl v1.1f without any issue. In my opinion your issue does not depend directly/only by the grbl version.

Based on my experience connection problems, hang, transmission related problem are 50% caused by electrical noise and 50% by poor quality chinese arduino clone.

Before say that the problem is software (windows, grbl, lasergrbl etc) please obtain and use an original arduino, a good USB cable and take the measures to have a proper shielding of motor and data cables.