arkypita/LaserGRBL #378: Problem!!! Error: Reset

Revenger72, 2018-06-07:
Hi. Please suggest how to defeat this problem: At an arbitrary moment, an error appears (in the photo) and the laser stops, and sometimes even the laser does not turn off. FAQ read, replaced the UNO on NANO (twice), reshaped and cleaned EPROM, made a separate power supply for arduino 5v., replaced the laptop with an ordinary computer, the machine was grounded. Laser CO2 40W. Nothing helps. It is impossible to leave unattended and start responsible work for processing.

[url=http://radikal.ru][img]http://b.radikal.ru/b30/1806/94/923c82ab7a92.jpg[/img][/url]

arkypita, 2018-06-08:
Two could be the cause of unexpected reset:
1) transmission problems put grbl in crisis in communication area
2) too fast axis movement together with high microstepping resolution put grbl in crisis in motor commanding area (too high step frequency to produce)

Possible hardware solution
1) use original arduino, not chinese clones. at least use clone with 16u2 chip, not ones with CH340
2) decrease microstepping (and consequently $100, $101, $102) if  too high frequency*speed. What is your $100, 101, 102 values?

Possible software workaround:
Configure threading and streaming as "Syncronous" and "Quiet"
[image]

However never left laser unattended.

Revenger72, 2018-06-08:
Thanks for the answer. Microsteps are not many - 100-101 =40 and 80.
The original Arduino use no opportunity.
Configure the "threads" I'll try the others.

-------------

I use M4 for engraving - an error appears. Also, if I use the M3 for cutting, it's also a mistake.

Revenger72, 2018-06-08:
Changing the parameter "**Threading Mode**" to "**fast**" gave a positive result!
For the whole evening of tests (etched for 20-30 minutes the same photo) just once the error appeared! Tomorrow I'll continue the tests.

closed 2018-06-12 by arkypita

dungjk, 2018-06-13:
Hi @Revenger72 
I think some possible cause:
1. Electrical noise cause communication be interrupted.
2. Electrical noise cause your board reset.
3. Your PC get to sleep make the communication interrupted

- Leave USB cable away from noise source (stepper motor wires, AC main)
- Try with better USB cable with noise filter ( https://en.wikipedia.org/wiki/Ferrite_bead)
- Turn off sleep mode on your PC

If you upload the LaserGRBL's log it may help for further issues.

Revenger72, 2018-06-13:
Thank you.
The log will try to find and download. The computer does not use sleep mode. The wire did not change, but with it everything worked perfectly earlier. Even Arduino Uno with a poorer wire worked without errors.
Probably there are obstacles, but I do not yet know how to find and clean them. Changing the program settings is better. The error appears rarely.

arkypita, 2018-06-14:
Are you using an original arduino board or a clone? What is the serial chip on your board? CH340 or 16u2

[image]

Communication problems between PC and arduino with CH340 serial chip [was deeply analyzed](https://github.com/gnea/grbl/issues/294#issuecomment-337801846) by the user @109JB and its result posted on grbl issue board.

Its result show that CH340 is not 100% reliable. Only solution is use original arduino board (or at least clones with 16u2)

So: if you want to leave your engraver unattended use original board (and install some kind of "fire alarm" too!).

Revenger72, 2018-06-14:
Of course not the original (ch340). But after changing the settings "thread", as I was told earlier, the error began to appear rather rarely.

There is an Arduino Mega with the original interface (16u2). Can I just stitch it and try it?

arkypita, 2018-06-14:
While grbl streams gcodes commands it also sometime query for grbl status (current laser position etc..) and grbl will reply with status message. This is needed to see the update of the position on screen and for info about buffer, machine state etc.

Once asked the status reply is sent by grbl to pc at unpredictable moment, usually at the same time pc is sendind some gcode.

Seem that CH340 has some issues managing transmitting and receiving at the same time, specially at high speeds.

**fast** or **quiet** mode will slow down sending speed and reduce the frequency of status query request. You can notice that position is updated with less speed.

This can reduce issue on CH340 but cannot guarantee that the problem is solved: it is only statistically reduced

arkypita, 2018-06-14:
Yes, you can use arduino mega but you must flash the specific grbl version for mega.

trustfm, 2019-01-11:
Hello there ! I am not aware of LaserGRBL. I was working into another private project using teacup but i think this issue is valid and at grbl. I had to move from grbl to teacup for other reasons. 
Now on the transmissions errors and restes of CH340 arduino clones. 
I have made a program called tinyserial in order to test the trasmission errors . 
I get constant errors at loop 40 / 50 at most with all mine arduino clones. 
The good news are that my custom Trustduino board with the CH340 chip works with tinyserial and with my private project without issues ! 
http://trustfm.net/hardware/Images/Trustduino/tinyserialv1.0.zip

In the zip file you can get PCB and schematcs of the trustduino board based on CH340 that at least for me does not have serial transmission problems ! 
Have a great day 
Please post if you got any success !

toll101, 2022-02-27:
Спасибо за интересную информацию.  Я построил лазерный ЧПУ на ардуино. В принципе всё работает но возникает ошибка. Подскажите, пожалуйста, что моет быть неправильно.  Если я получаю такие ошибки 
Спасибо. 
[image]

arkypita, 2022-02-28:
https://lasergrbl.com/faq