arkypita/LaserGRBL #379: Serial receive buffer size

mayhem2408, 2018-06-11:
I use both a grbl-Mega and a grbl-LPC on my raster laser engravers as they allow me to run my 7.5W and 40W lasers at much higher speeds than the standard grbl can handle. I have noticed that LaserGRBL does not utilized the full serial receive buffer on the mega or the LPC. LaserGRBL appears to be fixed at the 128 bytes available on the 328p. Grbl-Mega has a serial buffer of 256, and my Grbl-LPC has a massive 10240 byte buffer. Can you add either a feature to manually set the buffer size LaserGRBL will use, or have LaserGRBL read the buffer if $10 is set to 2 or 3.
With $10 set to 2 or 3, the return status from '?' with show the internal buffer state. On startup then the buffer is empty, it will show something like this:

```
?
<Idle|WPos:0.000,0.000,0.000,0.000|Bf:249,10240|FS:0,0|Pn:PXYZA>
ok
```

Bf shows the planner buffer and serial buffer available.

Grbl-LPC with it 250 planner buffer and 10240 serial buffer allows me to run the most complex dithered rasters at feeds up to F15000 without stuttering.

arkypita, 2018-06-11:
Hi mayhem
Supporting higher buffer size it's so easy in LaserGRBL that the only reason I did not do it is that I do not know about these differences.

I would prefer to read the value from the configuration rather than from the status report, because it is not guaranteed to be reported.

The welcome message is the same or not? Maybe it could be possible to detect mega and lpc version from welcome message?

mayhem2408, 2018-06-11:
@arkypita Unfortunately '$' and '$$' will only return the $10 reporting mode, but will not report the buffer size.
The best I and suggest is if '?' does not return Bf values, play it safe and default to 128 buffer size, but if '?' does show the Bf values, you can use it.

arkypita, 2018-06-12:
https://github.com/arkypita/LaserGRBL/releases/tag/v2.8.46

arkypita, 2018-06-12:
Config option added

closed 2018-06-12 by arkypita

mayhem2408, 2018-06-12:
@arkypita Perfect. I was testing 2.8.46 just as you released 2.8.47. 2.8.47 works great. I like how the progress bar shows two colours to show the serial data sent, vs data processed. A couple of my small rasters get 100% sent while still only at 10% processed. I have also tested the same board with $10=0 and it successfully defaulted to the smaller buffer. $10=2 and it detected the full buffer. Great work.