arkypita/LaserGRBL #381: Would it be a big job for you to make LaserGRBL a 3 Axis program?

StuartB4, 2018-06-12:
Adding a Z Axis for either focussing or for when burning/engraving thicker materials.
Just wondered as GRBL supports 3 Axis.

arkypita, 2018-06-12:
for focusing: custom button
for cutting thick materials: move z already added in #64

StuartB4, 2018-06-12:
Ok, thankyou.

Geekoid85, 2018-06-22:
Hello, the software is great for laser engraver. But like laserweb4 I want to use it with a plotter and a drag knif cnc cutter (vinyl cutter). Can you add the possibility to choose the "on" and "off" gcode command ? for exemple on and off = G1 Z0mm and G1 Z10mm for my machine.
 Also, it would be great to make a text box for custom start gcode and custom end gcode ;)

Thank !

arkypita, 2018-06-22:
Hi @Geekoid85 

I can guarantee that a seemingly simple thing like adding a custom on-off command is actually not that simple. I have recently tried to add a simple variant like M106 and M107 (fan ON and fan OFF) sometime used by 3d printer with marlin firmware ad when I have quite finished the task I have decide to undo all changes because of the impact and side effects of that change.

[image]

One of the  motivations is that ON OFF command are managed in LaserGRBL with a high level of optimization. In particular they are managed as "modal commands" (as they are) avoiding sending redundant commands.

Another reason is the need to recognize clearly in preview drawing: M3 and M5 are clearly recognizable in every place they are placed inside a loaded gcode, when G1 is not a "laser on", it is only a move. What should be if you define "G1 Z10" as "Laser ON" and then load a file with "G1 Z11" or G0 Z9 or G1 in one line and Z move in another line?

More: what define if that commands are able of power modulation or not? And with wich syntax they modulate?

To add a new set of ON/OFF commands it should be defined how they should be managed, by the point of view of modals behaviour, optimization, preview, power modulation etc and that definition could be conflicting with M3/M5, so the resulting code should take this complexity in manage all variety.

arkypita, 2018-06-22:
As the code is written now, I prefer not to "mess it up" with too many variations. In the future, when I have the time, I will try to rewrite the parts that deal with these works so that they can be more virtual, allowing to define custom command sets.

closed 2018-07-04 by StuartB4