arkypita/LaserGRBL #382: Append File during Run?

mikemorana, 2018-06-13:
I'm thinking of the 'append file' option working also as a sort of queue-up.
Where if a run had already begun, a dialogue box would confirm "you want to run this file at the end of the current run?"
This isn't pivotal for me at all. It occurred to me as an afterthought when while running an etch, I realized it was a bit too light. I didn't want to stop the etch because it was already several hours in. I didn't want to lose the expensive board of wood it was etching either, so I wanted to append a second run but with an increased S-MIN. If i had aborted and run again, there'd be an area darker than the rest where the laser had passed over twice. The first file was going to finish at night while i was out of the shop so, as a result i lost 5-6 hours of productivity.
This has happened a few times but I'm aware that some planning and double checking could avoid the need altogether. I just figured I'd offer the suggestion if it weren't particularly difficult to implement.

BallscrewBob, 2018-06-17:
Done similar myself but usually just tweak the settings when the first jobs has done.

Would be a nice addition as some wood even the same type doesn't always engrave the first pass even though the prev, piece did with the same code. Age of the wood and any pre-treatment can make a mjor difference so I am getting into doing a simple test burn on a scrap end first....Pine seems to be my worst offender with closer grain giving better results than coarse grain.

StuartB4, 2018-06-18:
How about using the Override sliders at the bottom status bar of the main window instead of re-running.
http://lasergrbl.com/en/usage/overrides

[image]

closed 2018-07-05 by arkypita