arkypita/LaserGRBL #383: Windoes desktop shortcut issue.

BallscrewBob, 2018-06-18:
Since the last two updates the desktop shortcuts are subject to getting wiped out with use is CCLEANER or WISE Registry cleaner.

It appears to be a combination of both those utilities.
Previous to that I had never had any issues.
It also looks like this extends to the program list too.
There was a recent update to CCLEANER but I am almost 100% certain this problem pre dates that.

arkypita, 2018-06-19:
Hi bob
I have not changed the installer package since long time, so i think it is not related by a change I made.
To build the installer I use Inno Setup studio, a tool for build software distribution package. I can try to send your issue to them.

BallscrewBob, 2018-06-19:
The copy / paste method seems to have STUCK so I think you can close this but another test with the original did see the shortcut dissapear again.
Dont see any immediate difference between the two..just a small but odd bug.

closed 2018-07-05 by arkypita