arkypita/LaserGRBL #384: problem drawing full circle

frikeldon, 2018-06-21:
When iI send this gcode to my grbl 1.1f laser machine it draw a circle, but in laserGRBL's preview, the cricle does not appear.

````gcode
G90
M4
G0 X25 Y12.5 F3000 S0
G3 X25 Y12.5 I-12.5 J0 F1000 S200
G0 X0 Y0 F3000

arkypita, 2018-06-21:
Thanks for issue reporting. I have found the buggy code but I need some time to find the best solution.

Arhimed-Ru, 2018-06-27:
I'm not a programmer
but like this - works:
G90
M4 
G0 X25 Y25 F3000 S0
G0 X25 Y12.5 
G2 X25 Y12.50001 I-12.5 J0 F1000  S200
G0 X0 Y0 F3000
M5
G0 X0 Y0

arkypita, 2018-06-27:
I already know the causes of described issue:
1) My drawing code has some optinization that skip drawing of segment that start and end in the same place. This code does not take into account that an entire circumference starts and ends in the same place by its own nature
2) c# circle function require computing start angle and end angle that give the same results, so the result is an arc of 0° instead of 360°

arkypita, 2018-06-27:
@Arhimed-Ru your example just work as because of 0001

arkypita, 2018-07-03:
Hi all
Please test https://github.com/arkypita/LaserGRBL/releases/tag/v2.8.51

StuartB4, 2018-07-03:
Just out of curiosity I tried the code as it is that Arhimed-Ru posted above.

[image]

Although it looks ok on screen, this is the result.

[image]

Looks like it needs an Offset to the right to correct it.

frikeldon, 2018-07-03:
@arkypita I've just tested my gcode with v2.8.51 and the issue is solved. Huge thanks! However I've noticed a different behavior on the preview screen but I'm not sure if it's intended. The drawed figure is scaled by fitting it to the width of the preview area, ignoring the figure's height.

Here's a screenshot:
[image]

@StuartB4 I belive that your problem is on the firmware side instead of lasergrbl. This software sends g-code commands to the grbl firmware and is this firmware who "moves" the laser. I've tested both codes (my original, and @Arhimed-Ru version) with my laser wich has grbl 1.1f and it works correctly.

Here's the result (left side is mine and right side is @Arhimed-Ru one):

[image]

arkypita, 2018-07-03:
@frikeldon 

> I've noticed a different behavior on the preview screen but I'm not sure if it's intended. The drawed figure is scaled by fitting it to the width of the preview area, ignoring the figure's height.

Behaviour is not changed. The same issue was present in previous version but you are not seeing it because the previous screenshot was taken in a 4:3 form factor (that limit image width) while last screeshot is taken in 16:9 factor.

The issue concern computation of minimum and maximum X and Y for the burned image that is computed cumulating min and max of each start-point and and-point of each segment.
While this is ok on segments there is an issue on arcs where more complex calc is required instead of start-point and end-point coordinates.

arkypita, 2018-07-03:
@StuartB4 I agree... you have a firmware or maybe hardware problem

StuartB4, 2018-07-04:
I don't seem to have a problem with either grbl or my hardware. I have grbl 1.1f.
I checked hardware and GRBL settings and all are ok.
Then I decided to just reset the Arduino using the Reset button and it's now working fine, it must have just got confused.

commit 141fcb43 (2020-12-08): Fix #384

Improve arc range computation
closed 2018-07-05 by arkypita