arkypita/LaserGRBL #385: Parts of image missing.

StuartB4, 2018-06-24:
Just wondered if you know why this happens on some images.
After clicking Create, the image on the burning screen has bits missing.

This shows all the dots ok.
[image]

This has some missing.
[image]

arkypita, 2018-06-24:
The elaboration of the preview is done on a small sized rescaled image to be sure that the computation time for the preview be reasonably low, to make ui controls responsive. The size is about the preview window pixel size.

The elaboration of the final vector image and gcode is done on a different scaled image, using the final output size (in your case 36x36 mm).

Due to the different scaling of the images the effects of vectorize options can act differently. In your case "spot removal" or other alghoritm is removing all the path that became to small to be significant.

StuartB4, 2018-06-24:
Ok, thanks very much.

closed 2018-06-24 by StuartB4