arkypita/LaserGRBL #386: HI FRIEND

Alive44, 2018-06-25:
I m use Arduino Uno and sns shield with laser 1 w TTL  I m connect the laser signal with spe En pins an cnc shield and  set The image was formatted like a video but it did not work correctly. 
please Help 
with With sincere modesty for all
 _

StuartB4, 2018-07-04:
Hi Alive44.
Could you please be a bit more clear, it's a bit hard to understand.
I understand you have a 1 Watt laser with TTL.

What make is your CNC shield? and what version number is it.
And what do you mean by "The image was formatted like a video but it did not work correctly."
Do you mean your settings were the same as you saw on a video on youtube?, if yes can you post the link for the video so I can have a look and better understand the problem.

Also, which version of GRBL is loaded in to the Arduino?

Thanks.

Alive44, 2018-07-27:
Hi StuartB4 
thank you for your time 
the CNC shield is V3 
yes the setting same as a video on youtube 
and the Grbl is 0.9 
but The problem still exists

StuartB4, 2018-07-27:
I think you mean a Protoneer V3 Shield.
What you should do is update GRBL from 0.9 to 1.1

Then when you have done that you have to go to the Arduino Libraries folder and find a file called config.h
It should be in C:\Users\*your user name*\Documents\Arduino\libraries\grbl
You can edit that file with a free program called Notpad ++ from here:
https://notepad-plus-plus.org/download/v7.5.8.html

You have to find line number 339 in config.h file which should look like this:
#define VARIABLE_SPINDLE // Default enabled. Comment to disable.

Make sure there are no // in front of the "#" , if there are, remove them and click File menu and then Save.
If there are not any // then just close the file without changing anything.

Then if you did have to remove the //'s then you will have to recompile GRBL 1.1 and reupload it to the Arduino for it to take effect. If you did'nt have to edit the file thats fine, it's good as it is.

Then the TTL wires from the laser go to the Z+ pin and a Ground pin on the Shield.

For a 3 wire laser: 
The Red should be +voltage supply to power the laser (this can usually be connected to the + power connector on the shield if you have 12+ volt supply going in), the Black should be Ground and the Yellow or whatever colour it is goes to Z+ on the Shield, it can also go to Z-  as they are both connected together.

For a 2 wire laser:
If the laser has 2 wires for the TTL as well as 2 power wires the TTL wires still go to  Z+ or Z- and Ground.
1 Black for Ground, and the colour of the other wire depends on which make of laser you have, usually Yellow, but sometimes Blue or White.

There are other things to do but this should be enough for now.
Hope it helps.

Alive44, 2018-08-15:
thank you for your time my dear . but the problem still my laser is TTL put dont work with PWM its fire  evry think in picture

Alive44, 2018-08-15:
and i found in line 339 this ((// and the planner blocks. Each segment is set of steps executed at a constant velocity over a))

StuartB4, 2018-08-16:
Ok.
Did you update GRBL to version 1.1

If you are using Notpad++ copy the line of text below:

#define VARIABLE_SPINDLE // Default enabled. Comment to disable.

Open Notepadd++ and open config.h again.
Click on the Search menu at the top or press CTRL+F.
Then click on "Find" and paste the text you copied into the "Find what:" box and click the Find Next button.

That shoud take you to the correct place in the config.h file.
Iif there are 2 //'s in front of the  #define just remove them and click Save on the File menu to save the config.h file.
[image]

If you did have to remove the //'s then you will have to recompile GRBL 1.1 and reupload it to the Arduino for it to take effect. 
If you did'nt have to edit the file thats fine, it's good as it is.

StuartB4, 2018-08-16:
Just wondered if you have PWM mode enabled in LaserGRBL.

[image]

Alive44, 2018-08-17:
thanks for your kind my friend this picture of my setting 
[image]

Alive44, 2018-08-17:
[image]
[image]

Alive44, 2018-08-17:
[image]

Alive44, 2018-08-17:
[image]

Alive44, 2018-08-17:
now is best from the past but the size is error i set the (W) and (H) 25mm but when its work change Without interference to 90mm for both Axis ?

gisep, 2018-08-17:
Maybe that's because of your travel resolution.

You can set it in Grbl/Grbl Configuration $100,$101,$102.
It depends on your hardware, for example my motors run at 80 steps/mm.

If your picture is 90mm but wants to be 25, you have to do the proportion: ActualResolution*90/25.

closed 2018-10-13 by arkypita