arkypita/LaserGRBL #389: Custom button not working to stop program

allendr59, 2018-07-02:
I'm trying to setup a custom button to pause the program and turn the laser off, but cannot get it to pause the program. I can see where it adds the code to the log if the program is halted, but not while the program is running.

What am I missing?
[image]

StuartB4, 2018-07-02:
The Hand button in the bottom right corner already does that.

allendr59, 2018-07-02:
Actually it doesn't. 

It does stop the feed, but leaves the laser on. I need it to stop the feed, and turn the laser off.

StuartB4, 2018-07-02:
Thats strange, it switches the laser off on mine, and it comes back on when I press the little man button.

arkypita, 2018-07-03:
> The Hand button in the bottom right corner already does that.

> Actually it doesn't. It does stop the feed, but leaves the laser on. I need it to stop the feed, and turn the laser off.

The hand button send the immediate command "!" (Feed Hold). Feed hold is documented as: 

- Places Grbl into a suspend or HOLD state. If in motion, the machine will decelerate to a stop and then be suspended.
- By machine control definition, a feed hold does not disable the spindle or coolant. Only motion.

So this behaviour is perfectly by specific. (by GRBL specific i mean... LaserGRBL is only a graphical UI that cannot change how command are managed)

> Thats strange, it switches the laser off on mine

The behaviour is the default for grbl v1.1 with [Laser Mode](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Laser-Mode).
More: https://github.com/gnea/grbl/issues/34#issue-188984182

> I thought that the ! command was supposed to break into the program wherever it's encountered, but apparently not.

LaserGRBL use a queue for sending gcodes.
All gcodes sent using custom button will be enqueued for sending after all previous enqueued command. When you are running a file/program all file commands is enqueued at file open, so your button codes will be added at the end.

allendr59, 2018-07-03:
OK, I can see where the M5 command is being sent when I press the hand button, and, as specified it only stops the motion, but leaves the laser on. Since the custom button commands aren't handled the same as the hold button, is there a way to add additional commands to the hold button so that I can at least turn the laser off when everything else stops?

arkypita, 2018-07-03:
> Is there a way to add additional commands to the hold button so that I can at least turn the laser off when everything else stops?

No, there isn't.

Why not consider to update your equipment with grbl v1.1 and activate laser mode?

allendr59, 2018-07-03:
I did last night, it's now at grbl v1.1z.

It looks like I was missing the activating laser mode step when I tried it this morning though, so thanks for that!

closed 2018-07-03 by allendr59