arkypita/LaserGRBL #391: Image scale help

allendr59, 2018-07-04:
I have a .png file that I'm trying to etch, and the scale is way off!
The finished image should be about 100mm square, but when I try to run it, the working area is much, much larger! Large enough that my machine (457mm x 600mm, roughly) hard limits before the image is finished. I'm using the default values for X & Y movement, so I'm not really sure what else to look at. Is this just an issue with the conversion from .png, or is there something else I need to look at.

Thanks!

[image]
[image]

StuartB4, 2018-07-04:
For a start the 2 red boxes around the Border speed and the Filling speed mean that the speed of 1000 mm/min is actually higher than the setting in GRBL $110 and / or $111. Check to see what it is set at in GRBL Configuration settings on the Grbl menu at the top left of LaserGRBL main window.

Then load the image again and select the Crop tool and draw a box around the outside of the image.
Then click Next.
[image]

Then scale the size to what you want in the Size settings W and H.
Then click Create.
[image]

StuartB4, 2018-07-05:
Thats good to know.
The easy way to get the steps/mm right is to tell the machine to move the X Axis about 30 mm's.
Then when it's moved measure the actual distance it moved with a ruler. If it's more or less than the 30 mm's then you can fine tune it buy changing the amount in $100 and repeat the test.
Do the same for the Y Axis and adjust the $101

StuartB4, 2018-07-05:
When you have the machine how you want it and have loaded an image and prepared it for burning, click the File menu and select Save Program.
Then you can open it with notepad and edit it.
That way you can change the speed, resave the file and load it straight back in to LaserGrbl.
Like this.

[image]

When you save the file make sure it has the .nc extension instead of the .txt one.