arkypita/LaserGRBL #395: Laser not turn on

sankajalakshan, 2018-07-09:
i made the coreXY cnc machine and added a 2500mw 3p laser which has only 3 pins (vcc gnd pwm). and i connect vcc gnd to external power source (6A) and connect pwm to Z+ / D11 pin on the cnc shield. and im using the GRBL 1.1 (latset version) and i did not modified it.  however, when i passing the code through the lasrgrbl laser did not turn one. when i turn on the weak light button the laser will turn on as a low light. but the burning is not working.
[image]
[image]
[image]

StuartB4, 2018-07-09:
You say " im using the GRBL 1.1 (latset version) and i did not modified it."
Did you enable laser mode $32 in GRBL settings? Enable it by changing the 0 to 1
Did you set $30 to 255 and $31 to 0 ?
[image]

sankajalakshan, 2018-07-10:
i have changed those settings. but i did not do any changes in config file or cpu_map file.

StuartB4, 2018-07-10:
You might have a faulty laser. When the laser diode breaks it ends up looking just like an LED light.
So you would get the dim light but not it's not lasing. 
You could also have a faulty Arduino, do you have another one you can try?
It could also be bad solder joints on the CNC shield.
What is the power source for the Arduino, is it just from the usb (as mine is) or a seperate supply?

sankajalakshan, 2018-07-10:
12V 6A power supply is used and week light is working. but laser not starting to burn.

sankajalakshan, 2018-07-10:
Yes. Its the same power source

arkypita, 2018-07-10:
> You might have a faulty laser

To be sure your laser is working you can detach it from the arduino side, and try to power it up giving necessary GND/VCC and put TTL at VCC to switch laser on. It should power up at maximum power.

If not, laser is death/faulty

Be care to your eyes

arkypita, 2018-07-10:
> power it up giving necessary GND/VCC and put TTL at VCC

or leave it as connected and just move TTL (yellow cable) from Z+ to 5V like this

[image]

StuartB4, 2018-07-10:
You can also try the yellow lead on the Z- because it is electrically connected to Z+.
If Z+ had a faulty connection then Z- should still work.

kyroncfel, 2018-07-11:
Hi, did you ground the laser driver board to the arduino so that both share a common ground.

StuartB4, 2018-07-11:
It should already be grounded, he said he's using the same power supply to power the laser and the Arduino.

1estrider, 2018-08-01:
When you are importing the image to what you have set the laser power?
You also could create a button for the laser on and off. Set the on to full power. You might have a faulty laser if that doesn't start burning.

sankajalakshan, 2018-08-02:
sorry for the late replying. I think I had a faulty laser. so I ordered another new one and its working well. thank you all for your kind support.

sankajalakshan, 2018-08-02:
my new laser is working. but I need to know how to engrave a photo with reducing the qulity

StuartB4, 2018-08-02:
What do you mean by reducing the quality.
Do you want a lower quality engraving.
If so you can just change the amount of lines here:
[image]

closed 2018-09-30 by arkypita