arkypita/LaserGRBL #397: COM Menu goes a little crazy sometimes.

buildlog, 2018-07-15:
Sometimes my COM menu gets a little crazy. According to Device Manager, I only had 3 and 10.

[image]

buildlog, 2018-07-15:
Here is what Device Manager showed at the time.

BTW: Is it possible to show the descriptive text in addition to the number to make it easier to know which one to pick?

[image]

StuartB4, 2018-07-15:
Also you should change the BAUD rate to 115200

StuartB4, 2018-07-15:
If you are using an Arduino and you have the Arduino IDE running you should be able to see which one it is on. Then select the same one in LaserGRBL.

[image]

buildlog, 2018-07-15:
Thanks for the reply

Yes, I understand how to find which one to use. I just thought it would save a step if it is in LaserGrbl too. 

Why are so many duplicates showing up in the LaserGrbl list? 

I am using a port of Grbl on a different processor (PSoC5). The USB can run at any speed up to 2M. I pick the fastest one in LaserGrbl.

StuartB4, 2018-07-16:
> I am using a port of Grbl on a different processor (PSoC5). 

Maybe thats the reason. Never seen that happen before.

arkypita, 2018-07-16:
Hi @buildlog thanks for issue submit

The code that fill the COM menu use the .net framework function SerialPort.GetPortNames() that return a list of the serial port names for the current computer.

https://msdn.microsoft.com/it-it/library/system.io.ports.serialport.getportnames(v=vs.110).aspx

```
CBPort.BeginUpdate();
CBPort.Items.Clear();
foreach (string portname in System.IO.Ports.SerialPort.GetPortNames())
{
	string purgename = portname;

	//FIX https://github.com/arkypita/LaserGRBL/issues/31
	if (!char.IsDigit(purgename[purgename.Length - 1]))
		purgename = purgename.Substring(0, purgename.Length - 1);

	CBPort.Items.Add(purgename);
}
CBPort.EndUpdate();
```

As documented the port names are obtained from the system registry (for example, HKEY_LOCAL_MACHINE\HARDWARE\DEVICEMAP\SERIALCOMM). **If the registry contains stale or otherwise incorrect data then the GetPortNames method will return incorrect data**. 

Furthermore GetPortNames has some known bugs (releted to incorrect data in registry) that produce wrong results, so i have added that "purgename" to remove the last character if it is not a number.

In my opinion you should clean out your HKEY_LOCAL_MACHINE\HARDWARE\DEVICEMAP\SERIALCOMM registry key.

More info here: https://github.com/arkypita/LaserGRBL/issues/31

arkypita, 2018-07-16:
> BTW: Is it possible to show the descriptive text in addition to the number to make it easier to know which one to pick?

This is a good hint but require me to use a different function to obtain this information because the system function in System.IO.Ports.SerialPort does not give me this informations. I'll get a look if possible by other way

closed 2018-11-06 by arkypita