arkypita/LaserGRBL #398: LaserGrbl doesn't run

istruttoresp, 2018-07-16:
I built a cnc laser from old DVD looking some examples on web.
i was testing the the software and it was running at the beginning but then crashed.
i tryed to reinstall,  upgrade or dowgrade version, read allover the web looking some feature i was missing but...no resoult!
i read about java 8 but laserGrbl  works well before crush...now i can't start it again.

[image]

[image]

istruttoresp, 2018-07-16:
this is situation programs

[image]

istruttoresp, 2018-07-16:
[image]

istruttoresp, 2018-07-16:
i can't install 3.5... winzoz is big shit...

istruttoresp, 2018-07-16:
sorry but... the tar.gz is for linux right? any instruction for installation?

arkypita, 2018-07-16:
Why did you talk of tar.gz? LaserGRBL installer is .exe

closed 2018-09-30 by arkypita

appolinari, 2018-12-10:
Hi! 
Can not run LaserGrbl on WinXP.
When starting, it gives an error message:
[image]
My Win:
[image]

StuartB4, 2018-12-10:
Looks like you may have to update Microsoft's .Net files.
Have a look here:
https://dotnet.microsoft.com/download/dotnet-framework-runtime

appolinari, 2018-12-10:
Hello!
I deleted and re-installed NETFramework several times. Used also the recovery utility. But it did not work. I think the fact is that laserGrbl cannot work with NETFramework version 4.0. 
I just can not install on WinXP NETframework newer than 4.0.
Does anyone use a laserGrbl with WinXP?

arkypita, 2018-12-10:
LaserGRBL use framework 3.5

[image]

Work on XP

[image]

appolinari, 2018-12-11:
HI!
Thank You! 
With N.fw 3.5 it works!

MARVINU, 2019-07-09:
I´m using windows 10 64x. I ran many times the software LaserGRBL but, some time ago I´ve tried to run, the first screen appear but after the program did not run. What can be wrong with?