arkypita/LaserGRBL #399: allow to delete some vectorized shapes with mouse

owai1980, 2018-07-21:
It would be nice if we could select and delete some vectorized shapes. This would allow us to print the drawing on wood, and after we could use the same input file to keep only the cut shape. (and repeat this shape several times to cut the wood, with a stronger power)

this will require to detect on witch shape the user clicks... not always easy with strange shapes :-)

i hope you understand what i mean, sorry for my english...

StuartB4, 2018-07-21:
Nice idea.
Don't apologise for your english, it's better than some of the english speaking people I know, and I'm english. :)

StuartB4, 2018-07-21:
A way of doing it without any program changes would be to make 2 identical drawings.
One for the first burn and one with only the vectors you want to go over to cut out.
Just load the second one in after burning the first one.

owai1980, 2018-07-21:
thanks :)
yes, this is how i do it now... and it works, but i have to be careful about the sizes / scale. it's ok. but... if you need ideas for the future... maybe it's a good one :-)

btw, thank you a lot for this software, very nice piece of code!! and open-source!

arkypita, 2018-07-23:
With the actual drawing code it is not possible, once preview is drawn, to recognize the source file by clicking on picture.

closed 2018-07-23 by arkypita