arkypita/LaserGRBL #400: Module connection ESP8266

Mikalay31, 2018-07-22:
Hey. I was going to transfer the machine to Wi-Fi, but I ran into a lot of problems. Did anyone discuss this issue at the forum?

Mikalay31, 2018-07-22:
I correctly edited the sketch?  http://prntscr.com/k9ge6h        ._path   -  I did not change               _username -  I inserted the name of my access point              _password     I inserted the modem access point password Wi-Fi         -      then I asked for this sketch module ESP8266.

StuartB4, 2018-07-22:
Heres the image from the link that did'nt work for you Mikalay.
[image]

arkypita, 2018-07-23:
Hi, i am the developer of LaserGRBL and I am currently using ESP8266 as the only mode to connect my engraver.

1) I suggest you to use the [telnet version](https://github.com/arkypita/ESP8266-SerialTelnet) of my esp sketch. Do not use LaserWeb sketch.

2) Once you have uploaded the sketch on ESP8266 the esp expose an it's own WiFi network, you can find it in wifi network list. This wifi is unprotected, so you can connect with no password.

3) Once connected to ESP network go to address 192.168.4.1, you should now see an admin pannel that allow you to connect the ESP to you home wifi network, choosing from a list and entering password.

4) Once the esp is connected to your home wifi you no more see the esp network, this confirm you that the esp is connected to your private wifi.

5) open your router admin panel, check in wifi config area and you should see the esp connected and read its own assigned ip address (like 192.168.x.x)

6) run LaserGRBL and connect to 192.168.x.x:23

Also, please use the most updated lasergrbl version https://github.com/arkypita/LaserGRBL/releases/latest

Mikalay31, 2018-07-23:
Hey. I know everything about you) I already talked with you and corresponded with you on Youtube. You have helped me more than once)

Did I understand correctly to me how to use this link to install and configure ESP8266?   --->    https://github.com/arkypita/ESP8266-SerialTelnet

I understood everything - I will try - then write - what happened. I will definitely install the latest version of your link. Thank you ! 

Nicholas.

Mikalay31, 2018-07-23:
Forgive me - I did not see that you already threw the exile. I will try. Thank you !

Mikalay31, 2018-07-25:
Good morning !
I fought yesterday with ESP8266 for six hours.
I did not succeed. I went to bed at two in the morning. I lost yesterday this fight)
I did not understand the problem.

I have the version installed arduino ide 1.8.5 -  Could there be a problem with this?
http://prntscr.com/kapnrr
I do everything by reference to the links    https://github.com/arkypita/ESP8266-SerialTelnet

I even installed the version exactly:
"Please note: require version >= 2.4.0 Replace package URL with https://github.com/esp8266/Arduino/releases/download/2.4.0-rc1/package_esp8266com_index.json"
http://prntscr.com/kapniq

I used the sketch telnetserver  in the module 8266, Then connected according to the scheme you specified.
http://prntscr.com/kaq1po

After power connection - a new network appears ESP8266. 
I connect to it (it without a password).
I open the browser and insert the link  "http:\192.168.4.1"

At this point I stopped - This link does not open in the browser.
Help or assist please) At me yesterday the eye has already started to twitch)
http://prntscr.com/kapqj9

Maybe this is the problem because I'm using version arduino ide 1.8.5 ( IDE 1.6.x I  did not find where to download).

Another question  - lines in the sketch -
const char* update_path = "/firmware";
const char* update_username = "admin";
const char* update_password = "admin";
Do not have to change? I mean my network passwords Wi-Fi ?
http://prntscr.com/kaplhe

I have a modem that uses the address of the new sample. Here is the old sample http://192.168.1.1/, I have such a http://192.168.100.1/ - maybe this is the problem?

In the network connection settings, I found this http://prntscr.com/kapqzf I do not know whether this is an IP laptop or IP ESP8266 - I did not understand.

Another question:
http://prntscr.com/kapsxy
Is it normal that not all libraries are highlighted? Maybe I do not have enough libraries (those that look pale).
Maybe this is the problem?

Another question:
What to choose in the program LaserGRBL when using telnetserver ?
Telnet or LaserWeb ESP8266
http://prntscr.com/kaptuu

Forgive that many questions! I have not been to your home for a long time)
The last question that which also interests me:
How to update the version correctly?
I followed the link    https://github.com/arkypita/LaserGRBL/releases/tag/v2.8.50
I downloaded all three files  install.exe  Source code (zip)   Source code (tar.gz)
I did not find there an EXE file.
http://prntscr.com/kaq7ab
How to update correctly?
There is an instruction?

Thank you for your help !

arkypita, 2018-07-25:
> After power connection - a new network appears ESP8266.
> I connect to it (it without a password).
> I open the browser and insert the link "http:\192.168.4.1"

I think that you mean http:\\

> https://prnt.sc/kapqzf

You should set "obtain IP automatically" or assign to your PC an IP in .4 network, otherwise you are in network 192.168.100.x and you cannot reach address of 192.168.4.x because the netmask 255.255.255.0 is limiting your PC to consider as local address only the IP that match in first 3 groups.

Alternatively you can change netmask to 255.255.0.0

> What to choose in the program LaserGRBL when using telnetserver ?
> Telnet or LaserWeb ESP8266
> http://prntscr.com/kaptuu

Telnet, of course.

>How to update the version correctly?

Just run the exe

[image]

Mikalay31, 2018-07-26:
Good afternoon !
I'm glad as an elephant)
I got it! Your advice was very helpful!
Changing the mask to 255.255.0.0 did not help.
Another thing helped: it was necessary to put the automatic assignment of the IP address.
After that, immediately earned the link "http: \ 192.168.4.1"
As you said.

True, my joy was short.
There was a very big problem with the definition of IP address - which was assigned by ESP8266.

I went into the settings of my modem at http://192.168.100.1/ and there I found the IP ESP8266
http://prntscr.com/kb6yn3

But many users do not know how to do this.
I found another option - use the Wireless Network Utility Watscher
http://prntscr.com/kb725k
In this method, you do not need to know how to enter the modem - it's much easier.

On average, I spent 7 days to reflash ESP8266 (
On that to launch LaserGRBL on Wi-Fi - 3 days.
It's a lot.

I plan to shoot a video instruction on setting up a wireless connection - in Russian with subtitles in English.

I can then drop you a link to the video - if you are interested.
Write to me - you need a link or not.

Thanks for the help !
I admire your work!

Thank you very much !

Nicholas.

arkypita, 2018-07-26:
I know that the procedure is not simple and require electronic and informatics skills that not everyone has.

Lucky most users has auto-assigned IP address, so some problems you have encountered does not exist, but if you have a fixed IP (and maybe you don't know and you don't have skills to manage this configuration) this could be a big deal.

> I plan to shoot a video instruction on setting up a wireless connection - in Russian with subtitles in English. I can then drop you a link to the video - if you are interested.

I think this could be a good idea. link is appreciated

I know that some things can be done by me to make it simpler:
- pre-compiled code for ESP sketch and library
- ESP IP auto-discovery in LaserGRBL (network scan)

Also electronics part could be difficult, especially because of the 3.3v level conversion needed.
I found a good solution using this board

[image]

Mikalay31, 2018-07-26:
Most of all there were problems with the firmware of the module.
I programmed the module through arduino nano, and not through the converter - like you.
Here is the circuit by which I was able to achieve 100% of the quality of the firmware module[image]

Regarding static IP, this is the requirement of our Internet provider.
I manually set it up) It's necessary for stable operation ... I did not think that there would be such consequences)

When I have everything ready I'll drop the link.

Regarding Shield for ESP8266. I agree - it's easier.
But it seems to me that there is an easier option.
For example:
[image]

Then there is no need to reconcile the logical levels.

Nicholas.

StuartB4, 2018-07-26:
Hi Mikalay31.
Just wondered if you diagram above is correct.
I thought TX had to go to RX and Rx to TX.
How can you tramsmit from TX and receive it at the other TX.

Mikalay31, 2018-07-26:
Hey.

When programming the ESP8266 module, this should be the case.
ТХ -> ТХ РКс -> РКс and there should be a jumper from MLA to Reset - on the board of arduino.
This is the nuance of programming an ESP module using arduino.
I repeat the diagram:

[image]

And in the operating mode contacts cross (vice versa)
ТХ -> КЧ РКс -> ТХ.
Here is the scheme:

[image]

I'm sure of that - I tortured the ESP8266 for a whole week. I use it exactly as in the diagrams)

arkypita, 2018-07-26:
> Regarding Shield for ESP8266. I agree - it's easier. 
> But it seems to me that there is an easier option.

The image I have posted is not a ESP8266 shield, it is a full arduino uno board with onboard esp8266 and lever converter. Both atmega and esp reside on the same board, they are serially connected one to the other and can be programmed individually via USB using selection jumper.

In my opinion, this is the final solution!

StuartB4, 2018-07-26:
Ok Mikalay31.
My mistake, thanks for explaining it.

Mikalay31, 2018-07-28:
Where is this address recorded? "http: \ 192.168.4.1"
Can try to change it?

Mikalay31, 2018-07-28:
I completely reinstalled everything.
Downloaded the telnetserver sketch - nothing changed (
At the same time ESP8266 occasionally blinks blue LED and clearly works.
It seems to me that something is wrong in the connection.
The network password tried to change. Then I brought it back.
How to untie the ESP8266 from a Wi-Fi network? So you can connect to it again?
The modem in the settings shows the connection to 192.168.100.12 but it is not active.
http://prntscr.com/kc2gcv

The program does not find this address at all (
http://prntscr.com/kc2gg6

Nicholas.

arkypita, 2018-07-28:
If esp cannot connect to wifi it goes in "programming mode" and expose its own wifi network that you can connect and go to 192.168.4.1to configure the connection to your router.

Vice versa if esp connect to your wifi, it does not expose its own network and act as client, acquiring an ip address given from your router.

This is how should work.

If working different or not working at all I cannot help you 'cause i have not write those part of code, so i cannot guess causes

arkypita, 2018-07-28:
Ps: please paste image directly in your post instead of linking

Alt+stamp or win+stamp and then paste (ctrl+v) on github textbox

Mikalay31, 2018-07-30:
Hey.

I understood the principle of work. Thank you.
But something went wrong. Also it is impossible to find the reason.

I'm trying to remove Arduino and repeat everything again.

I just dragged the photo into the dialog box and it was inserted.
Now it will not work.

Stump - what is this button on the keyboard?

I know the combination of Ctlr + C and Ctlr + V, but it does not work.

So I added a link ...

I also wanted to edit my comment which is higher, but I can not do it. Do not understand why. I do not have access rights?

You understand me ?

Nicholas.

arkypita, 2018-07-30:
> I understood the principle of work. Thank you. But something went wrong. Also it is impossible to find the reason.

I know, I also had to make many attempts before having a stable situation, but any situation is diferrent and I have no hint for you. Just try and re-try

About inlining image into post, simply drag and drop image file in github textbox OR capture screen with alt-stamp (a windows shortcut that acquire screenshot in clipboard) and paste.

Stamp or print-screen is this key:

[image]

Dont' worry, it's just a suggestion to make your future messages more readable!

>I also wanted to edit my comment which is higher, but I can not do it. Do not understand why. I do not have access rights?

I dont' know, it is a github problem or limit I think. Theorically it should be possible. I'll check if I have some options for that.

> You understand me ?

Perfectly!

Mikalay31, 2018-07-31:
I still can not post photos in the message. When I try to do what you said, I just open my photo. To return to the message, you have to press - back - to return to the message.

Mikalay31, 2018-07-31:
http://i.prntscr.com/e-FLd87iTAeOVVRxHo72rw.jpeg
http://i.prntscr.com/hHSh3lPwROuF4eG2uBd86w.jpeg
http://i.prntscr.com/qvhiceZdQyu1Qit5qGj8YQ.jpeg
http://i.prntscr.com/1Z5qJfkuRzagLhMm5pCP3Q.jpeg

Mikalay31, 2018-08-05:
Good afternoon !
I am fine.
I completely replaced the Arduino nano and the ESP8266 module.
Everything was done again - and the second day the flight is normal.
Then I replaced the arduino nano with the old one - everything is working fine.
So the problem is clearly in ESP8266.
But at the same time both ESP8266 modules I used are well-programmed and work in test mode. With other sketches.
I did not understand the reason.

Maybe the module ESP8266 - is afraid of overheating? I just soldered to her wires. Can the problem with overheating with a soldering iron?

Photos at https://github.com/ are never downloaded.
The customer service is silent.
StuartB4 - thanks for the link, but it does not work (

I took a video and made a video setup instruction. But I have not edited it yet. Link to the video will be discarded later.

Nicholas.

Will be added later.

Mikalay31, 2018-08-06:
Now your video works) Thanks!
But I still do not insert pictures (

Mikalay31, 2018-08-06:
Here is the video I promised. You can include captions in English.
https://www.youtube.com/watch?v=TeUozD0zQp8

closed 2018-09-30 by arkypita