arkypita/LaserGRBL #401: Parametric variables

RmnJn, 2018-07-25:
Hi, 
I would like to create some custom buttons on LaserGrbl like a square around my picture (I found the [program](https://github.com/arkypita/LaserGRBL/issues/306) - Thanks Vankes) but I would like to make a program which will draw a square around my picture but with a space all around. 

To do this I programmed my button like this and it's work, I draw a square around my picture 5mm larger on each side  : 
...
G0 X([left]-5) Y([bottom]-5) F500
M3 S250
G1 X([left]-5) Y([top]+5)
G1 X([right]+5) Y([top]+5)
G1 X([right]+5) Y[bottom]-5)
G1 X([left]-5) Y([bottom]-5)
M5
G0 X0 Y0 F3000
...
But now, I dont want to change all the "5" on my code one by one, I would like to use a local parametric variable. 
I read on some topics that the locals variables are writing like #1 to #33 on Gcode language so I try to write this in my code : 
...
#1=5
G0 X([left]-#1) Y([bottom]-#1) F500
M3 S250
G1 X([left]-#1) Y([top]+#1)
G1 X([right]+#1) Y([top]+#1)
G1 X([right]+#1) Y[bottom]-#1)
G1 X([left]-#1) Y([bottom]-#1)
M5
G0 X0 Y0 F3000
...

Unfortunately, this code it's not working, I think we don't have the access to write some locals variables.

Would someone have a solution to my problem?

Have a good Day
RJ

arkypita, 2018-07-26:
Variables is not natively supported by grbl firmware, so LaserGRBL can do nothing with them.

[image]

I can add some kind of user variable management in my LaserGRBL in future, to emulate this feature in LaserGRBL.

Mikalay31, 2018-07-26:
It would be great to create a separate topic where additional buttons with a text G-code file will be reset.
Everyone can add their own version of the buttons.
You can share your experience.
I am also very interested in this issue.

Nicholas.

arkypita, 2018-07-26:
https://github.com/arkypita/LaserGRBL/issues/306

RmnJn, 2018-07-26:
Thank you very much for your quick answer. 
I didn't read this notice, sorry.. 

For the moment I will make an excel VBA who can change the code to draw the border with a parametric variable an with the choise to make a circle, a rectangle or a square all around. 
I will post it on the #306 as soon as I will finish it. 

Thank you for your super program and good luck for all the tasks in your roadmap !!

closed 2018-10-13 by arkypita