arkypita/LaserGRBL #402: LaserGRBL Has Slowed Down.

StuartB4, 2018-07-29:
I thought it was my machine at first but it does it without my machine connected, just running LaserGRBL on it's own. Version 2.8.50. I've not changed anything since last time I used it.

When I do a Vector it runs at the correct speed.
But when doing a photo in Line to Line mode the engraving does'nt run at the correct speed.

Usually a picture like this takes between 45 minutes and 1 hour.

[image]

As you can see the Engraving Speed it set to 4000 mm/min.
Even if I save the GCode program and edit the F setting to a higer number and load it back in it still runs at the same speed.
The image is only 35 x 38 mm's. But it's running really slow and the Projected time to engrave has gone up to 7 hours. 
Video here:
https://youtu.be/e1hoyH2mte0

Sorry for the vdeo quality, my phone camera has a problem.

StuartB4, 2018-07-30:
I don't believe it.
For some reason and without me changing anything in LaserGRBL or in GRBL settings, $32 (Laser enable mode) has gone back to 0 instead of 1.
Changed it back to 1 and everything is back to normal.

closed 2018-07-30 by StuartB4

arkypita, 2018-07-30:
I would have bet on it!

Only this value is changed? Maybe you click on a your custom button with $32=0? Or have launched another program that could have set $32=0?

StuartB4, 2018-07-31:
> Only this value is changed? Maybe you click on a your custom button with 

> $32=0?

Yes only $32. I think it may be from clicking custom button for Laser on / Laser Off.
I did use it to focus after using engraving a thicker material.