arkypita/LaserGRBL #403: Two passes with horizontal and vertical engraving problem!

jranavaya, 2018-07-30:
Hi,

I really like LaserGRBL but one thing is causing me some trouble. I want to have two passes on an engraving, one with horizontal engraving and the second with vertical engraving. 

I am currently manually importing the image again after the first pass with exactly the same settings apart from changing horizontal to vertical. Problem is that the image shifts slightly on the second pass.

Is there a way to automatically have LaserGRBL do two passes in both horizontal and vertical methods?

Thanks!

gisep, 2018-07-30:
hi, you can load the first (horizontal) and save it in gcode without launching in, then load the second (vertical) and append the first one, launching both at the same time

arkypita, 2018-07-30:
@gisep give you a good hint

btw i thing that you have an issue on your hardware that produce different offset on x-y caused by different belt tension or slipering that cannot be corrected by LaserGRBL

Or maybe a problem of aliasing between requested resolution and realiable resolution.
What resolution did you set for line2line and what is your $100 and $101 configuration?

jranavaya, 2018-07-31:
I'll certainly check my belt tension as I think you may be right.

The resolution I currently have in line2line is 10. I've tested a few others too. 

My $100 and $101 settings are set to 100. This is the setting I've been advised to use since purchasing this engraver.

closed 2018-08-27 by arkypita