arkypita/LaserGRBL #406: Button for turning the laser on.

1estrider, 2018-07-31:
I was wondering if there would be possible to have a button that turns the laser on manually with 1/4 of the power (or something like that) so I can position my work and also adjust my laser beam?

StuartB4, 2018-07-31:
You can make a custom button.

I use GRBL 1.1 which has $32 set to 1 to enable laser mode.
This can be set from within LaserGRBL.
Click the Grbl menu (top left) and select "Grbl configuration"
look down the list for $32, move across and change the value from 0 to 1.
Then click the "Write" button at the bottom. That will save the setting in GRBL.

LaserGRBL 2.8.50 lets you use 1 button to do 2 things.
Just right click on an empty space on the bar with the buttons on at the bottom
and click "Add Custom Button"

[image]
 
This is mine to switch laser on and off with one button.

[image]

GCode.
$32=0 switches it out of laser mode.
M4 switches laser on to allow you to focus.
S255 is the laser power set to maximum, a quarter power is approx 64.
But you can set it to anywhere between 0 lowest and 255 highest.

GCode 2.
$32=1 switches back to laser mode.
M5 switches laser off after you have finished manually moving/focussing.

1estrider, 2018-07-31:
Thank you so much! This helps so much!  I ended up setting on S20, 1/4 power was still burning the mark on the cardboard.

closed 2018-08-01 by 1estrider