arkypita/LaserGRBL #409: When Stop the job,only x0,y0.

RDS356, 2018-08-05:
Hi...
My suggestion for new release;
1 - When job was stoped, not make g28, just g0x0,y0.
2 - Open dxf files.
Regards,
Reinaldo de Souza - From Brazil

https://www.youtube.com/watch?v=JT7A6z4mkQ0

arkypita, 2018-08-27:
> 1 - When job was stoped, not make g28, just g0x0,y0.

Why?

> 2 - Open dxf files.

Already in dev roadmap #64

RDS356, 2018-08-27:
1 - When job was stoped, not make g28, just g0x0,y0.

Why?
In my thought is not necessary, if you stop the job, the machine must back to you work area.The mach3, bcnc and others are like this.
I had a little accident when i stopped my first job and the machine backed to homming origin.

arkypita, 2018-09-03:
I read [G28 documentation](https://reprap.org/wiki/G-code#G28:_Move_to_Origin_.28Home.29) and I agree with you that this is not the correct code to send. For what I see Grbl react to G28 going to 0,0 without doing a homing procedure, but the behaviour if G28 is not well defined so different interpreter can act differently.

The reason i choose G28 instead of G0 X0 Y0 is that G0 is affected by the current motion mode that is active (realtive or absolute) that is something I don't know when send this command, so I should have push a command to force absolute movement with the side effect that this mode may change the behaviour of the code when you restart your job.

In my opinion the abort button should not exist, because its behaviour is not so clear.
I prefer to use pause button (red hand, bottom left of UI) and grbl reset, but a lot of people ask me for a abort button.

closed 2018-09-30 by arkypita