arkypita/LaserGRBL #410: How do you change the origin for the project?

beardedheathen, 2018-08-06:
Sorry don't know where else to ask. As far as i can tell i've got the grbl stuff set up right. It homes correctly for the origin being in the top left but when i import gcode into LaserGRBL it has the origin in the bottom left and tries to go that way directly into the end stop.

dfox1787, 2018-08-09:
I also want to know this

arkypita, 2018-08-11:
Set grbl conf with $$ to invert motor direction. Google for "grbl configuration"

dfox1787, 2018-08-11:
and allows me to set origin to the centre of a project? or top right?

beardedheathen, 2018-08-12:
Oh i did figure out that you can hit the globe button to set to zero.
@dfoxy that helped fix my problems.

dfox1787, 2018-08-12:
But that just zeros the x and y. 

Im wanting to know how to set work coordinates

mayhem2408, 2018-08-12:
If your machine coordinates have 0,0 in the top left, this means botton left must be in the negative. You need to zero you work coordinates  to the zero you want. You need to jog to the bottom left and zero the Y axis to set the work coordinates 0,0 in the bottom left. If you want 0,0 to be somewhere else, just jog there and zero the work coordinates again.

dfox1787, 2018-08-12:
i know how to zero the machine. I have cnc and when i create a job i can choose where the start point is. 

Id like to do the same in grbl. Set wcs to center for example.

thank you

mayhem2408, 2018-08-12:
@dfox1787 Machine coordinates and work coordinates are two different things. I'm obviously not understanding what you are trying to do.

dfox1787, 2018-08-12:
:)

closed 2018-09-30 by arkypita