arkypita/LaserGRBL #414: Sound when finishing a lob

christos-xithalis, 2018-08-10:
ok, this is a very simple request: the program plays a sound, custom or predetermined, to let you know when it has finished a job.
Thank you!

arkypita, 2018-08-11:
Already in dev roadmap

closed 2018-08-11 by arkypita

gerdvogel, 2020-05-01:
one question: when will the desired function be available?
I have a lot of wooden discs to laser and I am still working on other projects in my workshop on the side and sometimes I forget the laser and I don't get finished with the laser in time, if there would be a sound I could interrupt my active activity and put a new disc on the lasers.

gmmanonymus111, 2020-05-01:
My version now has a limited sound support, if you want, you can build that. I want to add the ability to change sounds and volume before opening a pull request, that's why it is delayed

gerdvogel, 2020-05-01:
Thanks for the answer.