arkypita/LaserGRBL #417: Am i engraving wrong?

dfox1787, 2018-08-12:
I might be getting the wrong end of the stick here.

But is the software telling me the sections in red are meant to be getting lasered?

see photos

http://s1239.photobucket.com/user/foxinola/library/Laser

rolx95, 2018-08-12:
Make sure your laser is focused. Go with one S-MAX (I use 230) and try lower border speed something like  a 1000. Looks like MDF there, 1000 mm/min should do enough, I have 2,5W laser and that is sufficient, with my 2,5W I for this picture into MDF I would go with around 1400-1500 mm/min on S MAX 230. All of this is your know-how, trial and error, try something easier and less time consuming picture first, to get a feeling for your wattage. First pick a logo and go with vectorize without filling, after the right burn then add filling, after that try the dog.

dfox1787, 2018-08-12:
Yeah it is MDF. If i laser a black and white image it works fine a looks good. As soon as i want to laser something that is grey scale it just burns black. 

See new photos

Mine is a 2.5 laser

http://s1239.photobucket.com/user/foxinola/library/laser2

mayhem2408, 2018-08-13:
@dfox1787 What Laser and Driver are you using? Can it handle the PWM signal used in grayscaling.

dfox1787, 2018-08-13:
Good question  after looking at my board im not convinced it is even setup for ttl. 

However does anyone know if it can? I read on some other posts that the laser module needs to be ttl compatible. How would i check if it is? 

Here is a pic of my main board and pin outs. 

http://s1239.photobucket.com/user/foxinola/media/Laser/Capture_1.png.html

mayhem2408, 2018-08-13:
What version of GRBL does that board support. It must support V1.1 for PWM support. If it's an older board running 0.9 or earlier, it will not support PWM output.

dfox1787, 2018-08-13:
Its grbl 1.1. 

Im getting very strange results when im testing. Enabling the Laser mode on the firmware gives me faster and better quality results. (using M4) 

Using M3 makes the engraving projects take 4 hours. M4 takes it to 14 mins. 

Ill share a link shortly with settings i have used

mayhem2408, 2018-08-13:
@dfox1787 I don't know why M3 and M4 are giving you different times. It only changes the was the power is controlled on moves. It looks like that board is not using TTL to drive the lasers power. It looks like it is directly driving the power of the lasers driver board. TTL requires the laser to have constent power, and then a TTL control signal can quickly turn the laser on and off.

dfox1787, 2018-08-13:
@mayhem2408 Its very odd. If i use M3 then it burns too much. If i enable M4 i get ok results. 

See vid and image below:

http://s1239.photobucket.com/user/foxinola/media/laser2/20180813_220141.jpg.html

https://www.dropbox.com/s/rmn2vxv6kmt44fe/My%20Movie.mp4?dl=0

arkypita, 2018-08-24:
Is it correct that laser mode give faster result because grbl was first developed for CNC where changing power it is in effect a change of speed of carving tool. This change of speed require some time (delay) to be effective, so grbl wait some milliseconds each change of S value.

As because with laser the change of power is instantaneous, with laser mode this changes are executed without delay by hardware.

https://github.com/gnea/grbl/wiki/Grbl-v1.1-Laser-Mode

[image]

closed 2018-09-30 by arkypita