arkypita/LaserGRBL #421: Dark lines

kimbo001, 2018-08-19:
Hi

After some trails i am giving grey scale a try but i get dark lines on the borders. like the laser switch on with a higher power and regulate to a lower power, if i burn horizontal i get them and the horizontal borders, vertical on the vertical borders and diagonal i get them on al four borders

2.5mW laser on Mana SE clone laserGRBL software GRBL 1.1f

StuartB4, 2018-08-19:
Have a look here:
https://github.com/arkypita/LaserGRBL/issues/373
and here:
https://github.com/gnea/grbl/issues/34

kimbo001, 2018-08-19:
Thank you

How do i change the M3 to M4 it does not appear in the drop down

StuartB4, 2018-08-19:
What version of GRBL do you have?

StuartB4, 2018-08-19:
Do you have Laser mode enabled in GRBL configuration?
[image]

StuartB4, 2018-08-19:
Click the Write button on the bottom of the GRBL config window, where you changed it.

StuartB4, 2018-08-19:
Just a thought, do you have the 2 settings just above the Laser enable setting the same as mine. $30 and $31.
They should be $30 255 and $31 0
That sets the laser power for PWM mode. Min power = 0 and Max power =255.

StuartB4, 2018-08-19:
Can't think why at the minute, do you have your laser focussed correctly.

kimbo001, 2018-08-19:
I will double check it. For some reason my $32 is not saving, all my other settings save but not that one, i even exported it, opened with notepad, made the changes and re-upload again but still not saving

kimbo001, 2018-08-19:
Apologies for asking , but i never done a hex file in Arduino IDE can you help me please

StuartB4, 2018-08-19:
Sorry, I don't use the Hex file.
I upload the GRBL source code (Version 1.1f) from the zip file here:
https://github.com/gnea/grbl/releases

This tells you how to do it but it's a couple of years old now and mentions version 0.9j, but you should get the Version 1.1f zip.
https://github.com/grbl/grbl/wiki/Compiling-Grbl

Then change any settings using the Serial Monitor in the Arduino IDE and re-compile it and re-upload it to the Arduino.

kimbo001, 2018-08-19:
Here is the vid of the app not saving https://youtu.be/i5VK8_LPmKA
Maybe you can see what i am doing wrong

StuartB4, 2018-08-19:
Oh yeah. It saves it until you actually connect to the com port, thats strange.
You seem to be doing everything correctly.
I hope you dont mind but I just opend an Issue on the GRBL site to see if anyone else has any ideas about what is happening, I included the video link you supplied.
Now we just have to wait to see if anyone replies.

One option you could try is to remove the Com port and let Windows reinstall it.
I just tried it and once it's removed it reinstalls it again.
See here:
https://youtu.be/cccKdNG8oYQ

kimbo001, 2018-08-19:
No Problem

I will do everything from start, uninstall the driver and then reinstall and see what it does

kimbo001, 2018-08-19:
Nope still get the same result

StuartB4, 2018-08-20:
Oh yeah. Haha, can't believe I did'nt think of that. Glad thats sorted out.

arkypita, 2018-08-27:
> Here is the vid of the app not saving https://youtu.be/i5VK8_LPmKA
> Maybe you can see what i am doing wrong

Maybe you can have a look at the clear red message that say "read and write enabled only when connected and idle"?

[image]

closed 2018-08-27 by arkypita

kimbo001, 2018-08-27:
Hi

Did you look at the whole Verdi?

 did connect it before i saved

when i reconnect after the shut down it did not save

arkypita, 2018-08-27:
Sorry, i was superficial. Now i have seen the whole video.

> In settings there is a soft reset ticked by default, untick and now it saves

Sound strange to me: soft/hard reset configuration should not affect grbl configuration at all.

I have tried the same sequence and saving is working for me, with and without soft reset.

kimbo001, 2018-08-27:
Yea i get sometime it saves some time it does not save, i checked the hardware nothing weird i even tried other firmware i still get the same. I am trying to eliminate anti virri software but time is not on my side. Very nice software BTW i really like it

kimbo001, 2018-08-27:
Done this with a 500mw laser at F1000
[image]