arkypita/LaserGRBL #423: Feature Request - Delete Single Path

corvette100377, 2018-08-21:
Ability to delete specific or single path prior to burning.
This would allow vector images to have more finite control.
i.e.: You want to cut just a circle, but because the vector conversion created 2 paths (an inside circle and an outside circle from tracing the line) it creates 2 cuts.

arkypita, 2018-08-22:
https://github.com/arkypita/LaserGRBL/issues/399

If you want to cut a circle vectorize this image:

[image]

This will produce a single path.

Centerline tracing is already in #64

closed 2018-08-22 by arkypita

corvette100377, 2018-08-27:
I just read through your comments on the centerline tracing issue. The ability to select individual paths on the main preview screen and delete them may be a good compromise since you cannot locate a good centerline detection code in C++. I am not a programmer, so forgive my ignorance.  This would at least get the desired effect close to the desired result.  I understand if this is not achievable in C++. Just trying to contribute my thoughts on the matter. I am not sure how hard it would be to code the ability to select the individually rendered vector paths.

arkypita, 2018-08-27:
The feature of "select individual paths on the main preview screen and delete them" is not possible to implement at all. The graphic code/library I use does not allow in any way to "go back" from a clicked pixel to the segment, and from the segment to the path that generate it.

Also it is not easy to add zoom on preview, so complex image could be difficoult to click. Moreover if the design is complex, a manual edit to delete double path could be very, very annoying.

arkypita, 2018-08-27:
The only "centerline tracing" library available on-line is the one you can test here:

https://online.rapidresizer.com/tracer.php

Here is a simple benchmark:

[image]

[image]

[image]

As you can see, as soon the image became a little complex, the centerline algorithm fail,

arkypita, 2018-10-21:
Hi all!

I have released a new version of LaserGRBL that include **centerline vectorization**.
This feature was requested by a lot of user and complete LaserGRBL with a "must have features" for all who need a simple way to engrave images.

This release was released as a pre-release so it is not automatically updated. 
To those who want to test it I recommend downloading the installer from: [here](https://github.com/arkypita/LaserGRBL/releases/download/v3.0.0/install.exe)