arkypita/LaserGRBL #430: Trying to change image size

yobosaeyo, 2018-08-26:
I am doing baseball bat logos and I am running into an issue where I am unable to increase width without increasing height. I am limited due to bat being round and have figured out that 1.5 inches is all I can go with my logo. When I try to do a players name at the same 34mm height it only allows for a width of 49mm. If there was some way to change this to say 60mm without changing the height it would be great. As it is I usually have to do the name as one project then the logo as a second one. Setup would be so much easier if I could do the logo and name in the same session without having to re set up everything.  Any help appreciated.

yobosaeyo, 2018-08-26:
[image]

StuartB4, 2018-08-26:
What software are you using to make the logo's and name text.
Surely you could use something like Inkscape to do both as one document / job in the sizes you need.
Then use it to generate the gcode, and load / import that code in to LaserGRBL.

yobosaeyo, 2018-08-30:
I am using inkscape, just not sure how to save it in inkscape with the dimensions I am looking for. I have only been working with this a short amount of time.

StuartB4, 2018-08-30:
This is a good place to look.
You can download the gcode plug-in for Inkscape and see how to create your images/text etc.
Tutorial is here:
http://jtechphotonics.com/?page_id=2012

GCode Plug-in download is here, the one you need depends on the version of Inkscape you have
https://jtechphotonics.com/?page_id=1980

yobosaeyo, 2018-09-01:
I was on version 48. I will install the latest and see if I can get it going.

yobosaeyo, 2018-09-01:
btw, how do I install the plugin you listed?

yobosaeyo, 2018-09-01:
I installed the latest version and the plugin. The issue I see now is that the text shows up but not the image.

yobosaeyo, 2018-09-01:
updated my inkscape and when I save the file as you suggested it misses the image and only shows the text. I tried multiple times. I did the trace image and deleted the original and same thing. I attempted to make it in the size I needed. I even tried loading the svg on makercam.com and it too only shows the text.

StuartB4, 2018-09-02:
Oh dear. Try this way instead.

1. Load your logo image in to Inkscape by clicking File menu, Import.
2. Position / resize image where you want it, then create text. 
3. Select just the text and group it by clicking the Object menu, Group.
4. Then select image AND text the same way. You can select one of them and hold Shift key and left click the other to select both.
5. Select File menu, Export PNG image. A window pops up on the right. Select Export As and give it a name. Click Export and then you can load it in to Lasergrbl as a PNG file.

yobosaeyo, 2018-09-03:
Got it all to work, now I am having an issue where the program will randomly stop the laser but and keep running commands while it says laser idle. Sometimes the laser is on and some off. This is a new issue for me. I made a rather long project back around mother's day that ran almost two hours without a hitch and now in a project that took an hour ten it stopped four times.

arkypita, 2018-09-04:
Look like a comunication issue. See http://lasergrbl.com/faq for possible solution

yobosaeyo, 2018-09-05:
Yeah, I messaged the place I bought from on Aliexpress and the are sending a new board. They have been most helpful and I would recommend.

StuartB4, 2018-09-05:
Thats good to know, never used them before.

closed 2018-09-30 by arkypita

yobosaeyo, 2018-11-24:
Hi, I got my new board and I am seeing that with your latest updates I am not getting clear text, it looks a little over burnt. Is there a way to turn down the power? I have the laser approx 2-3 inches away from the project.

StuartB4, 2018-11-24:
You could try increasing the burning speed in the Target Image Settings.

[image]

yobosaeyo, 2018-11-25:
I can try changing the speed to 1000 instead of the 800, I am not sure what the pwm setting is. See my pics of what I am sending and what i am seeing. Thanks
Kevin
[image]
[image]
[image]

yobosaeyo, 2018-11-25:
the bat has turned into a practice printing bat.

StuartB4, 2018-11-25:
>  I am not sure what the pwm setting is.

The PWM setting is the S-MIN and S-MAX setting in the Target Image settings in the image you posted above. It's the Laser power minimum and maximum.

If you change the S-MAX to 128 it will burn at half power as 255 is maximum power.
64 will burn at a quarter power etc. You can set it to any number inbetween 0 and 255 to adjust the laser power.

If you set it in GRBL Configuration settings to a lower number than 255 and store it using the Write button at the bottom of the window, Laser GRBL will still show S-MAX 255 in the Target Image settings, but it will put a red box around the number, so you know there is a problem. Just change the number S-MAX 255 in the Target Image settings to match the number stored in GRBL settings $30 and the red box will dissapear.

[image]

In LaserGRBL click on the GRBL menu at the top left of main window and click Grbl Configuration.
When it opens have a look at $30.
It says Maximum Spindle Speed, but when using a laser instead of a spindle motor it is Laser Power Max.
So if you change the number on that line (usually it's 255) it can't burn any stronger than that setting.

If that number is anything other than 255, say 1000, then setting it to 500 will be half power, 250 quarter power etc. That can be set the same but to any number between 0 and 1000.

yobosaeyo, 2018-12-09:
The final fix was increasing the speed, I tried lowering the power but it was still over burning. Thanks for the support.

StuartB4, 2018-12-09:
Hi yobosaeyo.

Glad to see you sorted it out.

yobosaeyo, 2018-12-09:
thanks again, can finally cross this project off my list!

Bazza1959, 2019-04-25:
I am having a problem where in inkscape the picture is 200mm x 200mm but when I then open it in laserGRBL its trying to burn a picture thats 759mm x 759mm. I am obviously doing something wrong, just not sure what