arkypita/LaserGRBL #431: Converting/creating files in to .nc or Gcode

1estrider, 2018-08-27:
I am at loss...trying to create some letters on the plaque. The real size of the laser work area is about 8x4mm or 3.25x1.5 inches. However all i get from the laser is a speck! How can I make the laser to burn desired size of the project? 
I am using Windows publisher to create files. Tried with pdf, and jpeg file formats.
Also, what kind of programs are you all using to create projects?

arkypita, 2018-08-27:
If output size is different from gcode size you should configure your grbl board params $100, $101 and $102 – [X,Y,Z] steps/mm

https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#100-101-and-102--xyz-stepsmm

closed 2018-08-27 by arkypita

StuartB4, 2018-08-27:
You seem to have your measurements wrong, 8x4 mm is not 3.25 x1.5 inches.