arkypita/LaserGRBL #432: Custom button - Framing

kimbo001, 2018-09-01:
Hi

I am not a GRBL coder can somebody help me i want to frame my work but it must have a 10mm offset

StuartB4, 2018-09-01:
10 mil X and Y ??

If so this should do it.
Create a custom button by Right clicking on an empty part of the bar at the bottom of Lasergrbl window, where the little man and hand icon.

Copy and paste this as it is in to the GCode box:
F100
M4 S255
G0 X10 Y10
G0 X[left] Y[bottom]
G1 X[right]
G1 Y[top]
G1 X[left]
G1 Y[bottom]
G0 X0 Y0
M5

Add a Tooltip, that just pops up when you move the mouse over the button so you know what it does.
Click Save button.

The F100 is the Speed. Change this to suit you.
M4 is Laser On. The S255 is the Laser power, change that to suit yourself between 0 and 255.
The G0 X10 and Y10 are the Offset setting.
If you just want X10 offset then leave out the Y10 and the same if you just want the Y10 offset, leave out the X10.

[image]

kimbo001, 2018-09-02:
Thank you very much :)

StuartB4, 2018-09-05:
Thats a better option than the one I posted.
You get a frame 10mm away from the image, my option only framed 10mm away the home position
but right up to the image.

StuartB4, 2018-09-06:
> continue jogging possibility ?
Please explain what you want to do.

manisaranbala, 2018-09-07:
X Y Z AXIS CONTINUE MANUAL KEY PAD MOVE

arkypita, 2018-09-16:
@manisaranbala 

Continuos jogging is not possible because grbl protocol does not support this. Because jog command are queued a user can easily queue up more motions than he want without realizing it and move well-past desired location.

P.s. please do not post "ad mentula canis" under other user thread. If you have a question or an issue, please open a new issue for your request.

arkypita, 2018-09-16:
@kimbo001 

> I am not a GRBL coder

GCode is very simple. Google for some tutorial www.google.it/search?q=gcode for beginner
There is a lot of video on youtube too.

closed 2018-09-16 by arkypita

manisaranbala, 2018-09-16:
https://github.com/gerritv/Grbl-Panel/tree/master/RepeatButton