arkypita/LaserGRBL #434: Why does LaserGRBL trace the outside of a line when Vectorizing?

montani76, 2018-09-01:
Every time I import a PNG file converted from DXF, just circles or straight lines, LaserGRBL traces the outside of each line on both sides not giving me a true line cut with my engraver. I can work with the brightness, contrast and B/W settings to minimize the doubles lines but always gives a skewed engraving not fluid lines. I am trying to cut gaskets and cannot seam to get the software to work well cutting converted dxf to png files from Inkscape.  Any suggestions? Be nice if Laser GRBL would import DXF files.

arkypita, 2018-09-02:
https://github.com/arkypita/LaserGRBL/issues/423
https://github.com/arkypita/LaserGRBL/issues/424
#64

closed 2018-09-02 by arkypita

tete87, 2021-08-16:
Hi. I'm trying to add dxf files on lasergrbl but I can't. Or it's the program that dont have that function? Can anybody tell me how to do it please