arkypita/LaserGRBL #436: Ricky23

Rick23n, 2018-09-05:
I have a chinese 16w Laser Engraver which came with Benbox to install which worked initially with NANO(324p)/LX-Nano with a CH340 chip but now the command buttons will only move the Y axis one way or activate the laser on full power.I have now installed LaserGRBL v2.8.52 which seemed to install correctly but when I import a file to print I cannot get the 'connect' button to activate. It tries to connect but then shows "disconnected" and the Session log reads:
5/09/2018 6:54:45 PM	SetStatus   	Machine status [Connecting]
5/09/2018 6:54:45 PM	OpenCom     	Open COM4 @ 115200 baud Ctrl-X
5/09/2018 6:54:45 PM	SendImmediate	Send Immediate Command [0x18]
5/09/2018 6:54:49 PM	SendImmediate	Send Immediate Command [0x18]
5/09/2018 6:54:55 PM	OpenCom     	Connection timeout!
5/09/2018 6:54:55 PM	CloseCom    	Close COM4 [CORE]
5/09/2018 6:54:55 PM	SetStatus   	Machine status [Disconnected]
5/09/2018 6:55:24 PM	Program     	------------ PROGRAM STOP ------------
Please help

gisep, 2018-09-05:
I bought a similar one and when trying to run it from LaserGRBL I had problems. With grbl firmware the motors became hot and it didn't run well, then when tried to put original firmware to run again the chinese software benbox i didn't find the firmware on the net and the seller told me it wasn't opensource.
I finally bought an arduino UNO with cnc shield, that's a lot better; the old shield neither had PWM.

Rick23n, 2018-09-06:
I reinstalled Benbox too and now all the controls are messed up. The Y axis will only move one way and the burn button will only burn in one continuous line and the only way to stop it is to turn off the power to the laser!

Rick23n, 2018-09-08:
I have flashed my chinese Nano328 with the 1.1GRBL and all works fine (would NOT flash with version 2.8) My question now is can I upgrade to V2.8.35 and still have functionality of the Arduino Nano328? My reason to upgrade is the laser will not engrave the first (or bottom) line of my image. I thought the upgrade might overcome this as suggested in post 22nd Dec 17 by arkypita.

StuartB4, 2018-09-08:
This is from the FAQ page for LaserGRBL.

> LaserGRBL work with any engraver that use Grbl as controller. This also include a lot of mounting kit from ebay, bangood, aliexpress, amazon, based on arduino uno/nano or other customized boards with ATmega328P chip.
> 
> Often some of these markers do not use Grbl but they use other control firmware/protocols (benbox or others) but is it possible to reprogram their firmware to use Grbl. You must google about how to do that with your specific controller.

So you should be able to upgrade to LaserGRBL 2.8.52, the latest and it should work.
Providing your GRBL settings are correct.

Rick23n, 2018-09-10:
10th Sept. Laser is now working with 2.8.52 but I do still have an issue when engraving 'horizontal'. It will only engrave the vertical lines. I am thinking it may be in the GBRL configuration setting.; Its ok if set to 'diagonal' but unfortunately this changes the size of the image when it's engraved (frustrating). Vertical is not an option either because then it misses all the vertical lines. I thought it also might be speed and power but if I set any higher, then any scrolls or words are burnt too much. My speed setting is 2000 and power 200 which engraves correctly except for the horizontal lines. Any suggestions????

StuartB4, 2018-09-10:
Can you paste a copy of your GRBL settings.
Just goto GRBL menu in LaserGRBL and click Grbl Configuration. That should show the list of settings.
Stretch the window down so it shows all the settings.
Press Print Scrn on your computer and then open MS Paint and click Paste.
You can draw a box just round the settings part of the image and Cut it, then click New. 
Click Dont Save, then click Paste and Save the image and post it here by clicking on "selecting them" at the 
bottom of this box.

Like this:
[image]

Rick23n, 2018-09-11:
[image] StuartB4 -  if you can view this file you will see that it is missing $122=10.000 m/sec^2, $130=230mm,$131=285mm,$132=200mm(I only have X & Y at the moment) This is because my laptop screen isnt big enough to show all at once. I do not have MS Paint and my anti-virus will not allow me to download it so I have done the conversion in Paint.Net

[image]This image shows what is happening to my engraved image.

[image]This image shows the setting I am using to send to the Arduino Nano328p

Thank you for your interest I really appreciate it

StuartB4, 2018-09-11:
You should upgrade GRBL to version 1.1, then you have an extra setting $31 which is Laser mode.

> The main difference between default Grbl operation and the laser mode is how the spindle/laser output is controlled with motions involved. Every time a spindle state M3 M4 M5 or spindle speed Sxxx is altered, Grbl would come to a stop, allow the spindle to change, and then continue. This is the normal operating procedure for a milling machine spindle. It needs time to change speeds.
> 
> However, if a laser starts and stops like this for every spindle change, this leads to scorching and uneven cutting/engraving! Grbl's new laser mode prevents unnecessary stops whenever possible and adds a new dynamic laser power mode that automagically scales power based on current speed related to programmed rate. So, you can get super clean and crisp results, even on a low-acceleration machine!
> 
> Enabling or disabling Grbl's laser mode is easy. Just alter the $32 Grbl setting.
> 
>     To Enable: Send Grbl a $32=1 command.
>     To Disable: Send Grbl a $32=0 command.

You have $110 set to 1000 maximum in GRBL but you have set the engraving speed to 2000 in the Target image settings.
Thats why it has the red box around it, it wont go 2000 when it's set to 1000 in GRBL, you need to change it to 2000 in GRBL.

Rick23n, 2018-09-11:
Thank you for getting back so quickly and yes I have Windows 10 and yes I found paint. 
LaserGRBL v2.8.52 is what I'm using to create the code for the Arduino so I thought that grbl v1.1f was a much earlier version. My mistake. When I downloaded grbl v1.1f it doesn't contain a Windows installer exe file. How and where do I find this. (I am a complete novice at this as you must have guessed by now) OR is this the version I need to flash to the ArduinoNano328p?? I made the assumption that the program that generates the g code was the one that contains the settings $30, $31 & $32. I am confused !

StuartB4, 2018-09-11:
Yeah thats correct, lasergrbl 2.8.52.
What you need to do is download GRBL 1.1f from here:
https://github.com/gnea/grbl/releases
You can just download the .Hex file

Then you download Xloader from here:
https://my.pcloud.com/publink/show?code=XZvPL17ZwQHksJSJSvRKcR4vzvGIwm1jUSQV
It's uploaded to my PCloud account.

Extract the XLoader file to anywhere you like.
When you have both run XLoader.exe from the folder you extracted it to and connect the Arduino to the correct com port, select nano from the dropdown list and set the correct baud rate, should be 115200, browse to where the .hex file is and click Upload.
Thats it.

Like this:
[image]

Then you can edit the GRBL settings from inside LaserGRBL in the list of settings in the image you posted.

Rick23n, 2018-09-12:
Thank you!!! My Laser is now up and running with the new settings added. It would not connect initially so I had to experiment with the baud rate (is 115200) and also the correct Arduino (NANO(324p)/LX-Nano) The image is crisp and it is engraving the first line, a little lighter than the rest but I guess that will be in the speed and laser power settings.
Again I can't thank you enough for your time. I will now switch my attention to building a "Z" axis as my setup does not allow for changing material thickness. I hope there is a build out there that will enable me to use grbl settings for this.

Rick23n, 2018-09-12:
I am still having a problem with the first horizontal line. Have tried different speed and power setting to no avail. The line is still faint compared to the verticals

StuartB4, 2018-09-12:
Hi Rick.
You could try changing this setting in the Settings on the GRBL menu.
Just to see if it makes a difference.
If it's enabled disable it or vice versa.
[image]

StuartB4, 2018-09-12:
Can you upload the original image you used here so I can try it on my machine.

StuartB4, 2018-09-12:
Just out of interest, did you enable Laser Mode in the GRBL settings, if not set $32 to 1 instead of 0.
Also set $30 to 255 (maximum laser power) and $31 to 0 (minimum laser power).
Then change these 2 values to M4 and 255.

Rick23n, 2018-09-13:
Hi Stuart
Tried the MultiDirectional setting - no change,yes I have the Laser Mode enabled also $30 = **9000** and $31 = 0. M4 did not make any change. I don't have the "optical correct" option, was it supposed to be included in grbl1.1f?? So I still have only 25% line image for the first line as compared to 100% for the vertical lines, the border vertical are darker than the internal vertical lines(it appears that the laser burns twice on the change of direction) $110 & $111, I have left at 500 This gives a projected time of approx 1hr. Using "Diagonal"the projected time is about 2 hrs
If I use the "vertical" line to line option, then the vertical lines are affected. (i.e. 25% only) I have left the line quality at 11 L/mm
[image]
This is the image I have created in MS Publisher, saved as a .png file to import to Lasergrbl.
I can get a reasonable image with diagonal set to Pwr2000 Speed 500 and one direction lasering , If I use Multidirection it burns the perimeter lines, and YES I had to restart grbl to activate Multidirection.
Just as a matter of interest Stuart, where are you located? I am on the Gold Coast Queensland

arkypita, 2018-09-13:
Can you also post a picture of the result, to better understand the issue?

arkypita, 2018-09-13:
For this kind of job I suggest vectorize + filling or vectorize only

fantilator, 2018-09-13:
@Rick23n 
Can you make you program again as on the photo where it failed.
Then save the program.
open the program in notepad or something and copy all the G-code in www.NCviewer.com
after this press the <plot> button and examing the drawing on errors.

When I did this with the drawing provided above I see this.

NCviewer:
[image]
LaserGRBL:
[image]

Here the bottom line is indeed not projected, maybe it's to close to the edge.
Try to lift it up a bit in the image.

StuartB4, 2018-09-13:
Hi Rick.
""OpticalCorrect" is in the "Grayscale" drop down box and is part of the Latest LaserGRBL 2.8.52 not GRBL.
It should be in the "Import Raster Image" settings.

Rick23n, 2018-09-13:
The best result yet has been : 
[image]
[image]
[image]
The best I could do with the G Code:
[image]
No Stuart  as you can see I dont have the 'Greyscale" drop down box  and I  re downloaded LaserGRBL 2.8.52 tonight. I have tried to shift the image up but it made no difference. The answer seems to be in getting the Laser power and speed right. Tried "Vectorizing" but only lost lines altogether.
Any thoughts are greatly appreciated

StuartB4, 2018-09-13:
> No Stuart as you can see I dont have the 'Greyscale" drop down box

If you go to File, Open and load an image the "Import Raster Image" box should open.
Grayscale drop down box should be there.

I also noticed your $110 and $111 in GRBL settings are set to 250, but you have the Engraving speed in the Target Image settings box set to 500.

[image]

It wont engrave at 500 speed if you have it set lower in GRBL.
Thats why theres a red box around the Engraving Speed box, to let you know something is not right.
You can set $110 and $111 in GRBL Settings to a higher number, up to the maximum speed your machine can move.
As long as you dont exceed that number in the Target Image Engraving Speed box it will be fine
and the red warning box will not appear.

[image]

StuartB4, 2018-09-13:
$30 should be set to maximum 255, that is the maximum power your laser will produce using from the  PWM signal from the Arduino. It's a 0 to 5volt signal,  0 is lowest power and 5volts is maximum (255)
0 is Lowest 255 is Full power, you don't need numbers in their 1000's.

StuartB4, 2018-09-13:
To see the GCode properly as a text file:
Load an image and get it ready for engraving but dont actually start the engraving process.
Goto File, Save Program.
Give it a name and save it.
Then you can open it in a text editor to view it or edit it as you wish.

StuartB4, 2018-09-14:
Hi Rick. Just a thought, have you tried the same image on a different type of wood? 
All materials react differently to laser and I know some types burn better than others.

Rick23n, 2018-09-15:
Thanks Stuart. The wood is ok. I have partially completed it (I stuffed up) and the image is fine except for the first horizontal line showing approx 50% lighter than the rest of the burn. With regard to the laser power, I saw somewhere on a forum it shows Laser Power 0-255 OR 0-12000 so when I wasn't having any luck I tried 12000 and got a better result but now I have it set to 255 and the result is 'becoming' acceptable. I still have to tweek something to get a crisp image. Thanks for the tip on File saving and yes I did get the engraving speed settings mixed up.
This is a screen shot where I can't see the Greyscale Setting
[image]
The other thing that I have a problem with is Image Size. I dont know how to control the 'vertical' size after I input the 'horizontal' measurement. Even then the figures do not correspond to the actual size.
To get 93.5mm x 274.5mm I have to input 30 x (and it automatically comes up with 90 as the vertical measurement) and it does show size in mm. And the offset I have to input 1mm to get approx 5mm. 
Any thoughts?

StuartB4, 2018-09-15:
Hi Rick.
Dont know why you dont get the Grayscale option when it's the same version as mine, you would have to ask the Arkypita, the developer why. You can just open a new issue about that, he may see it here but may not.

Yeah about the laser power settings. Some people have a 15 Watt laser which is 15,000 milliwatts.
So they set the $30 to 15,000 in GRBL, then when they change the setting in LaserGRBL to say 800 they see it as being 800 milliwatts,  or a setting of 7500 they see as 7500 milliwatts or 7.5 Watts, up to the maximum 15,000 /15 Watts. 
Setting $30 to 255 in GRBL and then setting it to 128 in LaserGRBL is the same thing, 128 would be just about equivalent to half the laser power or 7.5 Watts. 64 would be a quater power and so on.
If you were using a router spindle instead of the laser I think the setting would be for the spindle motor speed, 15000 rpm.

As for the image size problem I'm not sure what you mean by controlling the Vertical size.
Do you mean the Width and Height settings in the Target Image settings?
If you load an image say 200 by 300mm's it should show you that size in the Target image settings W and H boxes. W 200 and H 300.
If you change the Width setting the Height setting should change automatically to keep the image ratio correct.
The only problem you should get there is if you enter a W or H size which is larger than the GRBL settings for $130 and $131. 
If they were set for 150 and 200 and you try to enter 155 in the W or 205 in the H box you would get a Red box around the W or H setting box so you know there is a problem.

[image]

Rick23n, 2018-09-15:
Hi Stuart, my GRBL setting $131 is 230mmW and $132 is 285mmH As shown on the Target Image, to be able to burn the correct size I need to enter 30mm to get an image width of 97.5mm ???? I cannot see any conversion setting to be able to adjust this.
My laser is 16watt which should give me an average 8/11 watts.
[image]
I hope this makes it clearer.

StuartB4, 2018-09-15:
> Hi Stuart, my GRBL setting $131 is 230mmW and $132 is 285mmH

I think you mean 130 and 131, X, 132 is Z axis.

Just realised why you have no Grayscale settings box, it only to appears if you load a colour image like a photo in.

If the image you load in is 30 x 90 then it should show them numbers in the W and H boxes.

This image is 38mm's X 38mm's and thats what shows in the W and H boxes, as it should.

[image]

[image]

If I change number in the W the H changes to match.

[image]

Does the image you load in have a white border around it, or is the image right up to all 4 edges. 
If it has a border round it you can use the crop tool to just grab the image.
Just click the 6th icon along the bottom (Crop tool).
Draw a box around the actual image with the left mouse button held down then release it.
Click Next and see what the W and H numbers are.

[image]

Rick23n, 2018-09-16:
No border ,I made sure of that. To explain in more detail ; if I input 30mmW in the Target Image Size then when I create it, it shows 31mmW x 88mmH and that will produce an image size approx what I need. If I input 93.5mmW( which is the actual size of my imported image)93.5W x 274.5H) it will then it will try to burn an image approx 270mmW x 816mmH which is outside the constraints of the workspace $130 & $131 of 230W & 285H. Is it possible the problem is in the imported image from MS Publisher. Maybe if I learn a little more about Inkscape it may sort it out.

Rick23n, 2018-09-16:
I also see that M3 Laser Mode should allow me to focus the Laser. I cannot turn the laser on manually. The only time the laser is activated is when I start the Burn program. Have I missed something or is it the chinese version of the Arduino that doesn't support this. There is a switch on the laser but it appears to do nothing even with the laser in burn mode. Maybe the answer is to put a switch in from the 5v+ to the laser on very low power for focusing.

fantilator, 2018-09-16:
Try typing  M3 S100 and see what the laser does. M5 stops the laser.

Rick23n, 2018-09-16:
I'm sorry I dont understand. Type M3 S100 where? In LaserGRBL I only have the option to select M3 or M4.
 I dont have anywhere to input a command apart from changing the $ commands in GRBL Config.

Rick23n, 2018-09-16:
Ok , connect the laser and then I will have access to the command line!

Rick23n, 2018-09-16:
I dont seem to be able to adjust the laser power. Set at 5 or 255 is same. Yes I have changed the grbl settings $30 to match. With or without Laser Mode $32 enabled. Also with PWM enabled. ??

Rick23n, 2018-09-16:
Yes 3 pin connection

Rick23n, 2018-09-16:
Thanks Stuart I will work on that

Rick23n, 2018-09-16:
Set up the button all works except for Laser power still 100% full on

StuartB4, 2018-09-16:
You say
>  There is a switch on the laser but it appears to do nothing even with the laser in burn mode.

Are there any marks or is there any writing on the board close to the switch.
I know some of them lasers have a switch for TTL/PWM mode or Low Power mode which allows you too focus and then press it again for Hi Power or burning mode, depending on if the switch is on or off.
Is it a latching switch or just a momentery action, does it lock in to one position when you press it and unlock when you press it again?

StuartB4, 2018-09-16:
Oh ok. 
Seems strange that you can't vary the laser power.

Have you tried loading a colour image in to LaserGBL?
If not, can you try it.
Just load any colour jpeg image.
Set the Brightness and Contrast to suitable levels.
Select Line To Line Tracing.
Set amount of lines to 8 or whatever you want. More lines=higher quality but slower longer burn time.
Click Next.

[image]

Then in the Target Image settings just make the W something like 100, the H should change to the correct size.
My settings for the laser are: 
Engraving Speed: 1000
Laser on: M4
Laser Off: M5
S-Min : 0
S-Max: 255
Click the Create button.
Then burn the image on to whatever material you want/have. Not pure white though, try something a bit off white.
It should do a greyscale engraving.

fantilator, 2018-09-16:
@Rick23n 
Read this post and revert here if this is the same problem as yours.
Please also post brand, type and picture of the laser.

arkypita, 2018-09-16:
> Dont know why you dont get the Grayscale option when it's the same version as mine, you would have to ask Arkypita, the developer why.

The grayscale conversion option is available only if you load a colored image. If the image is already a greyscale or bw no conversion option is shown.

[image]

arkypita, 2018-09-16:
@StuartB4 

> Just realised why you have no Grayscale settings box, it only to appears if you load a colour image like a photo in. 

You have got it!

More in detail: lasergrbl "detect" if the image "look like" grayscale by sampling some pixels of the image. If all the sampled pixels are gray (same R-G-B values less or more, with some tollerance) the image is considered as grayscale and no color conversion option are shown.

Rick23n, 2018-09-17:
[image]
[image]
[image]
These are the photos of my Laser and control board containing the Arduino(NANO(324p)/LX-Nano) 
First to answer your question Arkypita ; the first horizontal line, the laser fires but it only burns at 25% of full power. I have set up a 5mm inset befor it starts.

fantilator, 2018-09-17:
@Rick23n 
First thing I spot, put switch on TTL mode.
Second: you TTL- (open hole next to the yellow wire) is not connected, does your manual say something about connecting the TTL-?

try to do following things please.
1) connect laser in software, type in "M3 S0" command
2) What does the laser (normally the fan should run, but laser is off (because you run it at 0%)
3) pull out yellow cable, while laser is running (TTL) see what this does. (2 options, laser stays off, laser shoots on)
4) turn machine completely off
5) replug cable back
6) connect TTL- (empty connection next to yellow cable) to your black wire of the power supply. (just strip 2 end of a wire and prone them in) make sure you do not short circuit with the red wire.
7) turn machine back on and connect.
8) type M3 S100 and see what it does.

If you should see or hear any unusual things switch machine off right away.

I also see that if laser mode is enabled, you should use M4 Sx instead of S3. with X a value between 0 and 255
The usage of the laser within g-code is simple:

M3 turns the laser on (with constant power),
M4 turns the laser on with adaptive power,
M5 turns the laser off.
S1 - S255 sets the power of the laser (0 - 100%)

Rick23n, 2018-09-17:
The "Manual" is very basic and mostly in Chinese. What little English there was isn't helpful so to get this far I've relied on people such as yourselves and am very appreciative of your help.
Something interesting - I experimented with the Laser power with the M4 Sxx command and found that to turn the Laser ON but NOT burn my final setting was M4 S.0125 so Stuart I have set this figure in the "Laser ON/OFF" button but it does open up yet another road to follow.

Rick23n, 2018-09-17:
fantilator - I did reply to you but it is not showing- I will repeat - Command M3 S0 does NOT turn the laser ON but when I remove the yellow TTL cable, the laser comes ON full power. I have found that I CAN adjust the power but with strange settings S .0125 +++. I am not sure about grounding the pin beside the TTL yet but may do if all else fails. Yes I am using M4 with the laser mode now.

arkypita, 2018-09-17:
- In my opinion grounding of TTL connector is not needed because the TTL ground is shared with power ground.

- If detaching TTL result in "ON FULL power" seem that your driver has an inverted logic (remember what I say some post ago about TTL control: 5V should be LaserON, 0V should be LaserOFF)

Did you have a multimeter? Can you measure voltage between GND and TTL pin and check that it vary with the S value?
- M3 S12000 (should be 5V)
- M3 S6000 (should be about 2.5V)
- M3 S0 (should be 0V)

If you don't have a multimeter, could you put a LED with 220ohm resistor series, between TTL and ground and see what append to its intensity? S12000 should be bright, S6000 half bright etc.

arkypita, 2018-09-17:
**More:** As I previously say "The TTL pin is on pin D11" This is the GRBL standard. I don't know nothing about benbox but maybe the pin is not the same. If grbl control the pin D11 but the TTL cable is connected on another pin the problem would be explained.

It is not possible to see it by the picture, you should check it using multimeter.

Disconnect your pwm connector from the driver side, put multimeter in "beep" continuity function, insert one tip inside pwm connector and with the other tip test each arduino soldering.

Only D11 should "beep" when probed

Rick23n, 2018-09-18:
Arkypita thank you for your interest. Yes only D11 shows continuity on TTL. At first at M3 S12000 when I measured the voltage there was no change at S6000. Then I started at S1 and these are the figures it showed on TTL ;
S1 = 38.2mv
S2 = 56.6mv
S5 = 111.7mv
S10 = .203v
S50 = .936v
S75 = 1.395v
S100 = 1.85v
S200 = 3.69v
S255 =4.7v
S1000 = 4.7v
S12000 = 4.7v
When I engraved an image and set Sxxxx ; Protocol Settings Buffered / Fast , Raster Unidirectional Engraving (the first voltage is on the "away" pass the second is on the "return" pass.
S2 = 2.03/2.36v
S10 = 2.03/2.8v
S100 = 2.03/3.09v
S255 = 2.03/3.13v
If I set  single burn (uncheck Unidirectional and restart GRBL ) then the voltage shows 2.03v on the first pass and 3.13v on the second pass.
I hope I have explained this clearly.

arkypita, 2018-09-18:
> Arkypita thank you for your interest. Yes only D11 shows continuity on TTL. At first at M3 S12000 when I measured the voltage there was no change at S6000. Then I started at S1 and these are the [...]
 
For what you say, look like that you modulate power in a range of S0-S255, not S0-S12000 as you say some post ago.

[image]

Do you have changed those values to 0-255, correct?

Rick23n, 2018-09-18:
Yes, in all cases I set the power to Sxxx in "Target Image" and also in grbl Config $30 with Laser Mode enabled changing from S0 up to S255. When I used setting above S255 it did not change from 2.03/3.13 Volts. on the TTL line. When I posted before using 0-12000 I was not sure of what I was doing and it seemed that I got a better burn at 12000 but now i am not so sure because 255 IS at full power (4.7v)

Rick23n, 2018-09-26:
Thank you to the above for your support. I think I have it all sorted out now except for the image size that is shown in Laser grbl in mm actually prints 3x the size but I can work with that unless there is an obvious setting that I can adjust.

StuartB4, 2018-09-26:
Don't know why it would burn bigger than it's supposed to.
Can you check your stepper motor movements are set correctly.

Easy way to do it is to open LaserGRBL and Connect.
Left Click the Right slider once then use the Up-Down keys on your keyboard to set it to 20 shown in the picture below.
[image]

Move the laser to about the middle of the X Axis rail.
Make a mark on the rail as close as possible to the laser on the left side of where the laser stopped. 
Click LaserGRBL Right arrow key once to move the laser 20mm's.
When it stops measure the distance it moved with a ruler from the mark on the rail
to the left side of the laser.
If the measurement is 20mm's then the stepper is set correctly.

If it's different to 20mm's you have to change the $100 and $101 settings.
Do the same for the Y Axis.

StuartB4, 2018-09-26:
Ok Rick.

Rick23n, 2018-09-27:
Settings $100 = 77 & $101 = 80.5 now is perfect thank you

arkypita, 2018-09-27:
Very happy of your resut.
Thanks to @StuartB4  for big effort giving support.

Will be nice if you want to draw a step-by-step guide to help other people converting a Benbox machine to a GRBL one. I think benbox is very limited when grbl has a lot more of support, UI programs etc

Rick23n, 2018-09-27:
I will try to explain to the best of my ability, but it may take some time to get it right. How should it be posted? On the forum? or is there somewhere else? I dont think Utube would be something that I could do.

StuartB4, 2018-09-27:
Hi Rick.
If you do decide to do a Tutorial / Guide, you might find a bit more info here that may help.
[Benbox Forum](http://benboxlaser.us/index.php)

Rick23n, 2018-09-30:
Thank you Stuart. I'm not sure if I will be able to yet, I am still learning. When I have it all sorted out I will certainly give more thought to sharing my experiences with everyone. I must admit the 16000mw laser is not living up to expectations. I would have thought cutting 3mm thick MDF (composite) board would have been an easy asignment but after full power at 50mm/min and 8 passes it leaves me puzzled

[45 more replies]

closed 2018-12-11 by arkypita