arkypita/LaserGRBL #438: Is my laser diode fried?

gisep, 2018-09-08:
Hi, today i encountered a problem while cutting some stuff.
It was a 3h cutting, so i divided it in 4 parts of 45min each.
Maybe because of this (I touched the laser more than one time and it never was hot) I encountered a power loseness, and controlling the laser i noticed that it has got now like a "double" laserpoint (i attached the image showing it). I've got another lens, so i changed the lens but the problem persists. Maybe it' starting to fry? 
Please tell me if you can imagine what kind of apocalypse is coming to my lasercutter.
Bye :)
[image]

arkypita, 2018-09-16:
A lot of chinese diode are sold declaring more power than their real power, usually with driver board that gives more current that the diode can sink/substain. This is why they die quickly.

This is because the buyer usually want to buy the more powered laser with the cheapest price, so if you sell a diode of 20W at 40$ you can sell tons of them. Don't mean if they die in a month or less.

The reality is that a good diode + driver and heatsink can cost 90$ for 2W of power.

I don't know nothing about your diode, price, power, heatsink, driver, power source etc, but if it shows two dots... good news for you.

Maybe you can ask on https://laserpointerforums.com community, there are a lot of skilled people (but actually the site look down).

gisep, 2018-09-18:
Thanks for the answer Arkypita,
I've got a "cheap" chinese laser with heatsink and driver, declared 15W, paid 100€.
I've seen that community but i can't find nothing really helpful despite didn't see it all.

Used for 4 months with no problems, then had this change and now i can (luckily) only engrave but no cut. I'm afraid i've got to become aware i wasted my money.
Can you indicate me a coarse spending to a laser that can cut at least a 3mm plywood and suggest some models and sites where to buy a decorous one?

I will eventually try to write a topic in that community..

fantilator, 2018-09-18:
@gisep 
I have bought an 8W laser from http://endurancelasers.com
Perfect support, quick delivery and very powerfull lasers.

I cut 12mm plywood in 25 passes.

Meanwhile they already have 20W lasers.

arkypita, 2018-09-19:
http://endurancelasers.com sounds to be very good quality.

Also [dtrlasershop](https://sites.google.com/site/dtrlpf/home/laser-cnc-engraving) is a good source for laser, but they don't sell a full module with embedded heatsink, driver, fan but only spare parts

gisep, 2018-09-21:
thanks for all suggestions friends!
I'll get a little pause with lasercutting (I don't have any money for this now and also have to finish my studies) but i'll take your words in some weeks as the new starting point. And I'll be back! :)

closed 2018-09-30 by arkypita