arkypita/LaserGRBL #441: Continuos jogging

manisaranbala, 2018-09-16:
Continuos jogging is possible i send sum links low grid .3 mm fr 1200 and hi grid. 2mm fr 4500. One time press button time
Delay again again press internally. So
Low and high slect. Jog running pl try
https://github.com/gerritv/Grbl-Panel/tree/master/RepeatButton

arkypita, 2018-09-17:
As you can see: https://youtu.be/OH7WgJeMukE

manisaranbala, 2018-09-17:
Sorry I did not test it properly and tested it and did not work thanks
i think grbl continuos jogging not working

arkypita, 2018-09-18:
In reality it is not entirely impossible, but it is complicated to obtain.

Grbl does not provide a jogging protocol that can be easily used for continuous jogging, so you have to do some trick to get it.

Since jogging commands are queued and executed, you need to avoid queues too many commands.

One possibility is to monitor the current position and see when you reach the target position (of a previous jogging command) and only then send the next command. 

This may sound good but you have to compete with the fact that the position reached may not be exactly equal to the one requested (because of stepper resolution), so you have to do some "near position" test.

It is also possible to continue jogging only with small increments of position, for example 1mm at a time, because otherwise you cannot stop moving at the exact moment when you release the button (as if there is a jog in progress from 10cm grbl stops only after moving 10cm)

arkypita, 2018-09-18:
Anyway I googled about it and looks that could be possible, so I will explore the possibility.

https://github.com/gnea/grbl/wiki/Grbl-v1.1-Jogging
https://github.com/grbl/grbl/issues/669

closed 2018-09-30 by arkypita