arkypita/LaserGRBL #444: Lead-in, lead-out lines to prevent scorching with M3 mode?

clintonefisher, 2018-09-19:
Hi!  I'm new to LaserGRBL, and I cannot use M4 to control acceleration-caused scorching because my laser is high-power and non-linear.   Using M4  get too many blank spaces and irregularities.   When I use M3, after some min/max power tuning, the grey scales come out decent *except* for the beginnings and ends of lines, where my CNC3018 slows down and scorches the wood.   An obvious fix, even recommended in the GRBL laser-mode docs, is to add blank lead-in and lead-out spaces before and after each linear scan-line, to allow for proper acceleration and deceleration before and after the laser is turned on.   
Has anyone attempted to implement this, or is it perhaps already there and hidden somewhere?
Thanks and thanks for everyone's hard work on this fantastic utility!

clintonefisher, 2018-09-19:
Upon further review, it looks like this feature has already been requested in issue #102 "Acceleration Offset".   The author of the issue mentioned 5mm as a buffer.    This would make a great default, but I would also make it a user-selectable variable.

BTW, I have some coding chops and might be available to help add this feature if one could point me to the appropriate modules, etc..

clintonefisher, 2018-09-20:
arkypita wrote:

"Another solution is to add a very light margin to the image   (not white, as white will be optimized/skipped by lasergrbl). An  "acceleration border" feature should be already in dev roadmap"

Arkypita, thanks for this workaround!   I'd considered adding a border ...just not a light (invisible) one.  Perfect!  Thanks!!

BTW, I took a quick look at the code and am excited to see if I can hack this feature in there myself.  
We shall see.  ;)

In the meantime, it's nice to have a solid workaround.   Thanks again for all your hard work!

arkypita, 2018-09-22:
Hi @clintonefisher 
I really appreciate to insert features submitted by LaserGRBL user. The best way to submit your code, once finished and tested, is to submit a pull request. If you are new in github world you can ask to @Effer or @Ranthalion how to do that.

> adding hard-coded accel and decel buffer zones as segments in the GetSegments() method.
> I'm not sure how to get these buffers past optimization.

Optimization remove all the white spaces. Maybe you can put an `mColor` of one in that segments to set them as very-light non-white `GrblCommand`. This will make them passing the optimization.

```
private abstract class ColorSegment
{
	protected int mColor;
```

A better way is to mark that `ColorSegment `as special, adding a `bool mAccel` variable, setting to `true` on that segments, and produce some special `GrblCommand` for them in the `foreach (ColorSegment seg in segments)` that produce GrblCommand from ColorSegment before optimize, and then let the OptimizeLine2Line skipping optimization fot GrblCommand of that type.

arkypita, 2018-09-22:
> Arkypita, any thoughts or advice before I dive in to the UI?

IMHO, my suggestion is to add the UI for this feature inside the `Import-Raster-Image dialog` both for Line2Line and Vectorization+Filling (filling use line2line).

Target-Image dialog should remain as-is, because the adding of an acceleration border should not impact in how the user compute the final image size and positioning. The image size should remain the same of the desired target image, regardless of using of acceleration offset

clintonefisher, 2018-09-22:
Thank you for the response and the encouragement and hints as to getting the accel buffer property into the ColorSegment and/or GrblCommand!   I've been coding for decades, but this will be my first Github contribution.  :)
> IMHO, my suggestion is to add the UI for this feature inside the `Import-Raster-Image dialog` both for Line2Line and Vectorization+Filling (filling use line2line).
>
> Target-Image dialog should remain [...]

Understood.   I do have some ideas as to how it might still fit into the Target image dialog to let the user see the desired target image size, the accel-border size, and the resultant overall image+border size all in one place ...and be able to make adjustments to all size parameters in one place.   

I will mess around with it and only submit the changes if and when when it feels correct from a user-standpoint and also have your blessing.  :)

And with that, I'm off to do the first engraving with my hard-coded border to see how it works!
Cheers!

clintonefisher, 2018-09-23:
Update:  The hard-coded buffer did the job..  No more burnt edges.
Though it's still un-optomized g-code, it did the job. 
Now, time-permitting, I'm on to the UI and then optimization bypass.   :)

In the meantime:  In case anyone's DESPERATE and wants to brute-force a 10mm buffer, here are the diffs off the latest version of **GrblFile.cs** as of this post.  WARNING: This shifts your zero point and is an awful awful very bad HACK!  As with any such things: USE AT YOUR OWN RISK!   :)

(I apologize for the formatting - I'm a GitHub first-timer trying to figure out how to post a patch and get the nice color-coding and properly deal with tabs and whitespace.  But I wanted to at least get this info out to the needful.)

```
@@ -365,11 +365,12 @@ private void ImageLine2Line(Bitmap bmp, L2LConf c)
 					temp.Add(new GrblCommand(String.Format("{0} {1}", fast ? "G0" : "G1", seg.ToGCodeNumber(ref cumX, ref cumY, c))));
 				else
 					temp.Add(new GrblCommand(seg.ToGCodeNumber(ref cumX, ref cumY, c)));
 			}
 
-			temp = OptimizeLine2Line(temp, c);
+            // accel decel buffers would be optimized out, so don't optimize
+            // temp = OptimizeLine2Line(temp, c);
             list.AddRange(temp);
 		}
 
 
 		private List<GrblCommand> OptimizeLine2Line(List<GrblCommand> temp, L2LConf c)
@@ -446,22 +447,38 @@ private List<ColorSegment> GetSegments(Bitmap bmp, L2LConf c)
 				{
 					bool d = uni || IsEven(i); //direct/reverse
 					int prevCol = -1;
 					int len = -1;
 
+                    int accDecBufferLenInMM = (int)(10 * c.res); // CEF set accel decel buffer to 10mm for now.
+
+                    // CEF: This is where we add the Accel buffer segment
+                    if (h)
+                        rv.Add(new XSegment(0, accDecBufferLenInMM, !d)); //Add Accel Buffer
+                    else
+                        rv.Add(new YSegment(0, accDecBufferLenInMM, !d)); //Add Accel Buffer
+
                     for (int j = d ? 0 : (h ? bmp.Width - 1 : bmp.Height - 1); d ? (j < (h ? bmp.Width : bmp.Height)) : (j >= 0); j = (d ? j + 1 : j - 1))
 						ExtractSegment(bmp, h ? j : i, h ? i : j, !d, ref len, ref prevCol, rv, c); //extract different segments
 
 					if (h)
 						rv.Add(new XSegment(prevCol, len + 1, !d)); //close last segment
 					else
 						rv.Add(new YSegment(prevCol, len + 1, !d)); //close last segment
 
+                    // CEF: This is where we add the Decel buffer segment
+                    if (h)
+                        rv.Add(new XSegment(0, accDecBufferLenInMM, !d)); //Add Decel Buffer
+                    else
+                        rv.Add(new YSegment(0, accDecBufferLenInMM, !d)); //Add Decel Buffer
+
+
+
                     if (uni) // add "go back"
 					{
-						if (h) rv.Add(new XSegment(0, bmp.Width, true));
-						else rv.Add(new YSegment(0, bmp.Height, true));
+						if (h) rv.Add(new XSegment(0, bmp.Width + (2*accDecBufferLenInMM), true));
+						else rv.Add(new YSegment(0, bmp.Height + (2*accDecBufferLenInMM), true));
 					}
 
 					if (i < (h ? bmp.Height - 1 : bmp.Width - 1))
 					{
 						if (h)
@@ -504,21 +521,26 @@ private List<ColorSegment> GetSegments(Bitmap bmp, L2LConf c)
 					int prevCol = -1;
 					int len = -1;
 
 					int z1 = slice < h ? 0 : slice - h + 1;
 					int z2 = slice < w ? 0 : slice - w + 1;
+                    int accDecBufferLenInMM = (int)(10 * c.res); // CEF set accel decel buffer to 10mm for now.
+
+                    rv.Add(new DSegment(0, accDecBufferLenInMM, !d));  // CEF: Add accel buffer
 
 					for (int j = (d ? z1 : slice - z2); d ? j <= slice - z2 : j >= z1; j = (d ? j + 1 : j - 1))
 						ExtractSegment(bmp, j, slice - j, !d, ref len, ref prevCol, rv, c); //extract different segments
 					rv.Add(new DSegment(prevCol, len + 1, !d)); //close last segment
 
                     //System.Diagnostics.Debug.WriteLine(String.Format("sl:{0} z1:{1} z2:{2}", slice, z1, z2));
 
+                    rv.Add(new DSegment(0, accDecBufferLenInMM, !d)); // CEF: Add decel buffer
+
                     if (uni) // add "go back"
 					{
 						int slen = (slice - z1 - z2) + 1;
-						rv.Add(new DSegment(0, slen, true));
+						rv.Add(new DSegment(0, slen + (2*accDecBufferLenInMM), true));
 						//System.Diagnostics.Debug.WriteLine(slen);
 					}
 
 					if (slice < Math.Min(w, h) - 1) //first part of the image
 					{

```

arkypita, 2018-09-24:
Hint about github usage:

- fork your own LaserGRBL repository by clicking "fork" button
- download and install [github desktop](https://desktop.github.com)
- download your repository clintonefisher/LaserGRBL  using "download button" and chose "open in desktop"
- now edit your code and use github desktop to push your changes in your own fork
- when you are satisfied of your job you can [create a pull request](https://help.github.com/articles/creating-a-pull-request-from-a-fork/) for merging your changes in original repository

[image]

[image]

clintonefisher, 2018-09-24:
Ahh, that makes sense to create my own local fork.    Thank you for the hint!

RunnyBore604, 2018-09-30:
@clintonefisher, you can try my tool for generating g-code from images. But it's just a tool, not controll software. https://yadi.sk/d/1WtJudo23ZugjZ

closed 2018-12-06 by arkypita

xsrf, 2025-02-08:
Sorry for digging up this old issue, but I cannot find any option to add a lead-in/lead-out in M3 raster mode in LaserGRBL v7.14.0. Was this ever implemented? Why was it closed?