arkypita/LaserGRBL #445: 3 AXIS CNC ROUTER

manisaranbala, 2018-09-20:
I used 3 axis cnc router. your software  v 2.8.52 successfully running but one problem z axis  not showing right side  corner and resume from specific line  (z axis )  please  add z axis WCO 
first flower damaged 
https://youtu.be/YjwxG4P5-fM

arkypita, 2018-09-20:
Dear @manisaranbala 

LaserGRBL is designed for laser engraver, however its robustness in gcode streaming makes it a good solution even for sending file to CNC.

I read your message and watched the video, but I could not figure out what the problem is. I ask you to try to explain better the problem you are talking about.

manisaranbala, 2018-09-21:
I used resume from specific line that time damaged job one line i send image.  I think your. Software x and y axis resumed program. Not z axis please. Add z axis. Resume safe hight than running .

manisaranbala, 2018-09-21:
https://drive.google.com/open?id=1bICFkJ65-pLv93NTID5CPF6-KOtBKlD5
[image]
[image]

arkypita, 2018-09-21:
Understand.
I'll get a look at this issue.

closed 2018-09-30 by arkypita