arkypita/LaserGRBL #450: Laser Head move (full speed with no laser) to show working area before engraving

Killthgz, 2018-09-28:
As the topic, is it possible to do that? if so I think this will help to align material before engraving or cutting.
I ever see this function in youtube. LaserHead will move as square at the edge of engraving area. Thank you.

StuartB4, 2018-09-28:
[image]

[image]

Changing the F setting will change the speed, up to the maximum your machine is set for in GRBL.

You can copy and paste this code:
F1000
G0 X[left] Y[bottom] 
G1 X[right]
G1 Y[top]
G1 X[left] 
G1 Y[bottom]

Killthgz, 2018-09-30:
Wow!! perfect!!!
Thank you very much for your advice ^____^. @StuartB4

closed 2018-09-30 by arkypita